Sevenstax/FreeV2G #93: mini cooper not charging

codderdreamer, 2022-10-12:
[minicooper.zip](https://github.com/Sevenstax/FreeV2G/files/9766992/minicooper.zip)

Hi,

It charges for a second and came Request Stop Charging. I activated port mirror. 
`           evse.setSchedule(schedule)
            evse.whitebeet.networkConfigSetPortMirrorState(1)
            # Start the EVSE loop
            evse.loop()
            print("EVSE loop finished")`

`    def networkConfigSetPortMirrorState(self, value):
        if not isinstance(value, int) or value not in [0,1]:
            raise ValueError("Value needs to be of type int with value 0 or 1")
        else:
            self._sendReceiveAck(self.netconf_sub_id, self.netconf_set_port_mirror_state, value.to_bytes(1, "big"))`

but I guess port mirror didn't open?

what's wrong with the mini cooper sending shutdown request?

jpo-stx, 2022-10-17:
Hi @codderdreamer 

You receive an update of the charge parameters (0x85) in frame no 560 where the error code is set to "6: Charging Voltage Out Of Range". But when I check the rest of the parameters it seems that the present current is too low:

target voltage: 411.6 V
present voltage: 411 V
target current: 5 A
present current: 1 A

jpo-stx, 2022-10-24:
Since you haven't responded I expect that the issue was resolved with the information given. If not fell free to reopen it.

closed 2022-10-24 by jpo-stx