Sevenstax/FreeV2G #95: No module named pylibpcap

kramzarales, 2022-10-17:
**Your Setup**
Platform:  EVSE
Firmware Version: i.e. V01_00_05
Host Controller Interface: Ethernet
Host: FreeV2G (EV_v1.0.4_0) 

**Describe the bug**
When running "sudo python3 Application.py "eth0" -m c4:93:00:22:22:22 -r EVSE"   after folowing the instructions, i get the folowing error:  ModuleNotFoundError:  No module named "pylibpcap".  I could not find this package anywhere, and from what i understand it was replaced by scappy in this project.   I ma runing python 3.7.3,   scappy version 2.4.5,   running on physical machine running ubuntu 22.04.

**Additional context**
Attempting to install FreeV2G and connect to WhiteBeet EVSE official dev board.

jpo-stx, 2022-10-17:
Hello @kramzarales ,

>Your Setup
Platform: EVSE
Firmware Version: i.e. V01_00_05
Host Controller Interface: Ethernet
Host: FreeV2G (EV_v1.0.4_0)

something is wrong here, there is no version 1.0.5 for the EVSE. First of all you should use the correct FreeV2G tag for your version.

To the lib problem. We changed from scapy to pylobpcap for Linux systems because it has a better performance than scapy. The package you need is python-libpcap:

```bash
pip install python-libpcap
```

We need to add this to the readme!

kramzarales, 2022-10-18:
Yes i incorectly labeled tthe request. I just used the git clone https://github.com/SEVENSTAX/FreeV2G  command to get the latest master branch, so i am guesing  that i have the latest [EVSE_v2.0.1_3] version.

As for the problem, python-libpcap is already installed (v 0.4) and it still gives the same error.

Interestingly when i run without sudo i get a different error:  The script appears to run correctly untill "Initializing framing interface", but then it gives a PermisionError for _socket.socket._init_(self,family,type,proto,fileno). My guess would be that sudo is required, but the documentation doesnt specify.

Failed to initialize the framing interface on \"{}\"".format(self.framing.sut_interface))
ConnectionError: Failed to inizialize the framing interface on "eth0"

eth0 is the correct (and only) interface on the machine

vbudko, 2022-10-18:
Hello, 
Please note that you have to specify correct MAC address as well.
In the log above you are using MAC address from the example, the actual MAC address of STM32 HCI is MAC address of PLC modem minus 2 (QCA7005, it is printed on the label on the shielding as well as coded into QR code).
BR, Vasily

kramzarales, 2022-10-18:
hello,
The mac adress is specified correctly, i just copied the comand from the instructions in this post.

jpo-stx, 2022-10-20:
> As for the problem, python-libpcap is already installed (v 0.4) and it still gives the same error.

When you run the command with sudo the package needs to be installed for sudo. You can get around this problem by installing the package with sudo or by using a python virtual environment.

closed 2022-10-20 by kramzarales