Sevenstax/FreeV2G #97: Very long delay between Welding Detection and Session Stopped

st4hli, 2022-10-25:
**Your Setup**
Platform: EVSE
Firmware Version: i.e. V02_00_01
Host Controller Interface: Ethernet
Host: FreeV2G (newest commit f4a8566a from PnC branch)

**Describe the bug**
When a running charging process is stopped by an EV, the EVSE whitebeet waits for many (often more than 20) seconds until it sends the Session Stopped status message to the host. According to our EV emulator, the ISO 15118 session is already stopped too (SessionStopRes sent), but the host still has to wait for the Session stopped message to finalize the charging session.
Since the documentation doesn't explain this behaviour (also, the diagrams for this part of the charging process are missing), please clarify why this happens and how we can end the charging process quicker. 
The current situation does not allow to start a new charging process sooner, e.g. by unplugging and plugging again. I can force close the session with self.whitebeet.v2gEvseStopListen() but this causes errors on both the EV as well as thr EVSE. 
In our own implementation, all of our repeated charging process test procedures fail due to this reason.

**To Reproduce**
Start a charging process with FreeV2G as EVSE and stop the charging process from the EV side.

**Expected behavior**
The delay between Welding Detection status message and Session Stopped status message to the host should be shorter.

**Logs**
"Request Stop Charging" received
Timeout: 4541
Timeout: no
Thu Oct 13 01:26:04 2022 "Welding Detection Started" received
Thu Oct 13 01:26:32 2022 "Session stopped" received
Closure type: 0
EVSE loop finished
Goodbye!

[wireshark_log.txt](https://github.com/Sevenstax/FreeV2G/files/9858481/wireshark_log.txt)

jpo-stx, 2022-10-25:
Hi @st4hli ,

thank you for the detailed description and the log file.

This is really strange that the session stop status message takes so long.

We see the TCP FIN from the EV in packet no 1088, shortly after the session stop response from the EVSE. The TCP FIN from the EVSE follows 2s later in packet no 1129. An then it takes ~25s until the session stop status message is received by the host. The session stop message should be send by the EVSE when the TCP connection is fully closed.

With the setup I have here I wasn't able to reproduce the issue. I will try to get a setup running which behaves like yours to analyze the issue.

st4hli, 2022-10-27:
Hi @jpo-stx ,

thank you very much, fixing this issue is crucial for us. If you need more information to reproduce the bug, just contact me.

jpo-stx, 2022-11-02:
I did further investigations and it seems that the TCP of your EV is not handling the TCP connection closure correctly:

[image]

As you can see in the log the EV is triggering a closure of the TCP connection with a FIN (1088) which is acked by the EVSE (1089). Then the EVSE sends the FIN (1129), but this FIN is not acked by the EV, which leads to a retransmission of the FIN packet (1212, 1377), which in the end results in a timeout and the session stop message is sent (1658).

I will see if we can implement a workaround for this behavior.

st4hli, 2022-11-03:
Thank you for the detailed analysis of the issue. I forwarded it to the developer of the EVSE communication board because it seems to be a problem on their side. 
Nevertheless, a workaround for this issue would be very helpful.

nststx, 2023-04-19:
Closing due to new firmware update which should fix this issue. If this issue still persists with the new firmware update, feel free to reopen it.

closed 2023-04-19 by nststx