Sevenstax/FreeV2G #98: Reference implementation in SPI?

kramzarales, 2022-10-26:
I would be interested to know if there are any plans for a version of FreeV2G that use the SPI interface on the whitebeet instead of Ethernet?

If there isnt, would someone perhaps have any contacts or code that someone who has succesfully done this SPI modification , would be willing to share?

jpo-stx, 2022-10-27:
Hi @kramzarales ,
we have an implementation for the Raspberry Pi, which is available in the plug and charge branch:
https://github.com/Sevenstax/FreeV2G/blob/plug_and_charge

Have a look at the SPI adapter:
https://github.com/Sevenstax/FreeV2G/blob/plug_and_charge/SpiAdapter.py

You can start the application with SPI interface like this:
```
python3 Application.py spi -i spidev0.0 -m c4:93:00:22:22:22 -r EVSE
```

kramzarales, 2022-10-27:
Thank you so much for this.

Just a heads up. https://github.com/Sevenstax/FreeV2G/blob/plug_and_charge  doesnt appear to work.

closed 2022-11-02 by kramzarales