Sevenstax/FreeV2G #106: sub id in the code are different with the one in the manual

zhanglongqi, 2022-11-15:
Your Setup
Platform: EVSE
Firmware Version: i.e. V01_01_07
Host Controller Interface: Ethernet
Host: FreeV2G (EVSE_v1.1.7_1) 
git commit hash: 12eb962

**Describe the bug**
as the title says

[image]
[image]

jpo-stx, 2022-11-15:
@zhanglongqi ,
I think you do not use the correct manual version for your WB firmware version. If you use the WB firmware version 1.1.7 please use the manual from the tag https://github.com/Sevenstax/FreeV2G/tree/EVSE_v1.1.7_1 which is manual version 2.11.2.

See 19.8.4 Charge Parameter Discovery chapter in the manual:

[image]

closed 2022-11-15 by jpo-stx

zhanglongqi, 2022-11-16:
Thanks, I can find the manual change log at the end of the manual. Is there a change log for the firmware? Who should I approach if I want to update the firmware?

jpo-stx, 2022-11-21:
Please read the introduction chapter https://github.com/Sevenstax/FreeV2G#introduction

You should be able to donwload the latest firmware from Codico https://downloads.codico.com/misc/plc/