Sevenstax/FreeV2G #109: Module did not accept command 27:a0, return code: 17

Malinda-G, 2022-11-18:
**Your Setup**
Platform: PEV/EVSE
Firmware Version: i.e. EV_v1.0.4_0 & EVSE_v1.1.7_1
Host Controller Interface: Ethernet 
Host: FreeV2G (EV_v1.0.4_0) & (EVSE_v1.1.7_1)

**Describe the bug**
We are using both EVSE and EV Whitebeet Carrier boards. The code of EV board hangs at "Start SLAC Matching" while EVSE hangs at "Start V2G" and then crashes soon after receiving "We did not receive a frame before the timeout of 30s"  and the code ends from the EVSE side. The output prints are attached below. 

[image]
[image]

jpo-stx, 2022-11-21:
Hi @Malinda-G ,

you have an EV with firmware version V01_00_04 and use the tag [EV_v1.0.4_0](https://github.com/Sevenstax/FreeV2G/tree/EV_v1.0.4_0)
and you use and EVSE with firmware version V01_01_07 with tag [EVSE_v1.1.7_1](https://github.com/Sevenstax/FreeV2G/tree/EVSE_v1.1.7_1)
correct?

I need to downgrade the modules I have here and see if I have the same problem.

There are also firmware updates available for EV and EVSE, I would recommend to use the latest versions.

Malinda-G, 2022-11-22:
Hi, 

Yes, the EV firmware version: V01_00_04 and using the tag EV_v1.0.4_0 and EVSE firmware version: V01_01_07 with tag EVSE_v1.1.7_1.  We also try using other versions just in case, but the results are still the same. From the EV side it gets stuck in the "Start SLAC Matching", and the EVSE in "Start V2g"

How can we update the module's firmware? We saw it is mentioned on the manual, but we can't find the necessary tools in sevenstax webpage.

Also, regarding the WhiteBeet Carrier, does the SW1 (FW VER selector) need to be in a special position? Now, both of them (EV and EVSE) are in the New FW position.

Thanks for the help!

jpo-stx, 2022-11-22:
The latest versions EV V01_00_06 and EVSE V02_00_01 are guaranteed to work with each other.

You should find the firmware update tool at Codico https://downloads.codico.com/misc/plc

Malinda-G, 2022-11-23:
Hi, 

We currently do not have access to the link you have provided. The NDA process might take a few weeks to get sorted from our end. Will it be possible to resolve the issue for the current versions we have mentioned in the mean time? 

For additional support I have attached the Wireshark logs for both EV and EVSE for the above mentioned issue if that is of any assistance. 

In addition to the above mentioned issues, if we un-plug and re-plug the CP when the code hangs at the specific places the code continues to run correctly and the charging process works as intended. 

It would be very helpful if you could resolve this issue. Thank you very much!
[WireShark Logs.zip](https://github.com/Sevenstax/FreeV2G/files/10071813/WireShark.Logs.zip)

jpo-stx, 2022-12-02:
Wo do not provide updates for older versions.

From the wireshark log I can see that there is something wrong with the EV host application. In paket no 687 the whitebeet informs the host that the SLAC process succeeded. Nearly 30s after this the host starts the application (paket no 719). At this point in time the host of the EVSE already shut down the application.

You can try changing the timeout value here:
https://github.com/Sevenstax/FreeV2G/blob/4b7b7ac3915dea45a850824fd797175e8045162e/Whitebeet.py#L1958

lho-stx, 2022-12-23:
Hey @Malinda-G,

does the proposed solution worked out for you? If not please let me know, so we can investigate this issue further. Else the issue will be closed.

Thank you,
lho

lho-stx, 2023-01-10:
Closing this issue due to inactivity.

closed 2023-01-10 by lho-stx