Sevenstax/FreeV2G #110: Module did not accept command 27:6f, return code: 17

zhanglongqi, 2022-11-24:
**Your Setup**
Platform: EVSE
Firmware Version: i.e. V02_00_01
Host Controller Interface: Ethernet
Host: FreeV2G  d59d389 - (tag: EV_v1.0.6_1, tag: EVSE_v2.0.1_4) 

**Describe the bug**
I tested WB EVSE board with WB EV board. It's OK. I got this error when I try the same WB EVSE board with a real EV.

**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**

## log from FreeV2G
```shell
# longqi @ l0 in ~/FreeV2G on git:master x [15:10:17] 

$  cd /home/longqi/FreeV2G ; /usr/bin/env sudo -E /home/longqi/FreeV2G/.venv/bin/pyto

pter/../../debugpy/launcher 42525 -- Application.py eth -i enp0s31f6 -m C4:93:00:34:AA:E5 -r EVSE 

[sudo] password for longqi: 

Welcome to Codico Whitebeet EVSE reference implementation

Initiating framing interface

iface: ETH, name: enp0s31f6, mac: C4:93:00:34:AA:E5

WHITE-beet-EI firmware version: V02_00_01

Set the CP mode to EVSE

Set the CP duty cycle to 100%

Start the CP service

Start SLAC in EVSE mode

Wait until an EV connects

EV connected

Start SLAC matching

Set duty cycle to 5%

SLAC matching successful

Set V2G mode to EVSE

Start V2G

"Session started" received

Protocol: 0

Session ID: fd1071b2f3898257

EVCC ID: 9012a100b94a

"Payment selcted" received

Selected payment method: 0

"Request Authorization" received
[... 375 lines cut ...]

[  386.307] [NOTICE] V2GDIN12: EV requested to stop session.

[  388.447] [NOTICE] V2GSVC: TCP: Client disconnected or connection closed by server

[  388.448] [NOTICE] V2GAPP: Received event 'Session stopped'

[  388.450] [NOTICE] V2GSVC: TCP: Listen on port 56851 for non-TLS

[  388.451] [NOTICE] V2GSVC: TCP: Listen on port 56852 for TLS

[  599.313] [NOTICE] V2GCPS: Change State from B to A

[  599.313] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()

[  599.314] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_STATE

[  599.315] [NOTICE] V2GCPS: STATE A�

 ```

 

**Additional context**
Add any other context about the problem here.

jpo-stx, 2022-12-02:
Which EV are you testing with?

Please capture a wireshark log on the ethernet interface of whitebeet and activate the port mirror.
Add `evse.whitebeet.networkConfigSetPortMirrorState(1)` before the evse loop:
https://github.com/Sevenstax/FreeV2G/blob/d7ec5823ee882e357162fd07e5798ddc3ac91ea1/Application.py#L103

zhanglongqi, 2022-12-15:
The car we test is IONIC 5. the log files are avaiable in this link. https://mega.nz/file/18QX2DLB#7mss1Te6TPCa6fcJP7DsSpB0h8N-mWqlSWuAni0tQr0

jpo-stx, 2022-12-20:
After ContractAuthorizationRes the EV should send ChargeParameterDiscoveryReq. In your case the EV sends a PowerDeliveryReq:

[image]

This is not according to specification. Before PowerDelivery there is also CableCheck and PreCharge procedure.

zhanglongqi, 2022-12-22:
Thanks. I will try to disable `DIN70121-2`.

Is it possible that we can follow the procedure of the EV to skip the `ChargeParameterDiscoveryReq` so it can continue the charging?

zhanglongqi, 2022-12-26:
May I know which car models you have tested and works well? Could you share a list of tested models so we can test?

zhanglongqi, 2022-12-30:
I tested another car (hyundai ioniq electric) and got the same error. It does not work at all if I remove the `DIN70121-2`.
the log in the link: 

https://mega.nz/file/E4YhGJrK#Kpv-SPEzpfpOeFhwpA_xyyX4lVsSpobEEeQ9Ery5r9w 
and 
https://mega.nz/file/F0ZHhTzB#S3pDq2y80wiAhteLrD6IXbPcV3ZVSpaZnV6rwh6dnck

jpo-stx, 2023-01-02:
The car you tested now only supports DIN70121-2 protocol. And it is not even doing the authentication sequence:

[image]

Can you try the same (disable DIN) with the other car?

And it is really strange that a Hyundai car is not communicating according to the standard. Where is this car comming from?

zhanglongqi, 2023-01-03:
we rent the cars from the open market. both of them can charge in Singapore.

May I know what car models you have tested and works well? Could you share a list of them?

zhanglongqi, 2023-01-03:
can we arrange a conference call after we test with ionic 5 on Jan-05?

zhanglongqi, 2023-01-05:
I tested with ionic 5 and disabled `DIN70121-2`, it seems working fine. 
https://mega.nz/file/N9RRCBxS#h6qHY4R_cn8jVq5D0__wPcdl7wvRNZ3gAa3Hl9-KBOw

closed 2023-01-17 by zhanglongqi

zhanglongqi, 2023-06-06:
> > I tested with ionic 5 and disabled `DIN70121-2`, it seems working fine. https://mega.nz/file/N9RRCBxS#h6qHY4R_cn8jVq5D0__wPcdl7wvRNZ3gAa3Hl9-KBOw
> 
> @zhanglongqi Could you please share what changes you have done to disable the DIN70121-2 beacause I am also facing same issue?
[image]

GovindMore, 2023-06-18:
> > > I tested with ionic 5 and disabled `DIN70121-2`, it seems working fine. https://mega.nz/file/N9RRCBxS#h6qHY4R_cn8jVq5D0__wPcdl7wvRNZ3gAa3Hl9-KBOw
> > 
> > 
> > @zhanglongqi Could you please share what changes you have done to disable the DIN70121-2 beacause I am also facing same issue?
> > [image]

Thank you for the response.