Sevenstax/FreeV2G #111: Firmware Update

ChargerH55, 2022-12-01:
Platform: EVSE
Firmware Version: V01_00_01
Host Controller Interface: Ethernet / SPI
Host: FreeV2G 

I want to update the Firmware of the board which is currently in version V01_00_01 to the latest version. I am running the .exe with the .fwu file provided on Windows as:

>StxFwUpdater.exe -f WHITEBEET_ISO15118_EVSE_V02_00_01.fwu -t 74:78:27:B0:33:6B -i "Realtek USB GbE Family Controller #3"
Welcome to the SEVENSTAX SFWU Updater 2.5

Initiating framing interface
Set time on the target to 2022-12-01T15:22:47Z
Starting update process
[!] Target did not answer acknowledge firmware update start, check whether it is in the right state

Can you tell me how to proceed from there?

Thank you in advance for your help.

jpo-stx, 2022-12-02:
Hi @ChargerH55 ,

have you tried rebooting the whitebeet before starting the firmware update?

If this does not help, can you provide a wireshark and console log?

ChargerH55, 2022-12-02:
Hi @jpo-stx ,
Thank you for your answer, I finally managed to update the firmware using a more recent .fwu file sent by CODICO Support.

However, I am still facing the same issue with the SLAC that I thought was due to the old version of the firmware I was using at first.
Here is the command :

_/FreeV2G-EVSE_v1.1.7_0$ sudo python3 Application.py enx0c3796785458 -m c4:93:00:21:50:8B -r EVSE
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: enx0c3796785458, mac: c4:93:00:21:50:8B
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
EV already connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching timed out
EVSE loop finished
Goodbye!

Do you know how I should proceed to tackle this problem ? 

Thank you in advance for your help.

Sincerely,

jpo-stx, 2022-12-02:
So you now have the firmware version 2.0.1?

Have you updated your FreeV2G host application to the latest version? https://github.com/Sevenstax/FreeV2G/tree/EVSE_v2.0.1_4

ChargerH55, 2022-12-02:
Now I have the firmware version V01_01_07 and I updated the FreeV2g to the one available FreeV2G-EVSE_v1.1.7_0.

jpo-stx, 2022-12-02:
Please provide a wireshark log of the ethernet interface in this case with port mirror activated as described here: https://github.com/Sevenstax/FreeV2G/issues/110#issuecomment-1334986058

From the log above it seems that there is no EV that starts the matching process within the time window and SLAC process times out on EVSE side.

ChargerH55, 2022-12-02:
[log_EVSE.zip](https://github.com/Sevenstax/FreeV2G/files/10141293/log_EVSE.zip)

jpo-stx, 2022-12-02:
When do you plug in the EV? When the CP state is read out the first time it is in state B, so the EV is already connected.

Can you try to start the whitebeet EVSE and plug in the EV when you see the print `Set duty cycle to 5%`.

ChargerH55, 2022-12-02:
FreeV2G-EVSE_v1.1.7_0$ sudo .venv/bin/python3 Application.py enx0c3796785458 -m c4:93:00:21:50:8B -r EVSE
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: enx0c3796785458, mac: c4:93:00:21:50:8B
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects
EV connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching timed out
EVSE loop finished
Goodbye!

jpo-stx, 2022-12-02:
What EV are you testing with?

Can you provide a console log? The UART USB port on the whitebeet, baudrate is 1Mbps.

ChargerH55, 2022-12-02:
I am trying to make it working with a Vector VCCU.

jpo-stx, 2022-12-02:
Do you get any helpful logs from this device?

codderdreamer, 2022-12-06:
why slack maching fails when the socket is already connected to the vehicle and it starts charging. slack mached when i plug in the socket afterwards. For example; If I don't remove the socket after starting and finishing the car charging once, there is a slack mahing error on the second. Also, the slack maching timeout mentioned here comes after 50 seconds when I try it, so when I extend the time a bit, slack maching failed comes up. For example; 60 instead of 50

nststx, 2023-04-19:
Closing due to inactivity. If the issue still persists, feel free to reopen.

closed 2023-04-19 by nststx

varshilgandhi, 2024-04-23:
Hey @CB-ctrl ,

As you mentioned in this about that you were facing the issue of the Firmware Update and SLAC matching timed out. 
Can you please tell me that did you able to solve this issue and if yes how did you solve this issue.
Becase I am also facing the same issue like you.
I update firmware version of EV board to V01_00_07 and EVSE board to V02_01_00.

But after that I give this below command I got the SLAC Matching Issue.

(.venv) varshil@varshil-Vostro-3520:~/Documents/Tirex/Latest Code 23-04-2024 (EVSE)/FreeV2G$ sudo .venv/bin/python3 Application.py eth -i enp1s0 -m c4:93:00:48:AC:45 -r EVSE
[sudo] password for varshil: 
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: ETH, name: enp1s0, mac: c4:93:00:48:AC:45
WHITE-beet-EI firmware version: V02_01_00
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects
EV connected
Start SLAC matching
Set duty cycle to 5%
We did not receive a frame before the timeout of 60s
EVSE loop finished
Goodbye!

Can you please help me to solve this issue if you already did it.

Your little help will be very much appreciated.

Thanks and Regards,
Varshil