Sevenstax/FreeV2G #116: Posibility of using Selected Energy Transfer Mode: 4: AC single phase

kramzarales, 2022-12-08:
**Your Setup**
Platform:  EVSE
Firmware Version: i.e. V02_00_01
Host Controller Interface:  SPI
Host: FreeV2G

This is a feature request/technical question, where i would like to know if it is possible to use FreeV2G in AC single phase charging mode.
We are mainly only interested in obtaining the SOC of the vehicle, which we can do with Selected Energy Transfer Mode: 1.
Other then this information we are mainly interested in charging the vehicle with only 16A single phase AC current.

In the Sevenstax documentation it states that Selected Energy Transfer Mode: 4 is used for single phase AC charging, but if we set it in FreeV2G it turns out an error, since it is not defined in the softwere.

My question is if having Selected Energy Transfer Mode set to 1, can cause problems if we then decide to charge with AC current via standard IEC 62196 CP pin comunication?
Would it still be possible to get SoC even if  Selected Energy Transfer Mode=4?

jpo-stx, 2022-12-09:
Hi @kramzarales ,

this is an interesting use case. Can you give me the whole configuration message? When you only select an AC charging mode you need to have ISO15118-2 protocol in your selected protocols list. What you cannot do is only select DIN70121-2 with only AC energy transfer modes, since AC charing is not supported by the DIN70121-2 specification.

For your use case if you start with DC energy transfer mode on your AC wallbox, I suppose that the EV will fail in the cable check sequence, but at that point in time we should already have received the state of charge of the EV. If you then stop the V2G and SLAC services, the EV should fall back the CP charging.

kramzarales, 2022-12-09:
hi,

This is my theory as well. This is my current config that works:
 self.evse_config = {
            "evse_id_DIN": '+49*123*456*789',
            "evse_id_ISO": 'DE*A23*E45B*78C',
            "protocol": [0, 1], 
            "payment_method": [0],
            "energy_transfer_mode": [1],
            "certificate_installation_support": False,
            "certificate_update_support": False,
        }

When it is in use it prints out the evse_id_ISO  , so i assume that it is as you said and it skips straight to using ISO15118-2 protocol.
If the solution is to first trick the EV into going into dc_extended mode to get the parameters and then terminating SLAC and just setting CP to the desired value, that works perfectly fine for us.

But i have another questing regarding this method. Would it also be possible to obtain the voltage and charging current data from the vehicle in the same way?

The command "Get DC Charging Parameters" (sub id 0x64) can be used to extract the current voltage and current as measured on the vehicle. My question is, is this the voltage and current that is measured on the lower DC charging pins, or is this voltage measured on the main AC pins of the connector?

There is also the command "Set AC Charging Parameters" (0x65). That talks about **the EV might easily
calculate the voltage between phases** Could that perhaps be used to get this reading from the vehicle?

jpo-stx, 2022-12-20:
> But i have another questing regarding this method. Would it also be possible to obtain the voltage and charging current data from the vehicle in the same way?

In AC charging the actual charging current is not sent by the EV to charging station.

> The command "Get DC Charging Parameters" (sub id 0x64) can be used to extract the current voltage and current as measured on the vehicle. My question is, is this the voltage and current that is measured on the lower DC charging pins, or is this voltage measured on the main AC pins of the connector?

This command fetches your own chargning parameters that were configured previously.

Present Voltage / Current: Measured levels by EVSE
Target Voltage / Current: Requested levels by EV

> There is also the command "Set AC Charging Parameters" (0x65). That talks about **the EV might easily calculate the voltage between phases** Could that perhaps be used to get this reading from the vehicle?

This command is used to set the EVSE parameters, these will be sent to the EV during the charging session-

nststx, 2023-04-19:
Closing due to inactivity. If the issue still persist or you have questions left, feel free to reopen.

closed 2023-04-19 by nststx