Sevenstax/FreeV2G #122: Sequence error

codderdreamer, 2022-12-12:
[sequence_error.zip](https://github.com/Sevenstax/FreeV2G/files/10207610/sequence_error.zip)

Hi, 

We are getting a sequence error during charging in the 2022 zoe vehicle. Why don't we get this in other zoe vehicles? In addition, we take it when the charge percentage is low, but not when it is high.

jpo-stx, 2022-12-13:
Can you change your setup so that you receive the mirrored plc frames from the whitebeet? You need to be directly connected to the Whitebeet or you need a smart switch with port mirroring anabled.

In the log I see that the EV requests a target voltage of 412V (0x1018ff) and you return as present voltage 364V (0x016c00) maybe this is not enough and the EV?

lho-stx, 2023-02-01:
Closing this issue due to inactivity. If the issue still persists, please re-open!

closed 2023-02-01 by lho-stx