Sevenstax/FreeV2G #125: No status messages from EV or EVSE Whitebeet Carrier Boards

Bobbybigears, 2022-12-14:
Hi,
I'm developing a prototype system using two Whitebeet Carrier Boards. One for the EV side of things (C-based SPI) and one for the EVSE side of things (python reference app). I'm able to get to a point where I can start a session, i.e. Control Pilot has been setup, SLAC matching performed, V2G mode (EV), configuration and dc parameters set (based on the manual example) and the V2G service has been started. All the commands are accepted, i.e. the acknowledge bytes are all 0. When I start a session (command is accepted), I only get a 'Session Stopped' status message (subid: 0xcb) when the 30s timeout occurs at the python app. This timeout is linked to the 'v2gEvReceiveRequest' function in 'Whitebeet.py' and called from 'Ev.py'. If I extend the timeout period, I get a 'Session Stopped' message at the EVSE end, i.e. printed from the python app. On a related note, there doesn't seem to be a description of the V2G start service command in the user manual. Does the service need a delay between issuing the start command and the 'Start Session' command?  Additionally, there are no framing or sync errors. I would be most grateful for any help you can provide me with regard to this issue .Best regards Rob.

Bobbybigears, 2022-12-14:
Apologies, the timeout is linked to v2gEvseReceiveRequest() in Evse.py.

Bobbybigears, 2022-12-14:
Under what circumstance(s) would a Whitebeet Carrier Board (EV) not generate a 'Session Started (0x27/0x80' status message for the EVSE having received and accepted valid 'V2G Set Mode', 'V2G Set Configuration', 'V2G Set DC Charging Parameters', 'V2G Start Service'& 'V2G Start Session' commands? Cheers Rob.

lho-stx, 2022-12-23:
Hey @Bobbybigears,

First I have some questions on your issue:

1. You are using whitebeet EV (own implementation) and whitebeet EVSE (FreeV2G)?

2. Can you please share some logs? Idealy you enable the port mirror feature on both of your boards and create a wireshark log of the communication. Also please attach the UART out of the whitebeets and the terminal output of the FreeV2G application (and maybe of your custom implementation).

I try to sort out all of the questions:

> Does the service need a delay between issuing the start command and the 'Start Session' command?

Before a V2G session can be started the two parties need to join the same HomePlugGP network. To join the same network there needs some control pilot handling and SLAC. This will introduce enough delay that there should be no problem.

>  This timeout is linked to the 'v2gEvseReceiveRequest' function in 'Whitebeet.py' and called from 'Evse.py'

When no frame sync error occurs, the whitebeet may have seen a timeout in the V2G session.

> Under what circumstance(s) would a Whitebeet Carrier Board (EV) not generate a 'Session Started (0x27/0x80' status message for the EVSE having received and accepted valid 'V2G Set Mode', 'V2G Set Configuration', 'V2G Set DC Charging Parameters', 'V2G Start Service'& 'V2G Start Session' commands?

There can be multiple reasons, like a mismatched EV and EVSE configuration.

I am sure we can work this out quickly when you send the requested logs. The wireshark logs are the most important ones!

Please allow longer response time because of holiday season! 

Thank you, best regards and happy holidays,
lho

Bobbybigears, 2023-01-23:
HI. Apologies for not getting back sooner. It was mismatched EV/EVSE configuration. Cheers Rob

lho-stx, 2023-01-24:
Hey @Bobbybigears,

ok then I will close this issue as solved!

closed 2023-01-24 by lho-stx