Sevenstax/FreeV2G #126: Control Pilot State (F) Being Returned for Mode EV

Bobbybigears, 2022-12-15:
In version 2.14 of the User Manual for the Control Pilot 'GetState' Command - Section 15.4.9, it states that in EV mode, the CP circuit can only detect States B to E due to the way it's designed.

I've setup the Control Pilot for EV mode and am getting State F (5/-12V) following calling this command (I've attached a screenshot for your perusal). 

This is the information for the Control Pilot prior to EVSE Connection and following execution of the 'Get Mode', 'Get Duty Cycle, 'Get Resistor Level' and 'Get State' commands.

I'm using C-based firmware using SPI as the host. The Whitebeet Carrier Board details are: 'WB-CARRIER-BOARD-P1 1.1'. PCB: PWL_700_03.

Many thanks in advance.
Rob

[image]

jpo-stx, 2022-12-20:
Hi @Bobbybigears ,
1. Are you using the latest firmware 1.0.6 and is the state changing at all?
2. Have you started the CP service?

We had a similar issue in the past (https://github.com/Sevenstax/FreeV2G/issues/77) where the state did not change, this was fixed in 1.0.6

lho-stx, 2023-02-01:
Closing this issue due to inactivity. If the issue still persists, please re-open!

closed 2023-02-01 by lho-stx

Bobbybigears, 2023-02-01:
Apologies. It was an old version of the firmware and repository.
Many thanks Rob.

-- 
Rob Davies
(Software Engineer)

Treharne Automotive Engineering Ltd
Treharne Workshops
Llanelli Gate
Dafen
Llanelli
Carmarthenshire
SA14 8LQ

Tel: 01554 775938