Sevenstax/FreeV2G #127: Malformed DC Charge Parameters Changed status message (0xC1)

dtomasi95, 2022-12-19:
When receiving the DC EVESE Charge Parameter from the EVESE, the WHITE-BEET (in EV mode) sends the DC Charge Parameters Changed status message (0xC1).
But the message consistently reports two time the EVESE Maximum Voltage, and it doesn’t report the EVESE Maximum Power Limit (see picture 1 below).
Is there a way to fix that?
Also, the number of bytes of the payload seems odd:
as you can see from my breakdown of the payload byte (picture 2), I cat find a way to divide them according to your documentation (picture 3) without ending with spare bytes that don't fit in the remaining fields.
If I am doing something wrong, can you provide me with an example?
I also attach a picture of the WHITE-BEET box for FW version and other info that may be useful for you.

[image]
[image]
[image]
[image]

jpo-stx, 2022-12-20:
Hi @dtomasi95 ,

you are right with the voltage being set a second time instead of the power. We need to fix that.

For the message you missed that the isolation status is optional so it is actually two bytes. First byte telling whether or not it is present and if it is it is followed by the second byte telling the value.
In your case after the status byte there is a `01` which tells you that the optional isolation status is present and the following byte `00` tells you that the status is "Invalid". The last two `00` bytes at the end of the message tell you that the last two optional parameters (Current Regualtion Tolerance, Energy to be Delivered) are not present

dtomasi95, 2022-12-20:
Hi @jpo-stx ,
thanks for your reply.
The breakdown now makes sense, so if the optional isolation status is not present, I will have the leading byte set to 00 and no following byte?
I was confused by the documentation example where the isolation status is just missing without a presence byte.
For the repeated voltage, can you tell me when a fix will be available?

Thank you,
best regards

[image]

jpo-stx, 2022-12-20:
Ok, the isolation status is missing completely in the example, need to fix that too.

There should be a new version available in Q1 next year, but I do not know the exact date.

dtomasi95, 2022-12-23:
hi @jpo-stx ,
i also encountered another error in these days.
When the charging is over, and I enter the post charging the EVSE don't send the EVSEPresentCurrent so the WHITE-BEET value for present current should be 0 instead is set to: 27.637000. (see picture below)

Instead, regarding the firmware update:  in our use case the  WHITE-BEET will be soldered directly tho the PCB of an arm board running Linux that will act as the host controller thought Ethernet, and we will have multiple units.
So if a WHITE-BEET firmware update will be necessary, we will need to update all the WHITE-BEET efficiently.
The documentation refer to a windows tool (that we can't use) and a Python tool.
Is the Python tool fully working and compatible with our use case?
Can you please provide with a link to the Python update tool?
Also, if in a later release of our software we would want to implement it,  the WHITE-BEET firmware update module supports also signed updates right?

thanks
kind regards 

[image]

jpo-stx, 2023-01-02:
> hi @jpo-stx , i also encountered another error in these days. When the charging is over, and I enter the post charging the EVSE don't send the EVSEPresentCurrent so the WHITE-BEET value for present current should be 0 instead is set to: 27.637000. (see picture below)

The current is only transmitted in the charging loop (CurrentDemandRes) so we do not really know the real value. Therefore, we decided to add a comment in the user manual:

[image]

It states that the present current value is only valid until "Charging Stopped" status message. However, it makes sense to set the value to zero when charging was stopped to reduce confusion on user side.

> Instead, regarding the firmware update: in our use case the WHITE-BEET will be soldered directly tho the PCB of an arm board running Linux that will act as the host controller thought Ethernet, and [...]

The python tool is fully working and should be compatible with your use case. The executable was generated from the python source code. Signed updates are supported.

lho-stx, 2023-02-01:
Closing this issue due to inactivity. If the issue still persists, please re-open!

dtomasi95, 2023-05-17:
issue still persistent

dtomasi95, 2023-05-17:
the issues mentioned in the first part of this thread to my knowledge are still unsolved please re open the issue

nststx, 2023-06-09:
Hi @dtomasi95,

The issue with the maximum voltage being reported twice instead of maximum power limit is indeed still unsolved. However we already have it in our internal tracker and will fix it with the next EV firmware release in Q3 2023.

Best regards,
Niklas Stoffers