Sevenstax/FreeV2G #128: Frame reception timed out error with codico EV

sabarish-sema, 2022-12-20:
I'm facing below issue on Codico EV Board while connecting with EVSE.
NOTE: We are using Linux environment for running application

Welcome to Codico Whitebeet EV reference implementation
Initiating framing interface
iface: ETH, name: enp4s0, mac: C4:93:00:34:D0:04
Traceback (most recent call last):
  File "/mnt/Projects/ev_codico/FreeV2G/Whitebeet.py", line 168, in _sendReceive
    response = self.framing.receive_next_frame(filter_mod=[mod_id, 0xFF], filter_req_id=req_id, timeout=time_end - time_start)
  File "/mnt/Projects/ev_codico/FreeV2G/FramingInterface.py", line 240, in receive_next_frame
    raise AssertionError("Frame reception timed out")
AssertionError: Frame reception timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/Projects/ev_codico/FreeV2G/Whitebeet.py", line 112, in __init__
    self.version = self.systemGetVersion()
  File "/mnt/Projects/ev_codico/FreeV2G/Whitebeet.py", line 306, in systemGetVersion
    response = self._sendReceiveAck(self.sys_mod_id, self.sys_sub_get_firmware_version, None)
  File "/mnt/Projects/ev_codico/FreeV2G/Whitebeet.py", line 189, in _sendReceiveAck
    response = self._sendReceive(mod_id, sub_id, payload)
  File "/mnt/Projects/ev_codico/FreeV2G/Whitebeet.py", line 183, in _sendReceive
    raise ConnectionError("Problem with send/receive - Please check your connection!")
ConnectionError: Problem with send/receive - Please check your connection!

During handling of the above exception, another exception occurred

Please let us know if more information is required

Thanks
Sai Sabarish

jpo-stx, 2022-12-20:
Can you send me a log of the Whitebeet UART?

sabarish-sema, 2022-12-20:
please find the Whitebeet Uart logs below

APPLICATION:
  Name:      ISO15118 (EV)
  Version:   V01_00_04
  Target:    WHITE-BEET
  CPU:       STM32F745

PRODUCT VERSION:
String: SEVENSTAX_LIB_V_10_02_01
Hex:    0x000a0201
Dec:    100201

Hostcontroller Interface: ETHERNET!

APPLICATION RUNNING ...
[    0.004] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from host
[    0.005] [NOTICE] APL-MAIN: BOOT-CFG: FW2
[    0.006] [NOTICE] APL-MAIN: BOOT-CFG: Module configured to EV
[    0.008] [NOTICE] Q7KCOM:  Boot from host selected
[    0.011] [NOTICE] SLAC: stxSLAC_Init()
[    0.013] [NOTICE] APL-ETH:   MAC of NIC[0]:    c4:93:00:34:d0:03
[    3.638] [NOTICE] APL-ETH:   MAC of NIC[1]:    c4:93:00:34:d0:02
[    3.644] [NOTICE] V2GCPS: stxV2GCPS_RegisterCallback()
[    3.649] [NOTICE] APL-ETH: Got link:
[    3.650] [NOTICE] APL-ETH:     Interface: eth1
[    3.650] [NOTICE] APL-ETH:     MAC:       C4:93:00:34:D0:02
[    3.855] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    3.856] [NOTICE] Q7K: =========================================
[    3.857] [NOTICE] Q7K:   QCA700x is READY
[    3.857] [NOTICE] Q7K: =========================================
[    3.858] [NOTICE] Q7KCOM: Reset QCA700x
[    3.860] [NOTICE] Q7K: =========================================
[    3.860] [NOTICE] Q7K:   CPU ON RECEIVED
[    3.861] [NOTICE] Q7K: =========================================
[    3.862] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    3.863] [NOTICE] Q7K: =========================================
[    3.864] [NOTICE] Q7K:   QCA700x is READY
[    3.864] [NOTICE] Q7K: =========================================
[    4.730] [NOTICE] Q7KCOM:  ======================================= 
[    4.731] [NOTICE] Q7KCOM:    FW and PIB was loaded successful 
[    5.034] [NOTICE] APL-MAIN: 
[    5.034] [NOTICE] APL-MAIN: =============================
[    5.035] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6): 
[    5.035] [NOTICE] APL-MAIN: for IP Config handle [fe01] 
[    5.036] [NOTICE] APL-MAIN: IPv6 Addresses:
[    5.037] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe34:d002
[    5.038] [NOTICE] APL-MAIN:   Local:    0:0:0:0 : 0:0:0:0
[    5.038] [NOTICE] APL-MAIN:   DNS1:    2001:4860:4860:0 : 0:0:0:8888
[    5.039] [NOTICE] APL-MAIN:   DNS2:    fd00:0:0:0 : be05:43ff:fe44:5280
[    5.040] [NOTICE] APL-MAIN: =============================
[    5.041] [NOTICE] APL-MAIN: 
[    5.245] [NOTICE] Q7K: =========================================
[    5.246] [NOTICE] Q7K:   CPU ON RECEIVED
[    5.247] [NOTICE] Q7K: =========================================
[    5.248] [NOTICE] Q7KCOM:  ======================================= 
[    5.248] [NOTICE] Q7KCOM:    BOOT FROM HOST READY 
[    5.249] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    5.250] [NOTICE] Q7K: =========================================
[    5.251] [NOTICE] Q7K:   QCA700x is READY
[    5.251] [NOTICE] Q7K: =========================================
[    5.275] [NOTICE] Q7KCOM:  Own QCA7K MAC = C4:93:00:34:D0:04
[   13.042] [WARN  ] APL-ETH: # Unhandled Ethernet notifycode 365

jpo-stx, 2022-12-20:
Try with MAC C4:93:00:34:D0:02

C4:93:00:34:D0:04 ist the MAC of the ethernet interface, See https://github.com/Sevenstax/FreeV2G/issues/96

closed 2022-12-20 by jpo-stx

tareksr4, 2023-03-18:
How to make a log of whitebeet uart?