Sevenstax/FreeV2G #130: Session Stopped after EVSE finished charge parameter discovery

ChargerH55, 2022-12-21:
**Setup**
Platform: EVSE
Firmware Version: V01_01_07 

Host Controller Interface: Ethernet 
Host: FreeV2G/tree/EVSE_v1.1.7_1

**Describe the bug**
I am trying to use the EVSE mode of FreeV2g with a VC-VCCU as an EV, but the application stop just after "Request Discovery Charge Parameters" finished. I tried to force the Charge Transfer Mode to 1: DC_Extended but nothing changed. 
Do you know what could cause this "Session Stopped" ? Is it coming from the EVSE or the EV ?

Please find attached the logs of the UART port of the Whitebeet, the wireshark log of the CAN port between EVSE and EV and the wireshark log of the ethernet port.

**Logs**

admsbo@desk-dev-01:~/Documents/vcevcc/FreeV2G-EVSE_v1.1.7_0$ sudo python3 Application.py -m c4:93:00:21:50:8B -r EVSE enx0c3796785458
[sudo] password for admsbo: 
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: enx0c3796785458, mac: c4:93:00:21:50:8B
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects
EV connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
Start V2G
"Session started" received
Protocol: 2
Session ID: df9b9a0ab23313db
EVCC ID: 00168105889e
"Request EVSE ID" received
Set EVSE ID: DE*ABC*E*00001*01
"Request Authorization" received
Authorize the vehicle? Type "yes" or "no" in the next 59s: yes
Vehicle was authorized by user!
"Request Schedules" received
Max entries: 1
Set the schedule: [{'id': 1, 'tuple': [{'start': 0, 'interval': 1800, 'max_power': 25000}, {'start': 1800, 'interval': 1800, 'max_power': 18750}, {'start': 3600, 'interval': 82800, 'max_power': 12500}]}]
"Request Discovery Charge Parameters" received
EV maximum current: 500A
EV maximum power: 200000W
EV maximum voltage: 4000V
Full SOC: 50%
Bulk SOC: 50%
SOC: 5%
"Session stopped" received
EVSE loop finished
Goodbye!

[image]
[log_ethernet.zip](h
[logs_EVSE.zip](https://github.com/Sevenstax/FreeV2G/files/10276553/logs_EVSE.zip)
ttps://github.com/Sevenstax/FreeV2G/files/10276535/log_ethernet.zip)

Thank you in advance for your help.

lho-stx, 2022-12-23:
Hey @CB-ctrl,

First, you are using an old firmware version. Please update to the newest firmware if possible.

Relating your issue:

In your log_ethernet.pcapng packet 83 the charge parameters of the EVSE are set. The values are:

Max. Current: 100A
Min. Current: 0A
Max. Voltage: 400V
Min. Voltage: 0V
Max. Power: 25 kW

In the first log there is:

> EV maximum current: 500A
> EV maximum power: 200000W
> EV maximum voltage: 4000V

These values are not matching with the max. EVSE parameters. If the max. values are exceeded by the EV the whitebeet will close the session. Please try to match the EV values with the EVSE values.

If the issue still occurs: Please enable the port mirror feature in the whitebeet (in Application.py put evse.whitebeet.networkConfigSetPortMirrorState(1) right before evse.loop() ) and create a new log_etherne.tpcapng. Then I can investigate the V2G session messages for any other error.

Best regards and happy holidays,
lho

CB-ctrl, 2023-01-09:
Hello @lho-stx,

Thank you for your answer, may I ask you what is the newest firmware version that I need to upload ? I am only seeing 
[image]  in the PLC documentation area.

I changed the values of max parameters to match the EV values with the EVSE values but I am still having the "Session Stopped".

I will send a new log_ethernet.tpcapng soon.

Best regards and happy new year,
CB

lho-stx, 2023-01-09:
Hey @CB-ctrl,

the newest firmware is V02_00_01. Please try this FW with the matching github tag.

I am looking forward to your logs!

Thank you and best regards,
lho

CB-ctrl, 2023-01-09:
Hello @lho-stx,

Here is what I have now with the newest firmware version.

(.venv) admsbo@desk-dev-01:~/Documents/vcevcc/FreeV2G$ sudo .venv/bin/python3 Application.py eth -i enx0c3796785458 -m c4:93:00:21:50:8B -r EVSE
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: ETH, name: enx0c3796785458, mac: c4:93:00:21:50:8B
WHITE-beet-EI firmware version: V02_00_01
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects
EV connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
Start V2G
"Session started" received
Protocol: 1
Session ID: ef2d61c6434a671f
EVCC ID: 00168105889e
"Payment selcted" received
Selected payment method: 0
"Request Authorization" received
54999
Authorize the vehicle? Type "yes" or "no" in the next 53s: yes
Vehicle was authorized by user!
"Energy transfer mode selected" received
Energy request: 5000
Maximum voltage: 400
Maximum current: 100
Maximum power: 20000
Energy Capacity: 5000
Full SoC: 100
Bulk SoC: 50
Ready: yes
Error code: 1
SoC: 50
Selected energy transfer mode: 1
"Request Schedules" received
Max entries: 1
Set the schedule: {'code': 0, 'schedule_tuples': [{'schedule_tuple_id': 1, 'schedules': [{'start': 0, 'interval': 1800, 'power': 25000}, {'start': 1800, 'interval': 1800, 'power': 18750}, {'start': 3600, 'interval': 82800, 'power': 12500}]}]}
"DC Charge Parameters Changed" received
EV maximum current: 100A
EV maximum voltage: 400V
EV maximum power: 20000W
EV ready: True
Error code: 1
SOC: 50%
EV target voltage: 0V
EV target current: 0A
Charging complete: 0
"Session stopped" received
Closure type: 0
EVSE loop finished
Goodbye!

I don't understand why the target current and voltage are 0, I am sending to the VCVCCU : 
    # Target Voltage
    VCVCCU_V2G_EVTargetVoltage = db.get_message_by_name('VCVCCU_V2G_EVTargetVoltage')

    target_voltage = VCVCCU_V2G_EVTargetVoltage.encode(
        {'VCVCCU_V2G_EVTargetVoltage_Value': 170, 'VCVCCU_V2G_EVTargetVoltage_Multiplier': 1,
         'VCVCCU_V2G_EVTargetVoltage_UnitFlag': 1, 'VCVCCU_V2G_EVTargetVoltage_Unit': 5})

    message_Targvolt = can.Message(arbitration_id=VCVCCU_V2G_EVTargetVoltage.frame_id, data=target_voltage,
                                   is_extended_id=True)
    can_bus.send(message_Targvolt)

    # Target Current
    VCVCCU_V2G_EVTargetCurrent = db.get_message_by_name('VCVCCU_V2G_EVTargetCurrent')

    target_current = VCVCCU_V2G_EVTargetCurrent.encode(
        {'VCVCCU_V2G_EVTargetCurrent_Value': 40, 'VCVCCU_V2G_EVTargetCurrent_Multiplier': 1,
         'VCVCCU_V2G_EVTargetCurrent_UnitFlag': 1, 'VCVCCU_V2G_EVTargetCurrent_Unit': 3})

    message_Targcurr = can.Message(arbitration_id=VCVCCU_V2G_EVTargetCurrent.frame_id, data=target_current,
                                   is_extended_id=True)
    can_bus.send(message_Targcurr)

Please find attached the log of the ethernet port:
[log_ethernet.zip](https://github.com/Sevenstax/FreeV2G/files/10373372/log_ethernet.zip)

lho-stx, 2023-01-09:
Hey @CB-ctrl,

Thank you for your logs!

In frame 203 the error code is set to errorCode: RESS temperature inhibit. Please have a look if your battery temperature is out of range.

With best regards,
lho

CB-ctrl, 2023-01-10:
Hello @lho-stx,

I changed some parameters for the battery temperature but I am still facing the same session stopped... Here is the new log:
[log_ethernet.zip](https://github.com/Sevenstax/FreeV2G/files/10381612/log_ethernet.zip)

Thank you in advance for your help.

Kind regards, 

CB

lho-stx, 2023-01-11:
Hey @CB-ctrl,

thank you for the log. I had a quick glance in it and on first sight the first problem (RESS temperature inhibit) is gone. I need some more time to look into this. I will come back to you as soon as possible.

Best regards,
lho

CB-ctrl, 2023-01-18:
Hello @lho-stx,

Do you have any insight regarding the logs ? 

Thanks a lot for your help again.

Best regards, 

CB

lho-stx, 2023-01-20:
Hey @CB-ctrl,

to answer your question regarding the target current and target voltage beeing 0 in the FreeV2G output: This is ok, because the Whitebeet sends a DCChargeParametersChanged messagre to the host application, because of the ChargeParamDiscoveryReq message send from your EV to the Whitebeet EVSE. There is no target voltage and current in this message, because this information is shared by EV in the CurrentDemandReq. This message is send only while the vehicle is charging. So FreeV2G just shows 0 for target voltage and target current, this may lead to confusion I will discuss this internally.

The problem in your charging session is that the EV is doing a TCP reset in frame 241, right after it received the ChargeParamDiscoveryReq. This TCP reset leads to the end of the charging session. On first glance the session parameters look good and match the Whitebeet configuration. 

Can you send me the output of UART and FreeV2G for the latest charging session?

Thank you and best regards,
lho

CB-ctrl, 2023-01-20:
Hello @lho-stx,
Thank you for your answer, please find the log of all port : 
[log_any.zip](https://github.com/Sevenstax/FreeV2G/files/10464123/log_any.zip)

Here is what I see from the Application.py terminal:

(.venv) admsbo@desk-dev-01:~/Documents/vcevcc/FreeV2G$ sudo .venv/bin/python3 Application.py -i enx0c3796785458 -m c4:93:00:21:50:8B -r EVSE eth
[sudo] password for admsbo: 
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: ETH, name: enx0c3796785458, mac: c4:93:00:21:50:8B
WHITE-beet-EI firmware version: V02_00_01
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects
EV connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
Start V2G
"Session started" received
Protocol: 1
Session ID: b72bc859f1abdf1a
EVCC ID: 00168105889e
"Payment selcted" received
Selected payment method: 0
"Request Authorization" received
55000
Authorize the vehicle? Type "yes" or "no" in the next 54s: yes
Vehicle was authorized by user!
"Energy transfer mode selected" received
Energy request: 100
Maximum voltage: 400
Maximum current: 100
Maximum power: 25000
Energy Capacity: 100
Ready: yes
Error code: 0
SoC: 22
Selected energy transfer mode: 1
"Request Schedules" received
Max entries: 1
Set the schedule: {'code': 0, 'schedule_tuples': [{'schedule_tuple_id': 1, 'schedules': [{'start': 0, 'interval': 1800, 'power': 25000}, {'start': 1800, 'interval': 1800, 'power': 18750}, {'start': 3600, 'interval': 82800, 'power': 12500}]}]}
"DC Charge Parameters Changed" received
EV maximum current: 100A
EV maximum voltage: 400V
EV maximum power: 25000W
EV ready: True
Error code: 0
SOC: 22%
EV target voltage: 0V
EV target current: 0A
Charging complete: 0
"Session stopped" received
Closure type: 0
EVSE loop finished
Goodbye!

Thank you again for your help. 

Best regards, 

CB

lho-stx, 2023-01-24:
Hey @CB-ctrl,

the session failes, because there is a bug in the EVSE Whitebbet firmware. The EVSE sends the duration of the schedule entries in the ChargeParamDiscoveryRes, but the duration shall not be send when the entry is not the last entry in the list.

This will be fixed with the new EVSE whitebeet firmware.

Thank you for your patience,
lho

closed 2023-01-24 by lho-stx

nikunjp26, 2023-04-19:
Dear IHo,

We are facing the same issue now so is there any conclusion on that? or can you please share us the experimental firmware in which you solved the issue beacuse not able to charge the Mahindra XUV400 EV car.

Best Regards,
Nikunj Patel