Sevenstax/FreeV2G #131: Not possible to select second schedule tuple

Malte9, 2022-12-21:
**Your Setup**
Platform: EVSE
Firmware Version: V02_00_01
Host Controller Interface: Ethernet
Host: FreeV2G EVSE_v2.0.1_4 /Main

**Describe the bug**
When the second tuple is selected the Tariff selection failed error is transmitted in the PowerDeliveryRes and the PowerDeliveryRes changes the charging context to DC from AC

**To Reproduce**
Create second schedule tuple by copying first one, start charging session and select second tuple

**Expected behavior**
A second schedule should be selectable by the EV

**Logs**
Python output:
[image]

From EVSE recieved CurrentDemandRes
[image]

PowerDeliveryReq for second schedule
[image]

**Additional context**
Testing against own EV implementation, when selecting the first schedule the charging process starts, only when selecting the second schedule the error is returned

lho-stx, 2022-12-23:
Hey @Malte9,

Can you do me a favor: Please enable the port mirror feature in the whitebeet (in Application.py put evse.whitebeet.networkConfigSetPortMirrorState(1) right before evse.loop() ) and create a wireshark log. Then I can investigate the V2G session messages for any error.

Please note: due to holiday season our response time may be longer than usual.

With best regards,
lho

jpo-stx, 2023-01-02:
This is a bug in the firmware. For validation in the PowerDeliveryReq the Whitebeet needs to save the IDs of the tuples. In case multiple tuples are present the ID of the first tuple is saved multiple times.

We need to fix this!

nststx, 2023-04-19:
Closing due to new firmware release which should fix this issue. If the issue still persists, feel free to reopen.

closed 2023-04-19 by nststx