Sevenstax/FreeV2G #132: wrong session Id

sabarish-sema, 2022-12-21:
We connected Codico EV with our EVSE. we are getting wrong Session Id error as below shown in logs
NOTE: we are setting session id as [01 02 03 04 05 06 07 08] in EVSE side as mentioned in the document.

Terminal Logs:

Welcome to Codico Whitebeet EV reference implementation
Initiating framing interface
iface: ETH, name: enp4s0, mac: c4:93:00:34:d0:02
WHITE-beet-PI firmware version: V01_00_04
Set the CP mode to EV
Start the CP service
Set the CP state to State B
Set SLAC to EV mode
Start SLAC
EVSE connected
Start SLAC matching
SLAC matching successful
Set V2G mode to EV
Set V2G configuration
Set DC charging parameters
Create new charging session
ControlPilot duty cycle: 5.6
"Session Stopped" received
Change State to State B
EV loop finished
Goodbye!

Serial Logs:

[ 2331.194] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6): 
[ 2331.194] [NOTICE] APL-MAIN: for IP Config handle [fe00] 
[ 2331.195] [NOTICE] APL-MAIN: IPv6 Addresses:
[ 2331.196] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe34:d003
[ 2331.197] [NOTICE] APL-MAIN:   Local:    0:0:0:0 : 0:0:0:0
[ 2331.197] [NOTICE] APL-MAIN:   DNS1:    2001:4860:4860:0 : 0:0:0:8888
[ 2331.198] [NOTICE] APL-MAIN:   DNS2:    fd00:0:0:0 : be05:43ff:fe44:5280
[ 2331.199] [NOTICE] APL-MAIN: =============================
[ 2331.200] [NOTICE] APL-MAIN: 
[ 2331.539] [WARN  ] V2GSVC: Security options satisfied, connect!
[ 2331.549] [NOTICE] V2GSVC: TCP: Connected to server without TLS
[ 2332.092] [ERROR ] V2GISO13: Wrong session ID from EVSE!
[ 2332.124] [NOTICE] V2GSVC: TCP: We were disconnected from the server
[ 2332.125] [NOTICE] V2GAPP: Received event 'Session stopped'
[ 2332.128] [NOTICE] V2GCPS: Changed Max Current EV State (7)
[ 2332.128] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[ 2332.129] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[ 2332.130] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[ 2332.130] [NOTICE] V2GCPS: Duty cycle is 1000 permill
[ 2332.179] [NOTICE] V2GCPS: Changed Max Current EV State (1)
[ 2332.179] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[ 2332.180] [NOTICE] V2GCPS: stxXBAR_V2GCPS_NotifyHandler()
[ 2332.181] [NOTICE] V2GCPS: # V2GCPS_INFOTYPE_CP_DUTY_CYCLE
[ 2332.181] [NOTICE] V2GCPS: Duty cycle is 56 permill
[ 2333.135] [NOTICE] SLAC: stxSLAC_Stop()
[ 2339.201] [WARN  ] APL-ETH: # Unhandled Ethernet notifycode 365

lho-stx, 2022-12-23:
Hey @sabarish-sema,

you are using an outdated firmware version, please update your firmware if possible, checkout the corresponding tag and try again.

If the error still occurs: Please enable the port mirror feature in the whitebeet (in Application.py put ev.whitebeet.networkConfigSetPortMirrorState(1) right before ev.loop() ) and create a wireshark log. Then I can investigate the V2G session messages for any error.

Thank you and best regards,
lho

sabarish-sema, 2023-01-03:
Hi,
while updating the firmware we are getting error as below.
We are using Linux machine for updating firmware.

Firmware version - WHITEBEET_ISO15118_EVSE_V01_01_06.fwu
python version - 3.9.16

Welcome the SEVENSTAX SFWU Updater
Initiating framing interface
Starting update process
Firmware update at 0 percent; Sent: 833 Bytes
[!] Error: Device rejected update file

please let us know if more information is required.
Thanks

sabarish-sema, 2023-01-03:
Hi,
In PLC documentation area, we are not able to find firmware for EV. We only found firmware for EVSE. 
Can you please let us know where we can find EV firmware.

Thanks

lho-stx, 2023-01-03:
Hey,

can you please confirm that you are using a Whitebeet PEV? If your are using Whitebeet PEV you may got the wrong firmware files from Codico.

Best regards,
lho

sabarish-sema, 2023-01-03:
Hi,
We are using Whitebeet carrier board EV ISO15118.

sabarish-sema, 2023-01-03:
How can we download Whitebeet EV board latest firmware from CODICO PLC documentation area.

lho-stx, 2023-01-03:
Hey,

Then you may got access to the wrong firmware files by Codico. Please contact your Codico sales representative and ask for the PEV firmware files. You can refer to this discussion.

If there are any problems, please let me know!

I apologize for the inconvenience,
lho

lho-stx, 2023-01-11:
Hey @sabarish-sema,

did you find a solution with Codico Customer Service? If yes, did this solve your issue? If not is there anything I can do for you? If not I like to close this issue.

Best regards,
lho

sabarish-sema, 2023-01-11:
Hi,
We got latest firmware from customer service.And updated the same firmware(V0_01_06) in EV Board.
But still we are facing same issue(Wrong session ID from EVSE!).

PFA wireshark logs and save as txt.As we are not able to attach as pcapng format.
[Codico_Ev2.txt](https://github.com/Sevenstax/FreeV2G/files/10390448/Codico_Ev2.txt)

lho-stx, 2023-01-11:
Hey,

thank you, I have no problem reading the Log. I will come back to you as soon as I have news!

lho-stx, 2023-01-24:
Hey @sabarish-sema,

It seems that the problem is the ServiceDetailRequest. Maybe your EVSE has not implemented the Message and therefore fails with a SequenceError and sets the session id to just 0s. 

I double check but I think there is no workaround in the moment.

Best regards,
lho

sabarish-sema, 2023-01-27:
hi,
I see error as EVSE Id received wrong.which we are sending in session setup response.So,my understanding is that it is failing before going to ServiceDetailRequest.

[image]

Thanks!
Sai Sabarish

lho-stx, 2023-01-27:
Hey @sabarish-sema,

I had a look into this log file you've send me:

> PFA wireshark logs and save as txt.As we are not able to attach as pcapng format.
> [Codico_Ev2.txt](https://github.com/Sevenstax/FreeV2G/files/10390448/Codico_Ev2.txt)

There you can see the SessionSetupResponse has a valid SessionID. please see the attached image:

[image]

In the top the wireshark window SessionSetupResponse is shown, in the bottom window ServiceDetailResponse is shown. Your EVSE is stopping the session because the processing of the ServiceDetailRequest has failed.

I checked internally and unfortunately I can not provide you a workaround for this issue at the moment (there may be the option to not send ServiceDetailRequest in future versions of the Whitebeet firmware).

Best regards,
lho

lho-stx, 2023-01-31:
Hey @sabarish-sema,

did you try DIN with your EVSE? Please let me know if this solved your issue.

Thank you,
lho

lho-stx, 2023-02-13:
Closing this issue. Please re-open if needed.

closed 2023-02-13 by lho-stx