Sevenstax/FreeV2G #133: EVSE, no more messages after Start V2G

nigaru05, 2023-01-03:
Hi, I'm new to the group and I've just started working with FreeV2G.
I have purchased a WB-CARRIER-BOARD EI - EVSE. Update the firmware to version V02_00_01, clone the git repository and change to TAG EVSE_v2.0.1_4.
I run Application.py and after the "Start V2G" message, it disconnects due to timeout, any ideas???

----------------------------------------------------------------------------------------------------
sudo .venv/bin/python Application.py eth -i enx70886b8d484e -m C4:93:00:34:AB:15 -r EVSE
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: ETH, name: enx70886b8d484e, mac: C4:93:00:34:AB:15
WHITE-beet-EI firmware version: V02_00_01
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
EV already connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
Start V2G
We did not receive a frame before the timeout of 30s
EVSE loop finished
Goodbye!
----------------------------------------------------------------------------------------------------

I also attach a capture made with wireshark between the notebook and the WB-CARRIER-BOARD

Thank you very much in advance.

[head_1_4.zip](https://github.com/Sevenstax/FreeV2G/files/10338846/head_1_4.zip)

lho-stx, 2023-01-04:
Hey @nigaru05,

I had a look into your log file. There's something strange happening:

In frame 438 FreeV2G host application sets Whitebeets V2G module to "mode EVSE", this is confirmed in the next frame. The setup of the EVSE parameter is also successfull.

In frame 458 FreeV2G queries the V2G module state of the Whiteneet. The Whitebeet answers in frame 459, but it says "mode: EV, code: module busy". This can also be seen in frame 7 and 8.

Somewhere in between frame 438 and frame 458 the mode of the V2G service module is changed.

I need to replicate your setup and investigate this issue further. In the mean time I kindly ask you to do the following:

1. If not done yet, please enable the port mirror feature by placing evse.whitebeet.networkConfigSetPortMirrorState(1) right in front of evse.loop() in Application.py. Now recreate the log you send me already.

2. Then, try to start the FreeV2G without connecting your PEV yet. Wait until the message "Wait until EV connetcs" appears and have a look if the issue still exists. Please create a log of this setup.

Thank you and I apologize for your inconvenience,
lho

nigaru05, 2023-01-05:
Hey @lho-stx ,

Thank you for your quick response and sorry for not having responded to you yesterday. The reason was that he did not have the vehicle to continue developing. Today I will attach the logs and I will comment on the new status.

While doing the tests, I was checking frames 438 and 458. When analyzing frame 458 the answer is EVSE, am I reading it wrong???

Packet 458
c0 27 41 df 00 00
35 c1

Packet 459
c0 27 41 df 00 02
00
01	-- EVSE
32 c1

In a few hours I will upload the new tests with the modification to log in (evse.whitebeet.networkConfigSetPortMirrorState(1))

Thanks a lot.

lho-stx, 2023-01-05:
Hey @nigaru05,

> While doing the tests, I was checking frames 438 and 458. When analyzing frame 458 the answer is EVSE, am I reading it wrong???

You are right, the documentation in the  WhitebeetManual v2.14 is wrong, this will be updated.

I am looking forward to your logs.

Best regards,
lho

nigaru05, 2023-01-12:
Hey @lho-stx ,

Yesterday I was able to perform the tests, sorry for the delay.

I attach the log files. Perform two tests:
1 - FW WHITEBEET_ISO15118_EVSE_V01_01_07.fwu with FreeV2G Tag EVSE_v1.1.7_1. EIM version
2 - FW WHITEBEET_ISO15118_EVSE_V02_00_01.fwu with FreeV2G Tag EVSE_v2.0.1_4. PnC version

In test number 1, after "Start V2G" the session starts but with some parameters that do not seem correct to me (Protocol 0 and Session ID: 0000000000000000), I receive the vehicle's mac but then "No EVSE ID available" and "Session stopped".

In test number 2, after "Start V2G" there is a 30 sec timeout. In the wireshark it can be seen that the vehicle sends an SDP (UDP multicast) packet and receives no response. I was reading and the PnC version does not have it implemented? On the other hand, the EIM version does, for this reason in test 1 the session starts, I think.

With neither of the 2 variants I can start my development.
I hope you can help me.

Regards.

[logs_v1.1.7_1.zip](https://github.com/Sevenstax/FreeV2G/files/10400132/logs_v1.1.7_1.zip)
[logs_v2.0.1_4.zip](https://github.com/Sevenstax/FreeV2G/files/10400134/logs_v2.0.1_4.zip)

nigaru05, 2023-01-17:
HI @lho-stx ,

Sorry to bother you, I wanted to know if you had been able to look at the logs?

Thanks in advanced.
Best regards.

lho-stx, 2023-01-24:
Hey @nigaru05,

Regarding 1:
You tried to use an older firmware with the latest FreV2G? This will fail due to mismatching in the framing protocol. Please do not try to use older firmware with the latest FreeV2G version.

Regarding 2:
First of all, your EVSE is still configured for EIM. If you want to use PnC you need to change `"payment_method": [0]` to `"payment_method": [1]` in the file Evse.py.

Also can you try to start the V2G service before starting SLAC? Please put         `self.whitebeet.v2gEvseStartListen()` in front of `cp_state = self.whitebeet.controlPilotGetState()` in Evse.py and see what happens. The EV send SECCDP before SLAC even finished and I want to see if the EVSE responds to the first SECCDP message.

You forgot to attach the UART out to logs_v2.0.1_4.zip can you please send this log too?

Thank you and best regards,
lho

lho-stx, 2023-02-13:
Hey @nigaru05,

is issue still present? If not I'd like to close this issue.

Best regards,
lho

nigaru05, 2023-02-13:
Hey @lho-stx,

Sorry for not having answered you sooner, I took a few days of vacation.
Thank you very much for her answer.

Regarding 1:
The WB-CARRIER that we bought a month ago has the firmware V01_01_07, therefore we are working with the TAG EVSE_v1.1.7_1 which is EIM.

Regarding 2:
The version EVSE_v1.1.7_1 only has the EIM.

Now I have good news. At this moment we have already been able to carry out the first load.
To solve our problems we have followed ISSUE 16 (https://github.com/Sevenstax/FreeV2G/issues/16).

The problem of "No EVSE ID available" with:
[image]

And then, when communicating by DIN70121 it is necessary to properly set the Energy transfer modes to 1 : DC extended.

At this time, I wanted to know if it is advisable to continue developing on the EIM version or go to the PnC? I open another issue and close this one? Or how do we continue?

Thanks in advanced.
Best regards,

lho-stx, 2023-02-14:
Hey @nigaru05,

thank you for your detailed answer. Please update to the PnC Version and then continue your developing. The latest version also includes several bug fixes and supports both EIM and PnC. 

We can use this ticket to track your progress, I will change the title if needed!

Like always, please create log files if you encounter problems and just send them to me.

Thank you,
lho

nigaru05, 2023-02-25:
Thank you very much for all the support @lho-stx !!!
Surely we will talk again when I continue with my development.
Thank you so much.

closed 2023-02-25 by nigaru05