Sevenstax/FreeV2G #139: V2G Communication doesn't continue after Charge Parameter Discovery

fjdicker, 2023-01-13:
**Your Setup**
Platform: EVSE
Firmware Version: V01_00_07
Host Controller Interface: Ethernet
Host: FreeV2G (EVSE_v1.1.7_1) / Own Implementation

**Describe the bug**
V2G Communication doesn't continue after Charge Parameter Discovery 

**To Reproduce**
Just use the WHITEBEET CARRIER BOARD and connect a Ford Mustang MACH E

**Expected behavior**
V2G Communication should continue with Cable Check

**Logs**
fjd@GfA-Server:~/FreeV2G$ git checkout EVSE_v1.1.7_1
Vorherige Position von HEAD war 99d7a37 Add AC specific functions to Evse class
HEAD ist jetzt bei b775bee Merge remote-tracking branch 'origin/eth_pylibpcap'
fjd@GfA-Server:~/FreeV2G$ git branch -a
* (HEAD losgelöst bei EVSE_v1.1.7_1)
  master
  remotes/origin/HEAD -> origin/master
  remotes/origin/master
  remotes/origin/plug_and_charge
fjd@GfA-Server:~/FreeV2G$ git checkout EVSE_v1.1.7_1
HEAD ist jetzt bei b775bee Merge remote-tracking branch 'origin/eth_pylibpcap'
fjd@GfA-Server:~/FreeV2G$ sudo python3 Application.py "eth0" -m c4:93:00:34:aa:55 -r EVSE
Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: eth0, mac: c4:93:00:34:aa:55
Set the CP mode to EVSE
Set the CP duty cycle to 100%
Start the CP service
Start SLAC in EVSE mode
Wait until an EV connects
EV connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
Start V2G
"Session started" received
Protocol: 2
Session ID: c9b5caa6056f51ea
EVCC ID: 0018230fd39a
"Request EVSE ID" received
Set EVSE ID: DE*ABC*E*00001*01
"Request Authorization" received
Authorize the vehicle? Type "yes" or "no" in the next 59s: yes
Vehicle was authorized by user!
"Request Schedules" received
Max entries: 128
Set the schedule: [{'id': 1, 'tuple': [{'start': 0, 'interval': 1800, 'max_power': 25000}, {'start': 1800, 'interval': 1800, 'max_power': 18750}, {'start': 3600, 'interval': 82800, 'max_power': 12500}]}]
"Request Discovery Charge Parameters" received
EV maximum current: 500.0A
EV maximum power: 207000W
EV maximum voltage: 414.0V
SOC: 64%
We did not receive a frame before the timeout of 30s
EVSE loop finished
Goodbye!
fjd@GfA-Server:~/FreeV2G$

**Additional context**
Issue #37 seems to describe the same problem
We think, 0x6B - Set Schedules generates the problem
The documentation for this command in the manual is ambiguous.
The PHYTON demo software sends a different frame as described in the manual.
If we send the command 0x6B as described in the manual we get a error.

lho-stx, 2023-01-25:
Hey @fjdicker,

indeed there is a problem with the schedule message in the whitebeet. I kindly ask you to enable the port mirror (put evse.whitebeet.networkConfigSetPortMirrorState(1) in the line above evse.loop() in Application.py) and send me a wireshark log of your session. We then can narrow down the problem.

Thank you,
lho

fjdicker, 2023-02-06:
[TRACE 18_44.zip](https://github.com/Sevenstax/FreeV2G/files/10668591/TRACE.18_44.zip)
Hi Iho,
here the log as requested.

Kind regards
F.J. Dicker

lho-stx, 2023-02-09:
Hey @fjdicker,

did you enable the port mirror, because I see no V2G messages in the log file? On first sight the command 0x6b looks ok, but I will have a closer look.

Is there any reason why you are using EVSE v1.1.7? If not please update to the latest firmware and try again.

Thank you,
lho

fjdicker, 2023-02-09:
Here the modified sourcecode.
Is the port mirror enabled?
I did the log with this software, you can find the debug output "Dicker" in line 9 of file CONSOLE 18_44.

[image]

lho-stx, 2023-02-09:
Yes this is how you enable the port mirror.
Is there a switch connected between your host PC and the Whitebeet?

fjdicker, 2023-02-09:
Yes, that's the problem.
I will send you new logs.

fjdicker, 2023-02-10:
Here the new logs!
[wireshark_2023.02.10.zip](https://github.com/Sevenstax/FreeV2G/files/10708080/wireshark_2023.02.10.zip)

lho-stx, 2023-02-13:
Hey @fjdicker,

thank you for the logs, it helps alot!

You are still using firmware version EVSE V1.1.7. Please consider upgrading to v2.0.1. There were major changes in the framing the framing protocol between the two versions.

Did you try the workaround mentioned in #37 (disabling the EVSECurrentRegulationTolerance paramter)? 

Best regards,
lho

lho-stx, 2023-02-24:
Hey @fjdicker,

did you try the workaround? Is the issue still present? If not I'd like to close this ticket. Please let me know if you just need more time!

Best regards,
lho

fjdicker, 2023-02-27:
Yes, I tried to disable EVSECurrentRegulationTolerance paramter, it's still the same.
Why should I update from 1.1.7 to 2.0.1?

lho-stx, 2023-03-07:
Hey @fjdicker,

I am still looking into this. 

Regarding the set schedules command: The example in the Whitebeet manual for firmware version 1.1.7 is wrong (16 bit integers are shown as one byte only), but the table is correct. Also the set schedules message in your log seems to be correct.

The session stops, after the EV receives the ChargeParameterDiscoveryResponse (it ACK the packet) and then does not continue with the V2G message flow. From the configuration of the Whitebeet and the session I see no reason why the EV does not continue with the session. The maximum values for the EV do not match with the EVSE maximum values, but normally this should not lead to an error. You can try to set your EVSE maximum values to at least the EV maximum values:

Maximum Current: 500A
Maximum Power: 207kW
Maximum Voltage: 414V

The latest version of the Whitebeet firmware and FreeV2G includes feature improvements and bug fixes. 

**NOTE:**  There where breaking changes in the framing protocol between 1.1.7 and 2.0.1

nststx, 2023-04-24:
Hi @fjdicker,

Does your issue still persist? Have you tried the suggestion by @lho-stx about the maximum values? Please let me know so we can help you further with this issue.

Best regards,
Niklas Stoffers

nststx, 2023-05-08:
Closing, due to inactivity. If this issue still persist, feel free to reopen.

closed 2023-05-08 by nststx