Sevenstax/FreeV2G #140: No message after SLAC matching - traces

fjdicker, 2023-01-16:
#############################################################
No message from whitebeet after successful SLAC matching
#############################################################
Is it caused by
0x05 | 0x80 | 0xFF | 1d = 0x00
Interface Down Access Point
#############################################################

APPLICATION:
  Name:      ISO15118 (EVSE)
  Version:   V01_01_07
  Target:    WHITE-BEET
  CPU:       STM32F745

PRODUCT VERSION:
 String: SEVENSTAX_LIB_V_10_00_03
 Hex:    0x000a0003
 Dec:    100003

 Hostcontroller Interface: SPI!

APPLICATION RUNNING ...
[    0.024] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from host
[    0.031] [NOTICE] APL-MAIN: BOOT-CFG: FW1
[    0.035] [NOTICE] APL-MAIN: BOOT-CFG: Module configured to EVSE
[    0.041] [NOTICE] Q7KCOM:  Boot from host selected
[    0.046] [NOTICE] SLAC: stxSLAC_Init()
[    0.565] [NOTICE] V2GCPS: stxV2GCPS_RegisterCallback()
[    0.573] [NOTICE] APL-ETH: Got link:
[    0.577] [NOTICE] APL-ETH:     Interface: eth1
[    0.581] [NOTICE] APL-ETH:     MAC:       C4:93:00:34:AC:8F
[    0.779] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    0.784] [NOTICE] Q7K: =========================================
[    0.790] [NOTICE] Q7K:   QCA700x is READY
[    0.794] [NOTICE] Q7K: =========================================
[    0.800] [NOTICE] Q7KCOM: Reset QCA700x
[    0.805] [NOTICE] Q7K: =========================================
[    0.811] [NOTICE] Q7K:   CPU ON RECEIVED
[    0.815] [NOTICE] Q7K: =========================================
[    0.821] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    0.826] [NOTICE] Q7K: =========================================
[    0.832] [NOTICE] Q7K:   QCA700x is READY
[    0.836] [NOTICE] Q7K: =========================================
[    1.652] [NOTICE] Q7KCOM:  =======================================
[    1.658] [NOTICE] Q7KCOM:    FW and PIB was loaded successful
[    2.167] [NOTICE] Q7K: =========================================
[    2.173] [NOTICE] Q7K:   CPU ON RECEIVED
[    2.177] [NOTICE] Q7K: =========================================
[    2.183] [NOTICE] Q7KCOM:  =======================================
[    2.189] [NOTICE] Q7KCOM:    BOOT FROM HOST READY
[    2.194] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
[    2.199] [NOTICE] Q7K: =========================================
[    2.205] [NOTICE] Q7K:   QCA700x is READY
[    2.209] [NOTICE] Q7K: =========================================
[    2.342] [NOTICE] Q7KCOM:  Own QCA7K MAC = C4:93:00:34:AC:91
[    3.622] [NOTICE] APL-ETH: Got link:
[    3.626] [NOTICE] APL-ETH:     Interface: eth0
[    3.630] [NOTICE] APL-ETH:     MAC:       C4:93:00:34:AC:90
[   13.337] [NOTICE] APL-MAIN:
[   13.339] [NOTICE] APL-MAIN: =============================
[   13.345] [NOTICE] APL-MAIN: NEW LAN CONFIGURATION (IPv6):
[   13.350] [NOTICE] APL-MAIN: for IP Config handle [fe00]
[   13.356] [NOTICE] APL-MAIN: IPv6 Addresses:
[   13.360] [NOTICE] APL-MAIN:   Link-Local:  fe80:0:0:0 : c693:ff:fe34:ac90
[   13.367] [NOTICE] APL-MAIN:   Local: 0:0:0:0 : 0:0:0:0
[   13.372] [NOTICE] APL-MAIN:   DNS1:  2001:4860:4860:0 : 0:0:0:8888
[   13.378] [NOTICE] APL-MAIN:   DNS2:  fd00:0:0:0 : be05:43ff:fe44:5280
[   13.384] [NOTICE] APL-MAIN: =============================
[   13.390] [NOTICE] APL-MAIN:
[   95.604] [NOTICE] SLAC: stxSLAC_Start()
[... 512 lines cut ...]
Command accepted 
WhiteBeet: 0x29 | 0x48 | 0x98 | 2d = 0x00, 0x00, 
WhiteBeet Send: 0x29 | 0x48 | 0x99 | 0d = 
Command accepted 
WhiteBeet: 0x29 | 0x48 | 0x99 | 2d = 0x00, 0x00, 
WhiteBeet Send: 0x29 | 0x48 | 0x9A | 0d = 
Command accepted 
WhiteBeet: 0x29 | 0x48 | 0x9A | 2d = 0x00, 0x00, 
WhiteBeet Send: 0x29 | 0x48 | 0x9B | 0d = 
Command accepted 
WhiteBeet: 0x29 | 0x48 | 0x9B | 2d = 0x00, 0x00, 
WhiteBeet Send: 0x29 | 0x48 | 0x9C | 0d = 
WhiteBeet: 0x29 | 0x81 | 0xFF | 1d = 0x01, 
Command accepted 
WhiteBeet: 0x29 | 0x48 | 0x9C | 2d = 0x00, 0x01, 
EV connected 
Start SLAC matching 
WhiteBeet Send: 0x28 | 0x44 | 0x9D | 0d = 
Command accepted 
WhiteBeet: 0x28 | 0x44 | 0x9D | 1d = 0x00, 
Set the CP duty cycle to 5% 
WhiteBeet Send: 0x29 | 0x44 | 0x9E | 2d = 0x00, 0x32, 
Command accepted 
WhiteBeet: 0x29 | 0x44 | 0x9E | 1d = 0x00, 
WhiteBeet: 0x29 | 0x81 | 0xFF | 1d = 0x00,

jpo-stx, 2023-02-14:
Hi @fjdicker ,
It seems that the EV is not starting the matching process. Is this after a new plug in of the vehicle?
You are also using an old firmware version for your Whitebeet. There is a version 2.0.1 available.
Can you activate the port mirror and do a wireshark log on the ethernet interface?

nststx, 2023-04-19:
Hi @fjdicker,

The newer firmware versions contain new features and bugfixes which might fix the problem you're facing. Regarding your issue: Does it still persist? If it does could you capture a wireshark log on the ethernet interface with port mirror active?

Best regards,
Niklas Stoffers

nststx, 2023-05-03:
Closing, due to inactivity. If this issue still persist, feel free to reopen it.

closed 2023-05-03 by nststx