Sevenstax/FreeV2G #142: Session stopped after receiving 'EVSE did not accept our ServiceDetailReq!' log message.

Bobbybigears, 2023-01-23:
**Your Setup**
Platform: Explorer16-dsPIC33CH512MP508/ Whitebeet Carrier Board EV ISO15118 (fw V1.0.6_0) / AQUA 75W DC Charger
Firmware Version: i.e. V01_00_06
Host Controller Interface: C/SPI

**Describe the bug**
When running the host firmware with the Whitebeet Carrier Board EV and a Whitebeet Carrier Board  EVSE ISO15118 (fr V1.1.7) to simulate the charger, the firmware will pass the point where a 'Session Started' status message is received and enter the charge loop.
When connecting to the AQUA DC Charger, the EV receives a 'EVSE did not accept our ServiceDetailReq!' message just prior to where it would enter the charge loop. Following this a 'Session Stopped' command is received and the firmware does not progress any further.

I've included both protocols in the initial configuration in addition to all of the DC Charge Modes.

I've included logs from the UART Debug Port for both configurations, i.e. the Whitebeet Carrier Board EVSE config and AQUA 75W DC Charger config.

Note: For convenience, I currently use the last command sent to clock data through. This is why there are multiple entries in the log for certain commands.

I would be most grateful for any help you could provide me with with regard to this issue.

Best regards
Rob
[whitebeet-ev.txt](https://github.com/Sevenstax/FreeV2G/files/10481213/whitebeet-ev.txt)
[aqua-ev.txt](https://github.com/Sevenstax/FreeV2G/files/10481215/aqua-ev.txt)

Bobbybigears, 2023-01-24:
Further to the issue described and with regard to the 'DC Energy Transfer Mode and Start of Session Sequence' - Figure 37 in section 19.9.2 in Whitebeet User Manual V2.11.2, do the V2G Service/EVSE interactions depicted return error codes and is there a way to determine what these are by means of the Whitebeet API? At the moment, the only feedback I receive is the 'Session Stopped' message which does not include any error information.
Best regards, Rob.

Bobbybigears, 2023-01-24:
Apologies, one more question.... 
Is this the type of issue that attenuation problems would cause?
Best regards
Rob

lho-stx, 2023-01-25:
Hey @Bobbybigears,

there is an issue regarding the ServiceDetail message. Per ISO15118 the message is optional  and it seems your AQUA 75W DC Charger does not support this message, therfore the session fails. I need to check internally if I can provide you with a workaround.

Unfortunately there is no API to receive the cause of failure of a V2G session at the moment, but I will discuss this feature.

Regarding your last question:
This is not due to attenuation problems. If the attenuation is to high (or if there is ambiguity to which EVSE the EV is connected), EV  does something called "BCB-Toggling" where it changes the voltage of the control pilot between to values. If the EVSE sees these voltage drops it will notify the EV, so both know to whom they are connected to.

Thank you and best regards,
lho

Bobbybigears, 2023-01-25:
Thank-you for your prompt reply. Much appreciated.

Bobbybigears, 2023-01-26:
Hi lho,
Is there any news with regard to a workaround? 
Cheers
Rob

LiamBattershell, 2023-01-27:
Hi Iho-stx, What does this mean for our development project? does this mean that the Whitebeet system for EV is not compatible with all CCS chargers? How do we tell what charger it would work with? We have put in a lot of development time into this for it to not work in the final stages.

lho-stx, 2023-01-27:
Hey @LiamBattershell,

first of all I apologize for not stating this clear:

There will be soon a firmware release (planned for late Q1/early Q2) that includes a solution for this specific problem. 

Until then the Whitebeet EV is not compatible with chargers that violate this specific requirement but you can try to use DIN with these chargers, because DIN hast no ServiceDiscovery messages.

Best regards,
lho

lho-stx, 2023-02-03:
Closing this issue. This will be fixed and tracked in interal issue tracker.

closed 2023-02-03 by lho-stx

Kristanpivk123, 2024-07-11:
Hi,
just today I encountered this exact problem on two of four EVSEs that I tested.
I am using WHITE Beet ISO15118 EV carrier board with V01_00_07 which I command over SPI.
Is it possible that this issue was not yet fixed in the latest FW? Or is there some instruction by which I can disable ServiceDetailReq?
Current FW on the board was uploaded by you (or CODICO, I had to send it since it was bricked because of V01_00_06 FW) a few months ago so I think it should be correct but it is true that when I ask module for FW version, it replies with 0.0.0.0 (until now there were no other problems).
Best regards,
Kristan

Bobbybigears, 2024-07-12:
Kristan. Have nothing to do with FW for WHITEBeet. Was developing a product
using WHITEbeet for a while and had the same issue. It's a while back now,
but believe I managed to get around this by using another profile. Ran into
more problems not long after this and called it quits.

-- 
Rob Davies
(Software Engineer)

Treharne Automotive Engineering Ltd
Treharne Workshops
Llanelli Gate
Dafen
Llanelli
Carmarthenshire
SA14 8LQ

Tel: 01554 775938

Kristanpivk123, 2024-07-12:
Thanks for the reply, Bobby.
If I may ask, did you create a charging profile and that fixed it? I currently don't send a charging profile so if that overcomes this, it will be great.

Bobbybigears, 2024-07-12:
Hi Kristan. Apologies, probably used the wrong terminology. When
communicating with the EVSE, there's a section where you select ISO15118
or DIN7012. Used to get a comms error having selected ISO15118. Managed to
get past the failure point by selecting the DIN7012 option. Not sure if
this is applicable to your situation, also used C as the programming
language to interface with the Whitebeet. Sorry I can't be of more help.

-- 
Rob Davies
(Software Engineer)

Treharne Automotive Engineering Ltd
Treharne Workshops
Llanelli Gate
Dafen
Llanelli
Carmarthenshire
SA14 8LQ

Tel: 01554 775938

Kristanpivk123, 2024-07-12:
Oh, believe me, you were already a lot of help, thank you!
But some additional questions if you don't mind: 
- did you encounter an EVSE that had this problem (ServiceDetailReq not accepted) but didn't support DIN 70121?
- you mentioned that it has nothing to do with WHITE Beet FW; does that mean that FW is working OK and that this is expected behaviour?
- do you perhaps remamber what those other problems were that you encountered? just to see if I can also expect them

Best regards,
Kristan

Bobbybigears, 2024-07-12:
Can't remember the Charger we used, but according to SevenStax it didn't
support a property that 'ServiceDetailReq' required for ISO15118.
Apologies, can't say if this is expected behaviour or not, don't know
enough about the standards involved and how errors are supposed to be
handled. I can't remember the specific error that occurred after the
'ServiceDetailReq' problem, but would have logged it on their GitHub page.
It was recommended to flash to the latest firmware to get around the error
and that resulted in additional errors. Gave up at that point .

-- 
Rob Davies
(Software Engineer)

Treharne Automotive Engineering Ltd
Treharne Workshops
Llanelli Gate
Dafen
Llanelli
Carmarthenshire
SA14 8LQ

Tel: 01554 775938

Kristanpivk123, 2024-07-12:
Thank you again for all of the answers.
Best regards,
Kristan