Sevenstax/FreeV2G #148: KIA EV6 charge problem with ISO and DIN protocols

Ivangel1972, 2023-02-09:
Hello, I am having a problem getting my KIA EV6 car charging to work.  After SLAC EVSE pairing, it sets WB module m/in other protocols type, DIN and ISO. The car sends Start Session where in this configuration it sends that it supports DIN protocol, then requests EVSE ID - it receives it and nothing more happens - where normally it sends a request for the rest of the parameters. After about 5.3s the WB module sends Stop Session.
If I connect the BMW i3 before trying to charge the KIA and start charging - then the connected KIA starts charging correctly, identifying itself as DIN. If I set EVSE to offer only ISO protocol - the KIA starts charging every time, this time also identifying itself as ISO (Start Session sends ISO) - but of course it is impossible to start charging on the BMW i3.
If I restart the EVSE, which is set by default to support both protocols, DIN and ISO, causing a hard reset of the WB module, through the corresponding input on it - the KIA does not charge, unless only the ISO protocol is set, as above.
That is, the car will start charging if - DIN AND ISO supported and previously charged the car with DIN support or EVSE supports only ISO - but this again leads to the inability to charge, for example: BMW i3.
Normally, the WB module is initiated after startup and after each charge according to the example from V2GFree.
WB - that is, the WhiteBeet module has FW version 01_01_07

lho-stx, 2023-02-09:
Hey @Ivangel1972,

this seems a more complex issue. I kindly ask you to enable the port mirror (put evse.whitebeet.networkConfigSetPortMirrorState(1) in the line above evse.loop() in Application.py) and send me a wireshark log of your sessions.

Also, is there any reason why you using firmware version v01_01_07? If not, please update to the latest Whitebeet firmware and try again.

Thank you,
lho

lho-stx, 2023-02-13:
Hey Marcin,

you are using Whitebeet EVSE, right? Then the latest version of the firmware is "Sevenstax_WHITEBEET_ISO15118_EVSE_PNC_V02_00_01_EXPERIMENTAL_update.zip".

If you are using your own implementation you can enable the port mirror via the framing protocol:

https://raw.githubusercontent.com/Sevenstax/FreeV2G/EVSE_v1.1.7_1/SEVENSTAX_WhiteBeet_UserManual.pdf#page=75

I kindly ask you to enable the port mirror prior running your application, create a capture of the SPI traffic and send it to me? If the KIA successfully receives the EVSE ID and there is a problem we should see the error message in the v2g communication.

Also if you have the possiblity please create a log of the UART output of the Whitebeet module.

Thank you,
lho

lho-stx, 2023-02-27:
Hey @Ivangel1972,

did the proposed solution worked out? If yes please let me know, I would like to close this issue ticket then!

Best regards,
lho

Ivangel1972, 2023-03-06:
Hi lho, no, i will capture in putty debug, from boot WB module (and whole DC contrroller) to trying charging KIA EV6:

``▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒

 APPLICATION:
               Name:      ISO15118 (EVSE)
                                           Version:   V01_01_07
                                                                 Target:    WHIT E-BEET
        CPU:       STM32F745

                             PRODUCT VERSION:
                                              String: SEVENSTAX_LIB_V_10_00_03
                                                                               H ex:    0x000a0003
                  Dec:    100003

                                  Hostcontroller Interface: SPI!

                                                                 APPLICATION RUN NING ...
        [    0.025] [NOTICE] APL-MAIN: BOOT-CFG: Boot QCA7005 firmware from host
                                                                               [     0.031] [NOTICE] APL-MAIN: BOOT-CFG: FW1
                                           [    0.035] [NOTICE] APL-MAIN: BOOT-C FG: Module configured to EVSE
                             [    0.041] [NOTICE] Q7KCOM:  Boot from host select ed
  [    0.046] [NOTICE] SLAC: stxSLAC_Init()
                                           [    0.565] [NOTICE] V2GCPS: stxV2GCP S_RegisterCallback()
                    [    0.573] [NOTICE] APL-ETH: Got link:
                                                           [    0.577] [NOTICE]  APL-ETH:     Interface: eth1
                            [    0.581] [NOTICE] APL-ETH:     MAC:       C4:93:0 0:21:4E:C0
          [    0.780] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
                                                                    [    0.785]  [NOTICE] Q7K: =========================================
                                                       [    0.791] [NOTICE] Q7K:    QCA700x is READY
                   [    0.795] [NOTICE] Q7K: =================================== ======
      [    0.801] [NOTICE] Q7KCOM: Reset QCA700x
                                                [    0.806] [NOTICE] Q7K: ====== ===================================
                                   [    0.812] [NOTICE] Q7K:   CPU ON RECEIVED
                                                                              [     0.816] [NOTICE] Q7K: =========================================
                                                                 [    0.822] [NO TICE] Q7K: QCA7K_SPI_REG_SIGNATURE = 0xAA55
                                           [    0.827] [NOTICE] Q7K: =========== ==============================
                              [    0.833] [NOTICE] Q7K:   QCA700x is READY
                                                                          [    0 .837] [NOTICE] Q7K: =========================================
                                                             [    1.519] [NOTICE ] Q7KCOM:  =======================================
                                                   [    1.526] [NOTICE] Q7KCOM:     FW and PIB was loaded successful
                                    [    2.014] [NOTICE] SLAC: stxSLAC_Stop()
                                                                             [     2.103] [NOTICE] Q7K: =========================================
                                                                [    2.109] [NOT ICE] Q7K:   CPU ON RECEIVED
                           [    2.113] [NOTICE] Q7K: =========================== ==============
              [    2.119] [NOTICE] Q7KCOM:  ==================================== ===
    [    2.126] [NOTICE] Q7KCOM:    BOOT FROM HOST READY
                                                         [    2.130] [WARN  ] V2 GCPS: stxV2GCPS_Stop() - Failed
                               [    2.135] [NOTICE] Q7K: QCA7K_SPI_REG_SIGNATURE  = 0xAA55
         [    2.141] [NOTICE] Q7K: =========================================
                                                                            [     2.147] [NOTICE] Q7K:   QCA700x is READY
                                        [    2.151] [NOTICE] Q7K: ============== ===========================
                           [    2.157] [NOTICE] Q7KCOM:  Own QCA7K MAC = C4:93:0 0:21:4E:C2
          [    2.281] [NOTICE] SLAC: stxSLAC_Start()
                                                    [    3.362] [NOTICE] APL-ETH : Got link:
           [    3.366] [NOTICE] APL-ETH:     Interface: eth0
                                                            [    3.370] [NOTICE]  APL-ETH:     MAC:       C4:93:00:21:4E:C1
                                          [    4.286] [NOTICE] APL-ETH: Lost lin k:
  [    4.289] [NOTICE] APL-ETH:     Interface: eth0
                                                   [    4.294] [NOTICE] APL-ETH:      MAC:       C4:93:00:21:4E:C1
                                 [    4.299] [NOTICE] APL-MAIN:
                                                                [    4.302] [NOT ICE] APL-MAIN: =============================
                                            [    4.308] [NOTICE] APL-MAIN: LAN C ONFIGURATION DELETED:
                      [    4.313] [NOTICE] APL-MAIN: for IP Config handle [fe00] 
 [    4.318] [NOTICE] APL-MAIN: IPv6 Addresses:
                                               [    4.322] [NOTICE] APL-MAIN:    Link-Local:  0:0:0:0 : 0:0:0:0
                              [    4.328] [NOTICE] APL-MAIN:   Local:   0:0:0:0  : 0:0:0:0
[... cut ...]
                         [   47.852] [NOTICE] APL-MAIN:   DNS2: 0:0:0:0 : 0:0:0:0
                                                                                [   47.857] [NOTICE] APL-MAIN: =============================
                                                           [   47.862] [NOTICE] APL-MAIN:
         [   48.804] [NOTICE] APL-ETH: Got link:
                                                [   48.808] [NOTICE] APL-ETH:     Interface: eth0
                [   48.812] [NOTICE] APL-ETH:     MAC:       C4:93:00:21:4E:C1

lho-stx, 2023-03-07:
Hey @Ivangel1972,

unfortunately github scrambles your log, maybe it is better to upload a .txt file next time?

Nevertheless, to investigate the issue further I need the SPI and Wireshark logs, from the Putty output I can not see if there are some V2G session related errors. Please enable the port mirror and create at least the Wireshark log by connecting the Whitebeet directly to a PC running Wireshark.

If you have trouble setting up the port mirror or creating the Wireshark log, please let me know!

Thank you,
lho

Ivangel1972, 2023-03-07:
Hey Iho, i have log i txt format - i thik its no diff between both  - but i can send you. But where?

I understand that you expect that via the ethernet port of the WB module it will connect to Wireshark and through the "port mirror" function I will capture communication between EVSE and EV?
I understand that now, in the project that we have been developing for a year, and which we are installing in ready-made EVSEs, I have to build an Ethernet system to send the data you are asking for?
Or rewrite half of the program to implement the new API of the module, which you call "new firmware", which in fact is definitely different than the previous one - to see if it solves the problem?
In the DC controllers that we have been installing in EVSE sent to customers for half a year?

As for the SPI log - in what format should it be readable for you? i have KingsVis software to capture any data from digital lines (with analyzer)

Regards

Marcin - Ivangel

lho-stx, 2023-03-08:
Hey @Ivangel1972,

I have to apologize, from your original post I did not read that you are using custom hardware and implementation. I assumed you are using the Whitebeet carrier board and the FreeV2G application.

Regarding the text logs:
Github applied markdown formatting to the logs, there are random whitespaces, newlines and part of lines are missing, therefore I asked you to send a textfile. Just append the text file in your next post, thank you.

Regarding the SPI logs:
I just downloaded the KingstVIS analyzer software. If you send me a KingstVIS log of your sessions I should be able to open and investigate it!

Thank you,
lho

nststx, 2023-04-19:
Hi @Ivangel1972,

Does your issue still persist? If it does could you send us the missing log files for further investigation?

Best regards,
Niklas Stoffers

Ivangel1972, 2023-04-22:
Yes issue stil persist.  Unfortunately, I don't have any of the problematic cars to log my work at the moment. As soon as there is a possibility, I will immediately let you know
Marcin Matusiak

Ivangel1972, 2023-05-13:
Hi Niklas Stoffers, i have logs in KingsVis (http://www.qdkingst.com/en/download) few "conversations" with CPU and WB in some situations:
Init - only init WB module, from Reset to periodic question about CP State
Bmw_i3 - short BWM I3 charging session
Ionic5 - short Hyundai Ionic5 charging session
Init_Bmw_I3 - Init WB module and short charging session
Init_ionic5 - init WB module and refuse charging by Hyundai Ionic5
Init_Bmw_I3_Ionic5  - init WB module and short charging session BMWI3 and nex short charging session Hyunai Ionic5

In the logs there is a small difficulty - the controller on which I collected data supports 2 WB modules and on the log you can see both "conversations" of the CPU with the modules. However, the SPI Enable signal applies only to the module we are interested in and only this data should be analyzed.

Please analyze this data and tell me what I am doing wrong.

p.s. Hyundai Ionic5 i the same as KIA EV6... 

https://drive.google.com/drive/folders/1Qr7BgE8fl-32VKTZ74-mGSpjryITpIcz?usp=sharing

Regards, Marcin Matusiak

Ivangel1972, 2023-05-14:
> Hi @Ivangel1972,
> 
> Does your issue still persist? If it does could you send us the missing log files for further investigation?
> 
> Best regards, Niklas Stoffers

Hi Niklas Stoffers, i have logs in KingsVis (http://www.qdkingst.com/en/download) few "conversations" with CPU and WB in some situations:
Init - only init WB module, from Reset to periodic question about CP State
Bmw_i3 - short BWM I3 charging session
Ionic5 - short Hyundai Ionic5 charging session
Init_Bmw_I3 - Init WB module and short charging session
Init_ionic5 - init WB module and refuse charging by Hyundai Ionic5
Init_Bmw_I3_Ionic5 - init WB module and short charging session BMWI3 and nex short charging session Hyunai Ionic5

In the logs there is a small difficulty - the controller on which I collected data supports 2 WB modules and on the log you can see both "conversations" of the CPU with the modules. However, the SPI Enable signal applies only to the module we are interested in and only this data should be analyzed.

Please analyze this data and tell me what I am doing wrong.

p.s. Hyundai Ionic5 i the same as KIA EV6...

https://drive.google.com/drive/folders/1Qr7BgE8fl-32VKTZ74-mGSpjryITpIcz?usp=sharing

Regards, Marcin Matusiak

lho-stx, 2023-05-26:
Hey @Ivangel1972,

thank you for your effort. Can you please give me a short overview of the channels? Channel 1 and Channel 5 are asserted and de-asserted at the same moment. Do you use one or two SPI interfaces?

Best regards,
lho

P.S.: If you have access to a logic analyzer supporting Saleae Logic2 I would prefer the .sal format

Ivangel1972, 2023-05-26:
Hey @Iho,
not the same moment. 
I suspect that you do not have a description of the channels, so:
0  -  Tx Pending
1 - Rx Ready
2 - SPI MISO
3 - SPI MOSI
4 - SPI CLK
5 - SPI SCS (chip select)
6 - Reset

if you take a closer look at Ch 5, i.e. SPI chip select is always active (low state) a little earlier than the Rx ready line (ch 1) - this is the "answer" from the WB module

P.S: I understand, but I do not have such equipment
Regards, Marcin

lho-stx, 2023-06-06:
Hey @Ivangel1972,

Thank you for clearification!
 
No problem, I converted your files into a format compatible with our tools.

I had a first look into the logs. First of all it seems that there is a problem with your host controller implementation right in the first frame:

[image]

On MOSI there is 0x55 set as size of the upcoming data exchange frame from the host. The next frame is a new (this time valid) size exchange frame. So maybe you need to adapt your host controller implementation.

Nevertheless, this does not lead to the describe issue. I inspected the logs further, but I need to continue investigation tomorrow.

Please allow some more time for investigation.

Thank you,
lho

lho-stx, 2023-06-16:
Hi @Ivangel1972,

we inspected the log files and I see no problem in the framing messages. We also tried to run into this error by reproduce your setup, but there is not enough information in the logs. For further investigation, I *need* logs of the V2G traffic between EV and EVSE. I see three options now:
1. Managing, somehow, to get access to the ETH port on the whitebeet
2. Swapping out the whitebeet with a whitebeet carrier board
3. Using a PLC sniffing box

I understand your frustration, but unfortunately the SPI logs do not contain enough information.

Best regards,
lho

Ivangel1972, 2023-06-16:
Hi @lho-stx - It's not about frustration, although a little trust would be useful. When I wrote that other cars load without a problem, I am convinced that there is no mistake on my part - we lost a lot of time to prove it.
Ok, I'll try to do it like in point 1 - add access to the ETH port - and then what? Please give full instructions on what to do so that I can provide you with further information.
Best regards
Marcin - @Ivangel1972

lho-stx, 2023-06-23:
Hey @Ivangel1972,

from my experience I can tell you there are some cars that do not comply to the ISO15118 specification, so it is not unlikely that the car is the problem, not your application!

In order to create a wireshark log, please connect the ETH port directly to your PC (without any network gear like switches in between). Then start sniffing on the interface you have the whitebeet connected to. Then you need to enable port mirror before you start the SLAC service (for the v1.1.7 firmware it is modID 0x05 and subID 0x55. Please see `12.3.10 Set Port Mirror State` in the Whitebeet manual v2.11.2  for more information).  Then start your charging session. If the session build up is done (so when the first CurrentDemandReq/Res appear) you can stop recording, I do not need the whole charging session, the negotiation of the session is enough!. Please try to create a wirehark log for every charging session, like you did with the SPI logs, this was great and helps a lot!

Thank you for your effort,
lho

Ivangel1972, 2023-06-23:
Hey @lho-stx 
In that case, maybe it would be good to make a list of cars that do not meet the assumptions of a given standard. 
OK, I know what to do, I have to organize it logistically and I will deliver what I can log in.
Regards
Ivangel1972

lho-stx, 2023-06-26:
Hey @Ivangel1972,

this is solely my subjective experience and I did not create a structured survey in any way which EVs are affected by this. I need to discuss internally if this is possible.

Looking forward to your logs!

Thank you,
lho

lho-stx, 2023-06-26:
Hey @Ivangel1972,

I see your points and personally I agree with you. I already informed the managment about the demand for such a list and will update you if I have an answer for you!

Best regards,
lho

Ivangel1972, 2023-07-18:
Hey Iho, look:
https://github.com/Sevenstax/FreeV2G/issues/110
it was more than half year ago, you know issue, you have log's, and?!
Gentlemen, time has become critical and we expect an immediate response from you
@Ivangel1972

lho-stx, 2023-07-18:
Hey @Ivangel1972,

I asked you to create the logs in my answer from last month: https://github.com/Sevenstax/FreeV2G/issues/148#issuecomment-1604482279

From the UART and SPI logs I can not identify any problem in the V2G communication. We need these logs. If you have problems creating the logs please let us know so we can support you. If you are unable to create such a log (physical constraints, non-working ETH port), please also let us know so we can work out another solution to get the V2G communication logs.

Best regards,
lho

ttu-stx, 2023-07-18:
Hi @Ivangel1972,
I have deleted your last comment due to misleading informatiuon. Pls focus on information for debugging purposes.
BR

lho-stx, 2023-07-21:
Hey @Ivangel1972,

I have reread your issue again and correct me if I am wrong, but wouldn't it be possible to:

- configure Whitebeet to DIN+ISO
- plug in KIA EV6/Inoiq5
- set duty cycle to 5%
- Whitebeet starts charging session with the KIA EV6/Ioniq5 (it will try DIN and fail)
- your user application sees the charging session failing
- change duty cycle to 100%
- keep KIA EV6/Inoiq5 plugged in
- reconfigure Whitebeet to ISO
- change duty cycle to 5% (thus reinitiating charging)
- after sucessful charging, reconfigure Whitebeet to DIN+ISO on plug out (state B > state A)

Based on your initial post, this would lead to a succesful charging session, because you mentioned the KIA EV6/Inoiq5 charges with ISO.

Best regards,
lho

Ivangel1972, 2023-07-22:
Hey Iho
I tried this. I did a special procedure, even reinitialised the connection to the car by setting PWM = 0% - unfortunately the car being still connected will no longer start charging.
I have an additional question: I connect an Ethernet module to the WB, does it get along with any Ethernet controller with MII/RMII protocol, or does it have to be a specific model somehow?
Greetings
Marcin

adirschl, 2023-07-25:
Hi Marcin, 
on WHITEbeet Evalboard we use KSZ8081RNBCA, but any other RMII ethernet PHY should work as well.
BR,
Andreas
CODICO

lho-stx, 2023-07-26:
Dear @Ivangel1972,

Thank you for trying the workaround.

We also have tested against an Ioniq 5 and the Ioniq was performing DIN based communication without errors. It is possible that there are different vehicles with different software versions, so we can not proceed with supporting this case without being provided with wireshark logs. Please equip at least your testing EVSE with ethernet access or use a PLC sniffer. Furthermore we strongly recommend to update White Beet and be prepared for future updates as the vehicle market is growing.

The latest version also includes an enhanced  reset handling of the QCA which might help to perform the reinitialization without unplugging the connector.

Thank you and best regards,
lho

closed 2023-07-26 by lho-stx

Ivangel1972, 2023-08-16:
Dear Iho - I have a question: will the ETH interface work if the WHITEBEET module is configured for SPI communication?
Because it looks to me like it doesn't - making it impossible to catch the data.
@Ivangel1972

lho-stx, 2023-08-16:
Hey @Ivangel1972,

when using SPI as host controller interface, the ETH interface only acts as port mirror (when enabled). This means you should see V2G communication on the ETH interface. Did you enable the port mirror prior running your charging session? I assume you connected the ETH port of the whitebeet directly to your PC?

Please let me know, if you tried the things mentioned above, so I can find a solution for you!

Best regards,
lho

Ivangel1972, 2023-08-16:
Hey Iho - Yes, i enabled port mirror and i have nothing on ETH - but if You are sure about this - I must have made a mistake somewhere. Thank you.
@Ivangel1972

lho-stx, 2023-08-16:
@Ivangel1972,

thank you for clarification. If you are sure the port mirror is enabled I will try to replicate your issue.

Thank you,
lho

Ivangel1972, 2023-08-16:
Iho - one more question: when i enable port mirror? after SLAC match ?
@Ivangel1972

lho-stx, 2023-08-16:
It is the best to enable port mirror right at start up. Please have a look into the [latest Application.py ](https://github.com/Sevenstax/FreeV2G/blob/f7039015eebd6d4cbca82fc463a31e62aa808135/Application.py#L94), there it is set before anything else is happening.

Best regards,
lho