Sevenstax/FreeV2G #153: WHITE-BEET empty session ID

RPA66, 2023-02-17:
I submit this issue with white beet P version:

Platform: PEV 

Firmware Version:  V01_00_06
Host Controller Interface: Ethernet
Host: FreeV2G (EV_v1.0.6_1)

We have tried establishing a communication session using DIN instead of ISO, since our SECC does accept empty session ID when using DIN. However we have encountered another problem. After receiving PreChargeRes with present voltage = target voltage "Start charging" is sent from the host controller (0x27 0xAC), but WHITE-BEET terminates communication, and after 5 seconds reports "Session stopped" to the host.

Attached all logs from the module.

[2023_02_14.zip](https://github.com/Sevenstax/FreeV2G/files/10768377/2023_02_14.zip)

lho-stx, 2023-02-20:
Hey @RPA66,

thank you for the logs. Did you enable the port mirror feature of the Whitebeet and are you directly connected to the Whitebeet module or is there a network switch in between? Unfortunately I see no V2G messages in the wireshark log. Also I can not see the SECCDP response from the SECC. I kindly ask you to adapt your setup accordingly and create a new log of your failing session.

Thank you and best regards,
lho

Dalachowsky, 2023-02-20:
Port mirroring is enabled and there is a switch in between the Whitebeet and the host controller. Should i connect to the Whitebeet directly in order to receive V2G messages?

lho-stx, 2023-02-20:
Hi @RPA66,

thank you for your fast response! Yes you need to connect the Whitebeet directly to your host controller!

Thank you,
lho

Dalachowsky, 2023-02-20:
Thank you for the info! Seeing V2G messages from Whitebeet is going to be very helpful.

I'm sending the logs that i was able to get from wireshark.

[wireshark_log.zip](https://github.com/Sevenstax/FreeV2G/files/10783211/wireshark_log.zip)

lho-stx, 2023-02-20:
Hey @Dalachowsky, @RPA66,

yes thank you! Now, are the last wireshark log and the debug prints from the same session? Because in the debug prints the SoC is set to 36%, in the wireshark log it is set to 60%.

Another thing I stumbled upon: the SECC maximum power is set to 250W. In the V2G messages the EV asks for 150V and 40A, this does not match. Also the debug prints states the EV wants a target voltage of 250V, not 150V, therefore my question.

Nevertheless, the EV fails after receiving PreChargeResponse due to an error in the V2G message parser implementation. The question is, if there is problem in the Whitebeet V2G message parser or is the send PreChargeResponse malformed.

I will check the implementation of the Whitebeet V2G message parser, in the meantime you may have a look in the above mentioned issues. Please can you give some hints about your used SECC? Is it your own implementation?

Thank you and best regards,
lho

Dalachowsky, 2023-02-20:
Regarding SOC and voltage mismatch it's possible that i forgot to switch back the EV configuration. Sorry for that.

However the issue with that maximum power value is another problem that we encountered. It seems that the Whitebeet just rewrites SECC maximum voltage from ChargeParameterDiscoveryRes into maximum power, since in the logs you can see that SECC reported 11,4 kW. When i changed SECC maximum voltage to 150V the maximum power that Whitebeet reported also changed to 150 W.

I'll also mention that we had to slightly modify FreeV2G by changing the Whitebeet.slacMatched() method. Previously it was waiting for the arrival of "SLAC process was succesful" frame. However during communication session the modules finished SLAC matching, our SECC was waiting for 60s and breaking the connection after not recieving any frames. The "SLAC process was succesful" frame was arriving right after SECC cut off the connection. That's why we changed the behavior of slacMatched() by polling WHITE-BEET module every few seconds for SLAC matching status. 

Our SECC is vSECC from vector.

lho-stx, 2023-02-20:
Hey @Dalachowsky,

regarding the wrong maximum power issue:
I have looked it up and this is a known issue which will be fixed in the next firmware version.

Regarding the SLAC issue:
This is strange, because the message in the wireshark can be seen (frame 144) right after the SLAC process . I wonder why it is not received in the FreeV2G application. Do run the application on a Windows host?

Regarding the parser:
I did not found any problems on first glance, but I will try to reproduce the bug and will come back to you if I have any news!

Thank you,
lho

Dalachowsky, 2023-02-21:
Yes, we are using a Windows host. Should we switch to Linux?

Thank you for the quick responses!

lho-stx, 2023-02-21:
Hey @Dalachowsky,

we had issues with some windows machines and timing of network packets. If you have access to a Linux host you can try to run the FreV2G application there, if the problem still persists there is another issue and we have to take a closer look.

Best regards,
lho

Dalachowsky, 2023-02-21:
I have a small question regarding port mirroring. If we connect Whitebeet through SPI to the host controller will it still work or do we have to connect it through ethernet? Because the documentation doesn't explicitly state whether mirroring to SPI is possible.
Thanks

lho-stx, 2023-02-21:
The port mirror will always use the ethernet interface, even when SPI is used as host controller interface. This is a good point, we will clarify this in the documentation!

Best regards,
lho

lho-stx, 2023-02-22:
Hey @Dalachowsky,

I was able to reproduce the bug. Adjustments in the Whitebeet firmware are needed. As a workaround you can try to charge without setting a charging profile when using DIN mode.

This will be fixed in the next Whitebeet firmware release.

Thank you for pointing out the bug!

Best regards,
lho

closed 2023-02-22 by lho-stx