Sevenstax/FreeV2G #154: After StartV2G, no response from the vehicle. ("We did not receive a frame before the timeout of 30s")

mckuss, 2023-02-21:
**Your Setup**
Platform: EVSE
Firmware Version: i.e. V02_00_01
Host Controller Interface: Ethernet
Host: FreeV2G EV_v1.0.6_1 + Own Implementation

**Describe the bug**
We did not receive a frame before the timeout of 30s

**To Reproduce**
customization.py includes the customized _handleNetworkEstablished() method.
[customization.py.txt](https://github.com/Sevenstax/FreeV2G/files/10797290/customization.py.txt)

**Expected behavior**
Expect a response to begin charging from vehicle, e.g. "Session started" received" printed to terminal.

**Logs**
Wireshark capture attached.
[230213_eTransit_EVacceptedChargeParameter_EVSEready_thenTimeout.pcapng.txt](https://github.com/Sevenstax/FreeV2G/files/10797293/230213_eTransit_EVacceptedChargeParameter_EVSEready_thenTimeout.pcapng.txt)

Since updating whitebeet to V2.00.01, uart output is corrupted - is there a new message rate from uart in the new version?

**Additional context**
Attempting to charge Ford eTransit

lho-stx, 2023-02-22:
Hey @mckuss,

yes the baud rate changed to 1000000, please try this baudrate and create an UART log. Furthermore, we need to have a look into your setup, because the frames shown in the wireshark log are not useful right now (packets missing).

1. Did you enable the port mirror feature? If not, please put evse.whitebeet.networkConfigSetPortMirrorState(1) in the line above evse.loop() in Application.py.
2. Are you connected via a network switch to your whitebeet? If yes, please connect directly to your Whitebeet, the network switch will not forward the packets from the port mirror.
3. In the Setup description you say you are using 

> FreeV2G EV_v1.0.6_1 + Own Implementation

 and later you say you are using an EVSE Whitebeet. I assume you made a mistake in the Setup description?
 
I kinldy ask you to check 1. + 2. and recreate the logs.

Best regards,
lho

crtg, 2023-02-22:
Hi,

I am facing exactly same problem, so decided not to open a separate issue. If this is a problem, feel free to move/delete this post.

**Your Setup**
Platform: Whitebeet EVSE + Whitebeet EV
Firmware Version: i.e. V02_00_01 and V01_00_06 
Host Controller Interface: Both boards connected to simple L2 ethernet switch which is connected to PC
Host: FreeV2G EV_v1.0.6_1 + EVSE_v2.0.1_4

**Describe the bug**
We did not receive a frame before the timeout of 30s

**To Reproduce**
I started two instances of Application.py from the same PC. Until I connect the EVSE and EV board everything looks fine. One board is waiting for the EV and the other one is waiting for EVSE. Once I connect the GND, CP, PE the EVSE side reports "We did not receive a frame before the timeout of 30s" error.

**Expected behavior**
Expect a response to begin charging from vehicle, e.g. "Session started" received" printed to terminal.

**Logs**
At the moment I have outputs from Application.py and EVSE UART output.
[evse_response.txt](https://github.com/Sevenstax/FreeV2G/files/10803657/evse_response.txt)
[ev_response.txt](https://github.com/Sevenstax/FreeV2G/files/10803658/ev_response.txt)

**Additional context**
After we received the WB from Codico we updated the boards to the latest version. We didn't change the PIB config file.

mckuss, 2023-02-22:
Hi Iho,
1. Yes, I have the port mirror enabled
2. Good to know about the ethernet switch, I will connect directly to the whitebeet. Do you suggest a hardwire tap for getting the wireshark traces this way?
My implementation is based on FreeV2G EV_v1.0.6_1, but with some small changes. I am overloading this function in a threaded classed based on Application.py, and have defined enums for configuration values for easier readability, but functionally my code is identical.
Will provide traces after re-running the test. 
Thank you!

lho-stx, 2023-02-22:
Hey @mckuss,

most likely directly connecting to your Whitebeet will be sufficient to create a wireshark log, now tap needed! Looking forward to your logs!

@crtg I have moved your issue (#156), but please also connected directly to your whitebeets (without a network switch in between), enable the port mirror of one of the Whitebeets and create a wireshark log of the interface connected to this Whitebeet. Everything else we will discuss in the other issue.

Best regards,
lho

lho-stx, 2023-02-27:
Hey @mckuss,

in your the log 230223_FordeTransit_slacok_v2gtimeout.pcapng.txt in frame 1325 the Whitebeets denies the TCP SYN of the EV with the ICMPv6 message "Destination unreachable". The EV asks for a TLS secured connection, but this fails, so there might be a problem with the certificates. Did you provide certificates to the EVSE Whitebeet (please see Whitebeet manual page 228)[https://raw.githubusercontent.com/Sevenstax/FreeV2G/EVSE_v2.0.1_4/SEVENSTAX_WhiteBeet_UserManual.pdf#page=228] ?

Maybe you can try to enable EIM only, because EIM does not need TLS for communication.

Best regards,
lho

lho-stx, 2023-02-27:
Hey @mckuss,

in your the log 230223_FordeTransit_slacok_v2gtimeout.pcapng.txt in frame 1325 the Whitebeets denies the TCP SYN of the EV with the ICMPv6 message "Destination unreachable". The EV asks for a TLS secured connection, but this fails, so there might be a problem with the certificates. Did you provide certificates to the EVSE Whitebeet [please see Whitebeet manual page 228](https://raw.githubusercontent.com/Sevenstax/FreeV2G/EVSE_v2.0.1_4/SEVENSTAX_WhiteBeet_UserManual.pdf#page=228)?

Maybe you can try to enable EIM only, because EIM does not need TLS for communication.

Best regards,
lho

mckuss, 2023-02-27:
We are indeed using EIM only, since we're using the included start CA certificate. Will follow up for creating a firmware update using PKI.

mckuss, 2023-03-02:
Ended up configuring SDP to unsecured TLS and we were able to establish comms with a colleages' Hyundai ioniq:

```python
        sdp_config = {'allow_unsecure': True,
                      'allow_secure': False,
                      'unsecure_port': 49153,
                      'secure_port': 49154,
        }
        self.whitebeet.v2gEvseSetSdpConfig(sdp_config)
```

In the process we did identify a small bug that was preventing us from operating unsecured TLS.
whitebeet.py, in method ```v2gEvseSetSdpConfig()```  Lines 1430, 1434, should look like:

```python
elif not (('unsecure_port' in sdp_config) and (isinstance(sdp_config['unsecure_port'], int)) and (sdp_config['unsecure_port'] in range(49152, 65535))):
```
and
```python
elif not (('secure_port' in sdp_config) and (isinstance(sdp_config['secure_port'], int)) and (sdp_config['secure_port'] in range(49152, 65535))):
```

We were getting an erronous error message about the port nuymber not being correct when we were setting allow_secure=False. This solves this error when the port is configured correctly (but disabled):
```python
sdp_config['secure_port']: 49154 | type(sdp_config['secure_port']): <class 'int'>
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/home/pi/git/py-beta-cubemaster/beta/cubemaster/main.py", line 130, in run
    self.loop()
  File "/home/pi/git/py-beta-cubemaster/.venv/lib/python3.9/site-packages/freev2g/Evse.py", line 632, in loop
    return self._handleEvConnected()
  File "/home/pi/git/py-beta-cubemaster/.venv/lib/python3.9/site-packages/freev2g/Evse.py", line 84, in _handleEvConnected
    self._handleNetworkEstablished()
  File "/home/pi/git/py-beta-cubemaster/beta/cubemaster/main.py", line 160, in _handleNetworkEstablished
    self.whitebeet.v2gEvseSetSdpConfig(sdp_config)
  File "/home/pi/git/py-beta-cubemaster/.venv/lib/python3.9/site-packages/freev2g/Whitebeet.py", line 1436, in v2gEvseSetSdpConfig
    raise ValueError("secure_port needs to be of type int with range 49152 to 65535")
ValueError: secure_port needs to be of type int with range 49152 to 65535
```

lho-stx, 2023-03-07:
Hey @mckuss,

good to hear that you proceeded. Thank you for the hint, we will have a look into this! If your setup is working now I close this issue!

Thank you,
lho

closed 2023-03-07 by lho-stx