Sevenstax/FreeV2G #155: Request: Ability to change interface from MII to RMII

Dalachowsky, 2023-02-22:
Hello,
In our implementation we want to connect Whitebeet directly to ethernet switch on our pcb. However Whitebeet uses MII interface and we already have another MCU that will be connected to MII interface of the switch. This is really problematic and we were wondering if you could implement some way to change ethernet interface of stm32 that is on the Whitebeet from MII to RMII since both of them are supported by that MCU.

lho-stx, 2023-02-22:
Hey @Dalachowsky,

I have looked into the sources to confirm, but actually the stm32 is configured to use the RMII mode, where did you find that Whitebeet uses MII mode? 

To switch the operation mode of the interface would be an interesting feature, but unfortunately hardware modifications are needed in order to use MII mode.

Thank you,
lho

Dalachowsky, 2023-02-23:
Oh I looked at the pinout in the datasheet and there are pins listed as "MII transmit data[0]" etc. I've assumed that if the RMII is not stated explicitly then it uses MII. Now i see that there are only 2 pins for Tx and 2 pins for Rx so that means it's RMII.
Thank you and sorry for the misunderstanding.

lho-stx, 2023-02-23:
Hey @Dalachowsky,

no problem, I'm glad that I can help you.

Best regards,
lho

closed 2023-02-23 by lho-stx