Sevenstax/FreeV2G #160: Control Pilot State Stuck at F for White-Beet PS 1.1

JPiruzza, 2023-03-13:
Hi, I have set up the CP line for EV mode for the White beet PS 1.1 but when I connect it to a charger, I am reading that the CP line voltage is negative (or in state F).

It seems like similar issues has been reported in the past (#126), but the solution relies on firmware updates that do not seem to be available for the SLAC/bridging board.

Are there plans to make this firmware update available to the PS board or do you know any solutions that can be done to fix this issue?

Thanks,
Josh

nststx, 2023-04-26:
Hi @JPiruzza,

Thanks for submitting your issue. We currently do not have an update for the SLAC firmware planned, however I will further discuss this internally. Regarding a solution to fix this issue, I will take a look at it and come back to you until the end of next week.

Best regards and thanks for your patience,
Niklas Stoffers

lho-stx, 2023-06-23:
Hey,

we are currently working on different issues and will come back to you as soon as possible!

Thank you for your patience,
lho

lho-stx, 2023-09-11:
Hey @JPiruzza,

unfortunately there is no update planned in the near future, but you can use a Whitebeet Carrier Board for this. Just do not enable the V2G and SLAC service, then it will behave like a SLAC bridging module.

Best regards,
lho

closed 2023-09-11 by lho-stx