Sevenstax/FreeV2G #161: Whitebeet slac/bridge runs customized code one time only

tareksr4, 2023-03-19:
i have a whitebeet slac/bridge module in runs on rasberryPI. when trying to operate the board and run send commands to it it work only one time, then it doesn't give any output and i'm not able to enter a charging session. also after this onetime operation that the board doesn't accept any commands through the ethernet.

nststx, 2023-04-17:
Hi @tareksr4,

thanks for submitting your issue. To help you with your problem I need further information. Could you please tell me which firmware version and platform (EV or EVSE) you're using? I will also need the logs from your Whitebeet and a wireshark capture of your session. To capture a session using wireshark please enable the port mirror by putting `evse.whitebeet.networkConfigSetPortMirrorState(1)` in the line above `evse.loop()` in *Application.py*.

Best Regards and thanks for your patience,
Niklas Stoffers

nststx, 2023-05-03:
Closing, due to inactivity. If this issue still persist, feel free to reopen it.

closed 2023-05-03 by nststx