Sevenstax/FreeV2G #162: Missing Documentation about Message Timeouts

st4hli, 2023-03-23:
Your Setup
Platform: EVSE
Firmware Version: i.e. V02_00_01
Host Controller Interface: Ethernet
Host: Own Implementation

Problem description:
It would be very helpful to have a clear documentation about all message timeouts between the EVSE Whitebeet and the host. It seems that the timeout description of  the examples in the documentation are not always correct (e.g. 500ms for Authorization Request - seems to be wrong).

Best regards,
st4hli

lho-stx, 2023-03-24:
Hey @st4hli,

I am not quite sure if I understand your issue well:

The timeout value in the Authorization Status Request message is a arbitrary value and can differ between different EVs. You can see different timeout values. The value give in the manual is just an example.

I apologize if I misunderstood your issue, I kindly ask you to elaborate it further?

Thank you,
lho

st4hli, 2023-04-03:
Hi @lho-stx,

I mean the timeouts for the messages between the EVSE host and the EVSE whitebeet. I doubt that all timeouts are set by the connected EV? I thought that the EVSE whitebeet itself defines certain timeouts for the messages it exchanges with its host.

BR,
st4hli

nststx, 2023-04-27:
Hi @st4hli,

Thanks for the further clarification. Take a look at the sequence diagrams in the "Application Examples" section in the Whitebeet User Manual. You should be able to find most if not all of the timeouts between WB EVSE and Host here. If that's not what you're looking for or you're missing a specific timeout, please let me know.

Best regards,
Niklas Stoffers

nststx, 2023-05-15:
Closing, due to inactivity. If the issue still persists or there are any questions left, feel free to reopen.

closed 2023-05-15 by nststx