Sevenstax/FreeV2G #167: Question: "ChargeParameterDiscoveryReq" parameter

ryofromjapan, 2023-04-13:
**Your Setup**
Platform: EVSE
Firmware Version: V02_00_01
Host Controller Interface: Ethernet
Host: Own Implementation

**Describe the bug**
"ChargeParameterDiscoveryReq" parameter below are not correct with the analyzed result.

- Energy Request
- Selected Energy Transfer Mode

**To Reproduce**
AC charge sequence is executed.

**Expected behavior**
I think that the below content should be correct.

Energy Request = 100[Wh]
Selected Energy Transfer Mode = 5(AC three phase)

**Logs**
Frame 274: 111 bytes on wire (888 bits), 111 bytes captured (888 bits) on interface ￥Device￥NPF_{8ADB4BE0-5E30-4FCA-B31D-EB9AD569150B}, id 0 Ethernet II, Src: 02:00:00:00:00:01 (02:00:00:00:00:01), Dst: 8Devices_34:a5:76 (c4:93:00:34:a5:76) Internet Protocol Version 6, Src: fe80::12, Dst: fe80::c693:ff:fe34:a576 Transmission Control Protocol, Src Port: 57967, Dst Port: 58840, Seq: 189, Ack: 140, Len: 37 V2G Transfer Protocol V2G Efficient XML Interchange
    V2G ISO1 Message
        Header
        Body
            ChargeParameterDiscoveryReq
                MaxEntriesSAScheduleTuple: 12
                [RequestedEnergyTransferMode: AC_three_phase_core (1)]
                AC_EVChargeParameter
                    [DepartureTime: 0]
                    EAmount
                        [Multiplier: 2]
                        [Unit: Wh (6)]
                        [Value: 1]
                    EVMaxVoltage
                        [Multiplier: 2]
                        [Unit: V (4)]
                        [Value: 5]
                   EVMaxCurrent
                        [Multiplier: 2]
                        [Unit: A (3)]
                        [Value: 5]
                    EVMinCurrent
                        [Multiplier: 0]
                        [Unit: A (3)]
                        [Value: 0]

Frame 276: 60 bytes on wire (480 bits), 60 bytes captured (480 bits) on interface ￥Device￥NPF_{8ADB4BE0-5E30-4FCA-B31D-EB9AD569150B}, id 0 Ethernet II, Src: 8Devices_34:a5:75 (c4:93:00:34:a5:75), Dst: TexasIns_00:1e:68 (a8:63:f2:00:1e:68) DEC DNA Routing Protocol DNA NSP message: Data segment continuation (0x00)
0000   a8 63 f2 00 1e 68 c4 93 00 34 a5 75 60 03 00 04
0010   00 1e c0 27 83 00 00 16 00 00 00 05 02 01 00 00
0020   00 00 05 02 00 00 00 00 00 00 00 00 00 00 ad c1
0030   00 00 00 00 00 00 00 00 00 00 00 00

c0          // Start
27          // Vehicle to Grid Service
83          // Energy Transfer Mode Selected
00
00 16       // Length=22[byte]
00          // Departure Time                   = none
00          // Energy Request                   = none
00 05 02    // Max. Voltage                     = 500[V]
01 
00 00 00    // Min. Current                     = 0[A]
00 05 02    // Max. Current                     = 500[A]
00          // Max. Power                       = none
00          // Selected Energy Transfer Mode    = DC core
00 00 00    // Energy Capacity                  = 0[Wh]
00          // Full SOC                         = none
00          // Bulk SOC                         = none
00          // Ready                            = False
00          // Error Code                       = No Error
00          // SOC                              = 0[%]
ad          // Checksum
c1          // End

**Additional context**
I apologize for posting so consectively from #164, but I'm not sure if this is a failed WB-CARRIER-BOARD (EVSE) behavior. Please let me know if my inquiry is not clear on some level, or perhaps if you have any question or concern.

Many thanks,

Ryo

lho-stx, 2023-04-14:
Hey Ryo,

you are welcome to post every question you have!

Can you please provide me the whole log file? This is needed to replicate your setup. If github is complaining that you can not upload .pcapng files, just append .txt to the file. Looking forward to your logs!

Thank you,
lho

ryofromjapan, 2023-04-14:
Hi Iho,

I will surely do that. Please give me a reasonable time. It feels like I am still not used to this place, but I will come back with the mentioned log file.

Have a nice weekend!

Ryo

lho-stx, 2023-04-14:
Hey Ryo,

You are doing good! Just by using the template you are helping us a lot!

Have a great weekend to!
lho

ryofromjapan, 2023-04-19:
Hi Iho,

Good day! I attach the log file for this inquiry. Would you kindly check about the content please?

Have a great day!

[20230419_EVSE_ChargeParameterDiscoveryReq.zip](https://github.com/Sevenstax/FreeV2G/files/11267633/20230419_EVSE_ChargeParameterDiscoveryReq.zip)

Ryo

lho-stx, 2023-04-24:
Hey Ryo,

I had a look into the log files and it seems the ChargeParameterDiscoveryReq send by the EV is malformed. I need to talk with a colleague about this. I will come back to you as soon as I have any news for you!

Thank you for your patience and best regards,
lho

nststx, 2023-07-03:
Hi @ryofromjapan,

Thanks for your patience and sorry for the late reply. I've taken a look at your issue today and this is actually a problem within the V02_00_01 firmware. However this issue has been fixed with the V02_01_00 firmware. Please try and update to the V02_01_00 firmware and let me know if that resolved your issue.

Best regards,
Niklas Stoffers

ryofromjapan, 2023-07-04:
Hi Niklas,

Thanks for the reply! It is duly noted that it is fixed with the V02_01_00 firmware, now.
However, we are unfortunately not able to have an environment where we can confirm that firmware, at the moment.
I apologize, but we currently don't know when we can confirm about if it's fixed.

I will come back here as soon as I hear anything new from my team. Have a nice day!

Many thanks,

Ryo

nststx, 2023-07-04:
Hi @ryofromjapan,

Thanks for coming back with that information. Take your time and let me know when you've got something new.

Best regards and have a nice day too,
Niklas Stoffers

nststx, 2023-08-02:
Hi @ryofromjapan,

I will close this issue for now due to organizational purposes. However when you are back working on it and still encounter your issue, please feel free to reopen!

Best regards,
Niklas Stoffers

nststx, 2023-08-02:
Closing due to temporary inactivity.

closed 2023-08-02 by nststx