Sevenstax/FreeV2G #169: SLAC Matching is failing if two Whitebeet module connected with Ethernet Switch

nikunjp26, 2023-04-13:
**Your Setup**
Platform:  EVSE
Firmware Version: i.e. V02_00_01
Host Controller Interface: Ethernet 
Host: Own Implementation

**Describe the bug**

We have integrated Whitebeet module running EVSE firmware version V02.00.01 with our CCS2 charger echo system and single gun echo system works very well. However, with the dual gun scenario we are facing couple of issues with some of the Vehicle. Same Vehicle works well if we connect only single Whitebeet module on the Ethernet.

Overall system setup is as per attached image.

[image]

Most of the time SLAC matching process is failing. Please find attached Wireshark traces for the same and please note that traces we have captured with port mirroring DISABLED.  

One thing we wanted to understand is even if Port mirroring is disabled, why SLAC matching packets are being routed on the ethernet interface? Image also attached for the specific packets from Wireshark captures. It appears that because of that routing, second Whitebeet module on the ethernet is also responding to the SLAC packets which is not directly connected to EV. In the traces you can see all SLAC matching packets are being rebroadcasted on the ethernet and multiple responses are observed. Is there any way to disable that?
[image]

- Please find the Attached log files which capture using the greenshark
- 
[SLAC_FAIL_MahindraXUV.zip](https://github.com/Sevenstax/FreeV2G/files/11223444/SLAC_FAIL_MahindraXUV.zip)

nikunjp26, 2023-04-17:
@nststx 

any update on this? as we are waiting for your reply.

nststx, 2023-04-17:
Hi @nikunjp26,

I tried to reproduce your issue but have yet failed to do so. If both Whitebeets have port mirror disabled I neither had any issues with SLAC matching nor was I able to see the SLAC messages sent to broadcast. However with port mirror enabled on the EVSE Whitebeet which is not physically connected to the EV I had the exact same issue you described here. 

What also struck me is message number 51 in your capture file. That message comes from communication between the STM32 and QCA700x. However you shouldn't be able to see those messages without port mirror being turned on.

What I currently suspect is happening is that port mirror is enabled on the Whitebeet which is not connected to the EV.  In general the Whitebeet acts as a tunnel between it's Ethernet and PLC interface for all messages that don't belong to it. With PLC communication being susceptible to electrical crosstalk this may result in the unconnected EVSE tunneling the PLC messages back to the Switch which probably results in timing constraints not being met for the ongoing SLAC process. However since the SLAC messages are broadcast messages this could only be the case if the port mirror is indeed enabled on the EVSE Whitebeet which isn't physically connected to the EV.

Could you do me a favor and double check that port mirror is disabled on **both** EVSE Whitebeets? Note that once you enable the port mirror it doesn't go back off until you either disable it manually again or reset the Whitebeet. It would also be interesting if the issue persists when you split your network into two separate VLANs (Host <-> WB1 and Host <-> WB2).

Best regards and thanks for your patience,
Niklas Stoffers

nststx, 2023-04-24:
Hi @nikunjp26,

I'm glad that the suggested workaround is solving your issue for now. Regarding the problem with port mirror and multiple Whitebeets connected to one switch, we are working on a solution for that and will fix it with the upcoming firmware release.

Best regards,
Niklas Stoffers

nststx, 2023-05-03:
Closing, as this issue is now tracked with our internal tracker.

closed 2023-05-03 by nststx

sametyazici, 2024-01-22:
> Hi @nikunjp26,
>
> I tried to reproduce your issue but have yet failed to do so. If both Whitebeets have port mirror disabled I neither had any issues with SLAC matching nor was I able to see the SLAC [...]

We are struggling with exact same problem. But VLANs does not work for us unfortunately.  Is there any right configuration setting for VLANs? We charging two seperate EVs with eth0 interface but one of them stop charging after. Port mirroring is closed too.