Sevenstax/FreeV2G #177: Charge session stops in Charge Parameter Discovery just after authentication

lb-rt, 2023-05-02:
**Setup**
Platform: EVSE
Firmware Version: V02_10_00
Host Controller Interface: SPI
Host: Own Implementation

**Describe the bug**
V2G Communication stops when Authorization Status is set to 0 (I don't even have the time to send the schedules).
Here it should wait for the schedules, if I try to send the schedules and the requested entries is 0, I have to send the Code to "Unable to deliver schedules" or I send anyways the schedules? In any case I can send both and I get an error in return, so the frame Set Schedules 0x27:0x6C seems that is never accepted.

**To Reproduce**
Here's my charge flow:

EV connected
Start SLAC matching
Set duty cycle to 5%
SLAC matching successful
Set V2G mode to EVSE
Start V2G (Start Listen sent)

V2G Session Started:
Protocol: DIN70121-2:2012
Session ID: EEF3BC19FC3BC653
EVCC ID: 000000050000

Selected Payment Method: External payment

Authorization Status Requested, timeout: 54999

-> Athorization status set to '0' (Authorization succeeded)

EV Energy Transfer Mode Selected:
Energy Request: 100.000000
Max Voltage: 500.00
Max Current: 50.00
Max Power: 25000
Selected Energy Transfer Mode: DC extended
Energy Capacity: 100
Full SoC: 99
Bulk SoC: 80
Ready: 1
Error Code: No Error
SoC: 0

Schedules requested: timeout = 54991, max entries = 0

EV DC Charge Parameters Changed:
Max Voltage: 500.00
Max Current: 50.00
Max Power: 25000
Ready: 1
Error Code: No Error
SoC: 0
Target Voltage: 0.00
Target Current: 0.00
Full SoC: 99
Bulk SoC: 80
Charging complete: 0

CP state changed: A

**Expected behavior**
It should wait for the schedules?

**Logs**
[log_bug_whitebeet_20230502.txt](https://github.com/Sevenstax/FreeV2G/files/11374656/log_bug_whitebeet_20230502.txt)

nststx, 2023-05-03:
Hi @lb-rt,

Thanks for submitting your issue. I will take a look at it in the upcoming days and come back to you until the end of next week.

Best regards and thanks for your patience,
Niklas Stoffers

lb-rt, 2023-05-04:
I can now correctly send the schedules and the cyclic 'Update DC Charging Parameters' frame, but the session closes anyways.
Seems like the same issue of ticket 130, https://github.com/Sevenstax/FreeV2G/issues/130

lb-rt, 2023-05-04:
In case here's the wireshark log
[20230504_EVSE_ChargeParameterDiscoveryReq.zip](https://github.com/Sevenstax/FreeV2G/files/11397713/20230504_EVSE_ChargeParameterDiscoveryReq.zip)

nststx, 2023-05-15:
Hi @lb-rt,

I've just taken a look at your logs. However I'm not sure if your Whitebeet log that you attached to your first message still matches the Wireshark log that you send in the last messages as you said that you were already able to go a step further in the mean time. Could you do me a favor and capture a new Whitebeet and Wireshark log just to make sure that both logs come from the same charging session?

Best regards,
Niklas Stoffers

lb-rt, 2023-05-17:
Hi,
It's not really the same charge but the result was quite the same, I can't do another test right now because my EV test equipment is not available until next week.
What I'm sure is that I had always these two messages at the end:

[  848.713] [CRIT  ] V2GLIB: ARRAY_SIZE_EXCEEDED: The number of elements in an subelement array is too large
[  848.831] [NOTICE] V2GCPS: Change State from B to A

But if I remember correctly the message just before was stating that the charge parameter discovery was processed.
I don't remember if for a brief moment it also went to C state or not, this I have to test it again.

lb-rt, 2023-06-01:
I put you again the logs of basically the same test as before:
[WB_parameterdiscovery.txt](https://github.com/Sevenstax/FreeV2G/files/11624593/WB_parameterdiscovery.txt)
[WB_parameterdiscovery.zip](https://github.com/Sevenstax/FreeV2G/files/11624595/WB_parameterdiscovery.zip)

Do you know what's wrong? There's maybe wrong data in what I send in the ChargeParameterDiscoveryRes ?

When I receive 0x27:0x84 I answer with 0x27:0x6C and the command is accepted, but then CP goes to A state

lho-stx, 2023-06-07:
Hey @lb-rt,

I suspect the framing message to set the schedules was received, but in the processing of the schedules there is a problem. Most likely we can sort the problem out when I have access to the specific framing message used to set the schedules! Can you provide a log that contains the framing messages? You are using SPI for communication, in [my comment](https://github.com/Sevenstax/FreeV2G/issues/176#issuecomment-1579119406 ) on #176 I shared the details of creating the SPI log with you.

I am looking forward to the SPI log!

Thank you.
lho

AlbertoGilardoni-eaton, 2023-06-19:
Hello,
So I have a SPI log, the log starts when the EVSE receives Session started, so 0x27:0x80.
Do you need more data?

I attached also the UART log.

[Session_2.zip](https://github.com/Sevenstax/FreeV2G/files/11787803/Session_2.zip)
[wb_log_uart.txt](https://github.com/Sevenstax/FreeV2G/files/11787805/wb_log_uart.txt)

lho-stx, 2023-06-23:
Hey,

we are currently working on different issues and will come back to you as soon as possible!

Thank you for your patience,
lho

AlbertoGilardoni-eaton, 2023-07-10:
Hello,
I tried to reconfigure the Peak Current Ripple and it worked, so yes the problem was the User Manual example.
I can finally continue with the charging process.
Thank you.
Alberto

closed 2023-07-11 by nststx