Sevenstax/FreeV2G #180: Stuck in user authorize stage and set key confirmation

nikunjp26, 2023-05-18:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host:  Own Implementation

We are doing testing of Sevenstax EVSE module(**02_01_00**) with Sevenstax EV Simulator (**V01_01_01**) Hardware revision 1.0.

We are facing the issue as per the below with attached logs

1) stuck at user authorize stage after "2023-05-17 17:21:31.114380" time in **diniso_3rd_slac_param_req** log file

2) CM_SET_KEY.CNF (Set Key Confirmation) at 2023-05-17 17:45:38.488201 in **v2g_config_error** in log file

[diniso_3rd_slac_param_req.zip](https://github.com/Sevenstax/FreeV2G/files/11508442/diniso_3rd_slac_param_req.zip)
[v2g_config_error.zip](https://github.com/Sevenstax/FreeV2G/files/11508445/v2g_config_error.zip)

Please provide us the support to solve this. feel free to contact if required more details.

lho-stx, 2023-05-23:
Hey @nikunjp26,
thank you for sending the logs, I will have a look into the files and comeback to you as soon as I have any news!
Best regards,
lho

lho-stx, 2023-05-26:
Hey @nikunjp26,

I checked the log files and have some questions:

In v2g_config_error the EV does not start V2G after the first SLAC matching. I do not know why and I would need the EV side of the communication to, in order to narrow done the problem. I also do not understand which CM_SET_KEY.CNF message you are referring to. Maybe you can give frame numbers instead of timestamps? This would help alot, thank you!

I also see no problem in the diniso_3rd_slac_param_req, there is a response to every authorization request and I also see charging sessions. Maybe you can also tell me the frame number?

In the User_Auth_Res_Not_Recevied log there is indeed something strange. The messages are all over the place, the EV does not continue when the EVSE sends the SECCDP response, but then it sends multiple TCP Syn. Later the EV continues to send SECCDP requests while the V2G session is running... This is impossible from the v2g state machine point of view. Do you have multiple simulators or whitebeets in the same network? Are the whitebeet and simulator connected to the same network (on the ethernet side)? If yes please use two seperated interfaces and try again.

Thank you,
lho

nikunjp26, 2023-05-29:
**Dear Iho,**

In v2g_config_error the EV does not start V2G after the first SLAC matching. I do not know why and I would need the EV side of the communication to, in order to narrow done the problem. I also do not understand which CM_SET_KEY.CNF message you are referring to. Maybe you can give frame numbers instead of timestamps? This would help alot, thank you!

**Ans** : Please ignore the CM_SET_KEY.CNF error for now. please help us to findout the solution of v2g_config_error because it is stuck point for us to clear the Certification test of our Final Product. let us know if need more logs of anything.

I also see no problem in the diniso_3rd_slac_param_req, there is a response to every authorization request and I also see charging sessions. Maybe you can also tell me the frame number?

**Ans** : In the logs there was the 1st session in which charging session was working fine and there was no any issue. you need to check the bottom logs of the files inwhich we got the error. 

In the User_Auth_Res_Not_Recevied log there is indeed something strange. The messages are all over the place, the EV does not continue when the EVSE sends the SECCDP response, but then it sends multiple TCP Syn. Later the EV continues to send SECCDP requests while the V2G session is running... This is impossible from the v2g state machine point of view. Do you have multiple simulators or whitebeets in the same network? Are the whitebeet and simulator connected to the same network (on the ethernet side)? If yes please use two seperated interfaces and try again.

Ans: yes we connected in same network and we enabled the ev simulator side mirrioring so this issue may happed. We will share the log again by connection as per your suggestion.

Thanks & best regards,
Nikunj Patel

nikunjp26, 2023-06-09:
Dear Iho,
waiting for your reply..

Thanks & Best Regards,
Nikunj Patel

lho-stx, 2023-06-09:
Hey @nikunjp26,

Regarding `v2g_config_error`:

> Ans : Please ignore the CM_SET_KEY.CNF error for now. please help us to findout the solution of v2g_config_error because it is stuck point for us to clear the Certification test of our Final Product. let us know if need more logs of anything.

Now I got your point. After successful SLAC matching the host tries a `V2G Service Set Mode: EVSE` (modID: 0x27, subID: 0x40). The Whitbeet answers with a `module busy` message. The following `V2G Service EVSE: Set configuration` messages are failing because the module is not in the correct state. You use your own host implementation? Try checking the response code of the answer of the Whitebeet and try again if it tells you `module busy`. An example of a failing setup are frame 634/635. In frame 745/746 the `V2G Service Set Mode: EVSE` is successful.

Regarding `diniso_3rd_slac_param_req.pcapng`:

> Ans : In the logs there was the 1st session in which charging session was working fine and there was no any issue. you need to check the bottom logs of the files inwhich we got the error.

There are mutliple charging sessions in this log (DIN and ISO). I double checked the log again and I am sorry but I do not see any problem with user authorization. I filtered the log file for every "Authorization Request" and there is always a successful "Authorization Response". I also filtered for "Contract Authentication Request" and there is always a successful "Contract Authentication Response". In both cases a charging sessions are taking place. 

Only in the last session (starting in frame 76792) there is no V2G communication because the EV does not send SECCDP after SLAC matching. Did you restart the simulator there? 

As you know the exact timestamp of the error, please tell me the frame number. Or send a log with the failing session only, maybe I just missed the error due to the size of the log file. 

Or did I not understand your issue well?

Regarding the `User_Auth_Fail_Log_3005.zip`:

> Ans: yes we connected in same network and we enabled the ev simulator side mirrioring so this issue may happed. We will share the log again by connection as per your suggestion.

Unfortunately I do not see any V2G messages in the simulator log, did you disable port mirror on the simulator? If yes please enable it again and rerun the tests. As an alternative you can enable the port mirror on the EVSE side. If you are using FreeV2G you need to place evse.whitebeet.networkConfigSetPortMirrorState(1) right before evse.loop() in Application.py. 

Nevertheless, it seems after SLAC was successful the EVSE waits for EV to send SECCDP messages, which are not seen in the log.

I understand your frustration, but please recreate the User_Auth_Fail_Log_3005 logs and review the diniso_3rd_slac_param_req.pcapng!

Thank you and looking forward to your updated logfiles,
lho

nikunjp26, 2023-06-12:
> Now I got your point. After successful SLAC matching the host tries a `V2G Service Set Mode: EVSE` (modID: 0x27, subID: 0x40). The Whitbeet answers with a `module busy` message. The following `V2G [...]
We will check and update after changes as per your suggestion. we are doing 10 retries with 500 ms now. so is it correct?

> Only in the last session (starting in frame 76792) there is no V2G communication because the EV does not send SECCDP after SLAC matching. Did you restart the simulator there?

No. we didn't restart simulator. why this issue is happen. 
We got CM_SLAC_MATCH.CNF reply in packet no.43 in "User_Auth_Fail_Log_3005" log file. as well as we got set mode confirm in packet no. 46. So why we are getting CM_SLAC_MATCH.REQ continuously from packet No. 53 again and again? FYI: at this stage our firmware is sending user authorization request and waiting for the user authorization confirm. 

> Nevertheless, it seems after SLAC was successful the EVSE waits for EV to send SECCDP messages, which are not seen in the log.

As per our second reply this is the real issue and we want to know why this happen. 

Thanks & Best Regards,
Nikunj Patel

lho-stx, 2023-06-12:
Hi @nikunjp26,

I would lower the time period to 400ms, because 500 ms is the timeout in the framing module after which the session terminates with a timeout error.

Regarding the CM_SLAC_MATCH.REQ, I need to replicate your setup with the simulator etc. and then have a closer look into this. Keep in mind this is an issue on the simulator side, not Whitebeet side.

I only have one question:

> FYI: at this stage our firmware is sending user authorization request and waiting for the user authorization confirm.

Are there frames missing in the log file? I see no user authorization request in the `User_Auth_Fail_Log_3005` log file and this would also not make any sense at this point, because there is no V2G session running.

Thank you,
lho

nikunjp26, 2023-06-12:
Dear Iho,
> Are there frames missing in the log file? I see no user authorization request in the `User_Auth_Fail_Log_3005` log file and this would also not make any sense at this point, because there is no V2G session running.

Apologies !. Here at this stage, we didn't requested for Authorization request. We will wait for USER Authorization confirmation for One min timeout. This stage is after Start Listening Successfully.

Thanks & Best Regards,
Nikunj Patel

lho-stx, 2023-06-23:
Hey,

we are currently working on different issues and will come back to you as soon as possible!

Thank you for your patience,
lho

lho-stx, 2023-06-26:
Hey @nikunjp26,

I have a clue what might be wrong. Can you please check which energy transfer mode in the Battery module is selected? Can you change the energy transfer mode to `DC: extended` and try again?

I rebuild your setup and observed the same issue you described: SLAC successful, but then no V2G session is taking place. 
The session was successful after I changed the energy transfer mode to `DC: extended` in the V2G simulator.

Thank you and best regards,
lho

nikunjp26, 2023-06-29:
Dear Iho,

Sorry for late reply.

EV Simulator has default mode is DC Extended and we didn't change it. 

So, every log file we provided to you is tested with DC Extended mode.

nststx, 2023-07-05:
Hi @nikunjp26,

I've taken a look at your issue and this is indeed a little bit strange. Looking at the EV simulator logs it seems that the EV never finishes SLAC matching (although it should). At least it's status flags indicate that. That's also why it starts sending the SLAC_PARAM_REQ again. To further investigate this issue could you please capture a UART log of the EV simulator?

Best regards,
Niklas Stoffers

nikunjp26, 2023-07-10:
Dear Niklas,

Please find the attached UART capture log file.
[EV_Simulator_UartLog2.zip](https://github.com/Sevenstax/FreeV2G/files/12000600/EV_Simulator_UartLog2.zip)

Best Regards,
Nikunj Patel

nststx, 2023-07-13:
Hi @nikunjp26,

Thanks for coming back with the logs. I will take a look at them in the upcoming days and come back to you once I found something.

Best regards,
Niklas Stoffers

nststx, 2023-07-14:
Hi @nikunjp26,

I've taken a look at your logs and it's all a little bit strange. It looks like the EV simulator actually finishes SLAC matching but detects a change to the PWM duty cycle to 100% shortly after that and thus stops the communication session. However that change should only happen if you set the duty cycle on the EVSE Whitebeet accordingly. Do you modify the duty cycle after or during SLAC on your EVSE Whitebeet?

Best regards,
Niklas Stoffers

nikunjp26, 2023-07-18:
Dear Niklas,

We are setting the pwm to 5% single time only. Then after we are not changing pwm.
Can you please help us to find the any frame in logs which indicate to changes in pwm so we can track and modify from our side?

Best regards
Nikunj Patel

nststx, 2023-07-18:
Hi @nikunjp26,

You can see the PWM change in the UART logs of the EV simulator. For example in line 409 or 560 "Duty cycle changed to 1000 permill" about 2-3 seconds after successfull SLAC matching. Do you maybe have some sort of timeout in your host application between SLAC matching and Session starting which ends up being a bit too short? If you have an oscilloscope you could also try if you're able to see the PWM change by yourself.

Best regards,
Niklas Stoffers

nikunjp26, 2023-07-19:
Dear Niklas,

apart from this can you please help for below points?

1. do you have any document for "how to run the freev2g python script" in windows system? right now we are checking with linux system only and we tried in windows system but not able to success.
2. can you please help to modify the freev2g script in which it will run in while loop for continious multiple charging session because rightnow we need to run script everytime for single session. if we stop forcefully the script using "cntl+c" then it will stop the session but next session also goes failed.

best regards,
nikunj patel

nststx, 2023-07-20:
Hi @nikunjp26,

> can you please elaborate more this sentence as we are not able to understand properly.?

I'm sorry if this wasn't clear. You use your own host implementation, right? What I meant was if in your host implementation you have timeouts after which you reset/stop the Whitebeet module. In this case especially between SLAC matching successful notification and Session started notification. This could explain why the EV simulator is detecting a change of the duty cycle. Or do you use FreeV2G here?
Were you able to check with an oscilloscope whether the duty cycle change is actually visible on the CP line? Otherwise it could be an issue with the change detection on the Simulator side, however I think this is much less likely to be the issue.

Regarding your second message:

1. Could you please specify which issues you're facing with FreeV2G on windows system? Normally it should work similar to Linux.
2. I think the easiest way here would be to create a batch/shell script (depending on which system you're using) which just continuously restarts FreeV2G (see example below). If you forcefully stop the script the Whitebeet may end up in some in-between state. That's why the next session may fail. In that case you would need to reset the Whitebeet.

File: *freev2g_loop.sh*

```sh
while true
do
    sudo python3 Application.py -m [mac_address] -r EVSE -i [interface] eth
done
```

Best regards,
Niklas Stoffers

nikunjp26, 2023-07-21:
Dear Niklas,

> I'm sorry if this wasn't clear. You use your own host implementation, right? What I meant was if in your host implementation you have timeouts after which you reset/stop the Whitebeet module. In [...]

Right now we are doing the retry of command again if incase timeout happen. so is it ok? or we need to start from the fresh?

For Windows i will again test and update you if need help because inpast i had tried but not success.

best regards,
Nikunj Patel

nststx, 2023-07-25:
Hi @nikunjp26,

> Right now we are doing the retry of command again if incase timeout happen. so is it ok? or we need to start from the fresh?

Yes this is alright. I thought you maybe had some sort of additional timeout if session is taking too long to start or something similar. 

I will need some more time to look into this. I'll let you know once I got an update for you. However in the meantime if you have the technical equipment to verify it would still be interesting whether the PWM change is actually visible on the CP line.

Best regards,
Niklas Stoffers

nikunjp26, 2023-07-25:
Yes Niklas,

i will check the PWM using CRO and update you soon.

Best Regards,
Nikunj Patel

nikunjp26, 2023-08-25:
Dear Niklas,
 We tried to reproduce the issue but now able to generate. Still our testing is on going.

Here is the video of one session. Does this session is correct? If you see the video for 1 second pwm generate then it goes to zero and then 2/3 second it becomes stable. This behaviour is correct or we need to modify that?

https://drive.google.com/file/d/15S-jjUjNpwyqMHWDLCaOGtqqwPcoUKdt/view?usp=drivesdk

Beat regards,
Nikunj Patel

lho-stx, 2023-09-04:
Hey @nikunjp26,

on first sight it looks ok, but let me try to capture the PWM timing by myself, so I can give you a wave file as reference.

Best regards,
lho

nikunjp26, 2023-09-25:
**Dear Iho,**

any update on this?

lho-stx, 2023-09-29:
Hey @nikunjp26,

yes. I did a test and my waveforms are also looking like yours! PWM is turned on and it stays at 5% continously!

It looks ok for me!

Best regards,
lho

closed 2023-09-29 by lho-stx