Sevenstax/FreeV2G #181: ABB Terra CJG 50 kW

svendriesen, 2023-05-25:
Setup: Whitebeet EV connected to an ABB Terra CJG 50 kW
Platform: PEV
Firmware Version: i.e. V01_00_04
Host Controller Interface: Ethernet
Host: FreeV2G (EV_v1.0.4_0)

**Describe the bug**
I am not sure if this should be a new issue or as an answer to the previous one, but the setup and problem is almost the same as #157 . The differences are the application output and some frames in the pcapng file. The application does not terminate, but gets stuck in the session starting state. I tried both DIN and ISO, but the output of the application remains the same. Is this a fault from Terra as in 157 or can I solve this with the Whitebeet? 

**Logs**
[Terra_53_CJG_DIN.pcapng.txt](https://github.com/Sevenstax/FreeV2G/files/11564296/Terra_53_CJG_DIN.pcapng.txt)
[Terra_53_CJG_ISO.pcapng.txt](https://github.com/Sevenstax/FreeV2G/files/11564297/Terra_53_CJG_ISO.pcapng.txt)
[image]

lho-stx, 2023-05-26:
Hey @svendriesen,

in your logs I do not see any messages from the charger. Did you enable the PortMirror feature? Also, is there any particular reason why you are using an old firmware version? If not please update to the latest firmware and try again.

Best regards,
lho

svendriesen, 2023-05-30:
Hey lho, 

Thanks for the fast response!
The problem is resolved in the meantime. The reason we didn't have a firmware update is because there was no business case related to our project, so the NDA procedure was stopped.

You can remove this issue if you wish.

Kind Regards,
Driesen Sven

lho-stx, 2023-06-02:
Hey @svendriesen,
thank you for your feedback!

If you have any more questions, please feel free to open another issue in the issue tracker!

Best regards,
lho

closed 2023-06-02 by lho-stx