Sevenstax/FreeV2G #182: Charging problem with battery level less than 76%

nigaru05, 2023-05-27:
Platform: WB-CARRIER-BOARD-EI as EVSE
Firmware Version: V02_00_01 PnC
Host Controller Interface: SPI
Host: FreeV2G (EVSE_v2.0.1_4)

I am working with a WB-CARRIER-BOARD-EI, the FreeV2G implementation is running on a Compute Module 3, which is connected to the WB by an SPI interface. The Compute Module control a 23KW Power Module by an CAN interface. Every time FreeV2G receives a 0x85 (DCChargeParametersChanged) message, we send that configuration to the Power Module. The first test was performed with the battery level above 80% and the charge worked correctly until I stopped it manually from the car. The second test was performed from 83% to 100% battery level and it worked correctly. But when I tried to charge it from 30%, it didn't work. After running many tests with different battery levels, so I established that when battery level is below 76%, the charge don't work.
[debug_egolf_bat74_dont_work.txt](https://github.com/Sevenstax/FreeV2G/files/11582934/debug_egolf_bat74_dont_work.txt)
As it can be seen in the log file, debug_egolf_bat74_dont_work.txt
Apparently, the car stop charging:
[   69.414] [NOTICE] V2GISO13: EV requested to stop charging.

The difference that I noticed is that when the battery is below 76% there is a significant difference in the output voltage of the Power Module and the voltage desired by the car. Each 0x85 message specifies EV target voltage: 360.0V, but with the battery at 74% the output voltage of the Power Module (the one seen by the car) is about 334V, when the battery is 76% or more the battery is above from 341V.
FreeV2G uses a function (calcEvsePresentVoltage) to calculate the "assumed" present voltage, the function uses some deltas. Wouldn't it be better to consult the Power Module for the current voltage???
def _calcEvsePresentVoltage(self):
"Calculates the present voltage at the current point in time, based on the given delta_u and the minimum and maximum values of the EVSE and EV."

I also started reading the ISO 15118 standard that specifies maximum voltage and current variations. But it is very extensive and I am just beginning.

On Monday I will try to make the captures with the mirror port.

- Attached log files with test at 80% (work) and 74% (don't work)
- Attach the log from debug port (J11 UART) of the Whitebeet.

[terminal_egolf_bat80.txt](https://github.com/Sevenstax/FreeV2G/files/11582935/terminal_egolf_bat80.txt)
[terminal_egolf_bat74_not_work.txt](https://github.com/Sevenstax/FreeV2G/files/11582936/terminal_egolf_bat74_not_work.txt)

I hope you can guide me to continue with my development.

Best regards.

lho-stx, 2023-06-07:
Hey @nigaru05,

first of all I apologize for the long delay in my answer. Then I would like to thank you for the detailed issue description!

To answer your questions:

Indeed, there are different voltages applied when using bulk charging and normal charging. The thresholds when bulk charging is used is different for every type of car/battery.

Please keep in mind that the FreeV2G application is a demonstration. If you want to use FreeV2G for actual charging you need to make changes to the code. FreeV2G "simulates" a charging controller, therefore there may differences in the values calculated by FreeV2G and actual physical values. My theory is, that the EV most likely does see this misalignment too and triggers an emergency shutdown. To confirm this I need to see the wireshark logs, looking forward to this!

You are free to modify the python code in order to use the actual physical values you obtain from your power electronics. You also may need to implement a real charging control loop that is acting according to standards and regulation.

Please send me your wireshark logs as soon as they are available!

Thank you,
lho

nigaru05, 2023-06-19:
Hey @lho-stx,

Thank you very much for your reply.
I've already been able to do full charges, replace getEvsePresentVoltage with the actual value.
We can end this problem.
But when trying to solve this, I realized that the port mirror is not working for me, at the beginning of the development I connected to the WB-CARRIER-BOARD by eth, the application and the port mirror worked, but now the connection to the WB it's over SPI and I'm not getting any output on the eth port.
This is why I can't send you the wireshark log. But it is very important to have the mirror working again to continue development.

I am looking forward to your response.

Thank you again.
Regards.

lho-stx, 2023-06-26:
Hey @nigaru05,

normaly there is no issue with using SPI as HCI and using the port mirror on the ETH port. Just for my sanity I tested this feture again and I see the V2G traffic when sniffing on the ETH port while using SPI as HCI. Do you have network gear (routers, switches) between your whitebeet and your PC? If yes, please connect the whitebeet directly to your PC and try again.

Thank you,
lho

nststx, 2023-07-19:
Closing, due to inactivity. If your issue still persists, feel free to reopen.