Sevenstax/FreeV2G #183: getting 0x11 error code in v2g_sub_set_configuration command.

nikunjp26, 2023-05-29:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: Own Implementation

We are doing testing of Sevenstax EVSE module(02_01_00) with Sevenstax EV Simulator (V01_01_01) Hardware revision 1.0.

During different testing we are getting 0x11 error code in v2g_sub_set_configuration command. We checked the updated user guide but not able to found the meaning of that code so please help us to get that or provided the updated guide if you have.

Attached the log file for reference and snap-shot of the document.

[image]

nikunjp26, 2023-05-29:
[User_Auth_Init_V2G_Config_Fail.zip](https://github.com/Sevenstax/FreeV2G/files/11592562/User_Auth_Init_V2G_Config_Fail.zip)

Please find the logs.

lho-stx, 2023-06-09:
Hey @nikunjp26,

this is the same as in #180 for the log `v2g_config_error.` In frame 111153 Whitebeet tells you that the module is busy, therefore the setup commands are failing with `0x11` (`Wrong state, not in the state to receive the command`). These are general error codes and can are found in 18.5 Error handling.

If you use your own implementation, try the `V2G Service Set Mode: EVSE` again later, until the module tells you '0x01: command accepted' in its response.

Closing this issue because it is a duplicate.

Thank you,
lho

closed 2023-06-09 by lho-stx