Sevenstax/FreeV2G #184: Session error: 23: Charge parameter timeout reached        Warning: Module did not accept command 27:6f, return code: 17

GovindMore, 2023-06-01:
**Your Setup**
Platform: EVSE
Firmware Version: i.e. V02_00_01 
Host Controller Interface: Ethernet 
Host: FreeV2G (FreeV2G EVSE_v2.0.1_4) 

**Describe the bug**
I connected this upgraded Whitebeet with Tata EV (Tiago) and then took logs. After resetting the WB, we got expected output for 2 to 3 times then after that we were facing issue.

"Session Error" received
Session error: 23: Charge parameter timeout reached
Warning: Module did not accept command 27:6f, return code: 17

**To Reproduce**
Modifications done in the code default “FreeV2G EVSE_v2.0.1_4” before taking the logs:
1] Modified the code as per “Fix frame filtering #175” issue 
( changes in two files 1)Framinginterface.py 2)Whitebeet.py )

2] Also disabled the DIN by following code changes:
        self.evse_config = {
            "evse_id_DIN": '+49*123*456*789',
            "evse_id_ISO": 'DE*A23*E45B*78C',
            "protocol": [1], 			# only one protocol ‘ISO’ is passed
            "payment_method": [0],
            "energy_transfer_mode": [0, 1, 2, 3],
            "certificate_installation_support": False,
            "certificate_update_support": False,
        }

3] In Whitebeet.py,
evseIdIso = self.payloadReaderReadBytes(lenEvseIdDin) 
 line changed to 
evseIdIso = self.payloadReaderReadBytes(lenEvseIdIso) 

I connected this upgraded Whitebeet with Tata EV (Tiago) and then took logs. After resetting the WB, we got expected output for 2 to 3 times then after that we were facing issue. 

**Logs**
I have connected WB directly using Ethernet with laptop (without switch)
I have also set port mirror state to 1.
I have attached the wireshark logs and python output logs for both correct and faulty(with issue).
[correct_2_0_1_4__wireshark.txt](https://github.com/Sevenstax/FreeV2G/files/11625681/correct_2_0_1_4__wireshark.txt)
[faulty_2_0_1_4_wireshark.pcapng.txt](https://github.com/Sevenstax/FreeV2G/files/11625686/faulty_2_0_1_4_wireshark.pcapng.txt)
[correct logs.txt](https://github.com/Sevenstax/FreeV2G/files/11625694/correct.logs.txt)

[faulty logs.txt](https://github.com/Sevenstax/FreeV2G/files/11625695/faulty.logs.txt)

**Additional context**
Default WB(EVSE) version was V01_01_07.
Upgrade from V01_01_07  to  V02_00_01 
I have attached the upgrade process logs in “WB_firmware_upgrade_log”

[WB_firmware_upgrade_log.docx](https://github.com/Sevenstax/FreeV2G/files/11625574/WB_firmware_upgrade_log.docx)

GovindMore, 2023-06-07:
Waiting for the response. If some more information needed, please let me know.

lho-stx, 2023-06-07:
Hey @GovindMore,

Thank you for using the template, providing decent logs and additional information!

I had a look into your logs and them main problem is that the PowerDeliveryResponse is not send in time (Therefore the `Session error: 23: Charge parameter timeout reached`). The `Warning: Module did not accept command 27:6f, return code: 17` occurs, because the modul is already in stopped state.

One difference between the two logs are, that the EV does a router solicitation between the PowerDeliveryRequest and PowerDeliveryResponse. I need to look in detail what is going on there.

In the meantime, can you create other faulty logs, so I can see if the problem is related to the router solicitation at all? Also can you send me the UART Debug prints of the Whitebeet module of such sessions?

Thank you,
lho

lho-stx, 2023-06-08:
Hey @GovindMore,

I had another look into the logs and found something I missed yesterday:

The timeout occurs, because your host controller does not respond in time. If you compare the timestamp of frame 489 (EVSE: Update DC charging parameters) and 492 (TCP FIN) or 493 (EVSE: Session Error) you'll see that you are above the 500ms performance timeout for communication between host and Whitebeet. 

In the latest firmware a new framing command is added: `Set Session Parameter Timeout`. You can set the performance timeout for communication between host and Whitebeet with this command. Please be aware that this is intended for testing purposes! For more information please see the latest Whitebeet manual, chapter 18.10.21

Best regards,
lho

GovindMore, 2023-06-09:
Hi @lho-stx,
Thank you for the response. As you suggested I upgraded the WB Firmware Version to V02_01_00 and used code EVSE_v2.1.0_0.
I used this default code as it is and only added two lines given below just before evse.loop()
            evse.whitebeet.networkConfigSetPortMirrorState(1)
            evse.whitebeet.v2gEvseSetSessionParameterTimeout(1000)

I got three errors and I have attached logs for the same:
1]code stuck at v2g start- 
[fault_start_v2g_python.docx](https://github.com/Sevenstax/FreeV2G/files/11704248/fault_start_v2g_python.docx)
[fault_start_v2g_usb.docx](https://github.com/Sevenstax/FreeV2G/files/11704250/fault_start_v2g_usb.docx)
[fault_start_v2g_wireshark.txt](https://github.com/Sevenstax/FreeV2G/files/11704252/fault_start_v2g_wireshark.txt)
2]Session error: 23: Charge parameter timeout reached
[fault3_pyhon.docx](https://github.com/Sevenstax/FreeV2G/files/11704272/fault3_pyhon.docx)
[fault3_usb.docx](https://github.com/Sevenstax/FreeV2G/files/11704274/fault3_usb.docx)
[fault3_wireshark.txt](https://github.com/Sevenstax/FreeV2G/files/11704275/fault3_wireshark.txt)
3]two warnings at a time
[fault2_python.docx](https://github.com/Sevenstax/FreeV2G/files/11704281/fault2_python.docx)
[fault2_usb.docx](https://github.com/Sevenstax/FreeV2G/files/11704283/fault2_usb.docx)
[fault2_wireshark_logs.txt](https://github.com/Sevenstax/FreeV2G/files/11704292/fault2_wireshark_logs.txt)

Thank you,
Govind

pubaek, 2023-06-09:
@GovindMore

lho-stx, 2023-06-12:
Hey @GovindMore,

I inspected your logs and it seems there are frames missing. There are ack of unseen TCP packets, jumping sequence numbers etc. Can you please give me a brief overview of your setup?

Nevertheless, in the log file `fault3_wireshark.txt`, there is still a timeout. Packet 284 `subID: 0x63 (EVSE: Update DC charging parameters)` timesout in packet 306 `subID: 0x8e (EVSE: Session Error)`.

I kindly ask you to recheck your host application timing and the logs, if there are indeed any missing frames.

Thank you,
lho

GovindMore, 2023-06-13:
Hello @lho-stx,

I have connected Laptop directly with WB through ethernet. So, host application is running on Laptop.
Did you get anything related to the 'fault_startv2g' and 'fault2' issues?  is reason behind them same as that of fault3? please give some more detail.

Thank you,
Govind

GovindMore, 2023-06-18:
Hello @lho-stx ,

I changed the host and checked again. Now the frequency of timeouts has been drastically reduced, and it is working fine. So, for now, I am closing the issue. If there are any further issues, I will let you know. Thanks for your support.

Regards,
Govind

closed 2023-06-18 by GovindMore