Sevenstax/FreeV2G #185: Slave is Responding to Size Exchange Frames in an Unexpected Way

keenanjohnson, 2023-06-03:
**Your Setup**
Platform: PEV / EVSE
Host Controller Interface: SPI
Host: Custom C Implementation

**Describe the bug**
Hello! I am trying to implement a client for the WhiteBeet PI module using the White Beet Dev board.

I am seeing some very weird behavior that doesn't seem to match the software implementation guide.

In the software guide, it shows  an example size exchange frame as follows. The slave module responds with AAh, AAh, 00h, 00h.

[image]

However in my testing, the slave module responds slightly differently. Below is a capture from my logic analyzer.

My host implementation sends a Size Exchange frame as I desire with a size of 8 (AAh, AAh, 00h, 08h). 

Confusingly, the slave responds with a size exchange frame that also lists a size of 8 instead of 00 as expected in the guide.

Then when I attempt to send the frame to request the firmware version: uint8_t get_firmware_bytes[] = {0x55, 0x55, 0x00, 0x00, 0xC0, 0x10, 0x41, 0x01, 0x00, 0x00, 0x2B, 0xC1}; the slave responds with some seemingly random data byes of it's own.

[image]

Can anyone help shed some light on what I am doing wrong? Is there a reference C or C++ implementation I could take a look at?

Channels in the Logic Analyzer Capture Above:

0 - Slave Select (CS
1 -  CLK
2 - MOSI
3 - MISO
4 - RX Ready
5 - TX Pending

keenanjohnson, 2023-06-03:
The software usermanual implies that the slave should just respond with zeros to a data-exchange frame after the header: 
[image]

keenanjohnson, 2023-06-03:
The dev board I am using: https://www.codico.com/en/wb-carrier-board-pi-1-1-pev-embedded-iso15118-sw-stack

lho-stx, 2023-06-05:
Hey @keenanjohnson,

W need to have a closer look into this. Can you provide me the SPI logs? Preferably in a Saleae Logic2 or .csv format?

In the meantime, please have a look into #150, there are more details about the SPI protocol. Unfortunately there is no C/C++ reference implementation at the moment.

Best regards,
lho

keenanjohnson, 2023-06-05:
Thank you! Yes here is a csv export of the capture I screenshotted above.

[get_firmware.csv](https://github.com/Sevenstax/FreeV2G/files/11654239/get_firmware.csv)

[Attached is a log from a DSLogic Logic Analyzer](https://www.dreamsourcelab.com/product/dslogic-series/)

Channels in the Logic Analyzer Capture Above:

0 - Slave Select (CS
1 - CLK
2 - MOSI
3 - MISO
4 - RX Ready
5 - TX Pending

lho-stx, 2023-06-09:
Hey @keenanjohnson,

I apologize for being unprecise: I need the CSV file with the high level SPI analyzer already applied. Is it possible to provide me such a log file?

Thank you,
lho

lho-stx, 2023-06-20:
Hey @keenanjohnson,

do you keep the SPI timing in mind? You need to wait 20us between asserting of CS and the transmission of the payload.

Nevertheless, I kindly ask you to send your .dsl file? The .csv is really hard to work with and gather information from...

Thank you,
lho

keenanjohnson, 2023-06-26:
Hello @lho-stx !

Thank you for the response. I believe I have sorted out my issue. The issue was there was a bug in the SPI transfer function of my vendor BSP that was toggling the CS line after each four bytes.

I have now created an initial implementation in C++ and things seem to be working. I have attached a zip file below containing a saleae logic analyzer log file of the exchange for anyone in the future.

[get_fw_ver_6_26_2023.zip](https://github.com/Sevenstax/FreeV2G/files/11875361/get_fw_ver_6_26_2023.zip)

I am reading out the firmware version as V01_00_04 in the capture above.

One last questions before I close this out. I tried reading out the serial number using the 0x47 Get Serial Number Sub ID and I get an unknown Sub ID response. Any idea why that might be?

I have attached a second logic analyzer log below showing that behavior.

[get_serial_number_6_26_2023.zip](https://github.com/Sevenstax/FreeV2G/files/11875420/get_serial_number_6_26_2023.zip)

lho-stx, 2023-06-29:
Hey @keenanjohnson,
thank you for sharing the logs!

First of all you are sending your data exchange frame on the MOSI line again, when you are reading the data from whitebeet.

[image]

This is no problem, because you set the length in the size exchange frame to zero, I just wanted to give you a hint!

May I ask you where you read an unkown Sub ID response? In the log you have send me there is a valid answer for the Get Serial Number. It is just set to 00:

[image]

Please keep in mind that you are using an old firmware version. If it is possible please update your board.
**IMPORTANT NOTE:** Do **NOT** update to **v1.0.6**, because there is an issue that will lock your whitebeet to this FW version. There will be a new FW available soon.

Thank you,
lho

keenanjohnson, 2023-06-29:
Thanks @lho-stx ! I was confused as the serial number is not zero on the sticker on my unit. 

Is that serial number not set at the factory I assume? 

I saw the note on the bug related to the v1.0.6 and will wait for the update.

nststx, 2023-07-03:
Hi @keenanjohnson,

You're right. The serial number is not set at the factory. Thus the serial number being empty is expected behavior. If you want you can set the serial number yourself by using the Framing API (chapter 11.3.4 in Whitebeet User Manual). However this will only change the serial number in the user configuration. Thus once the board is reset to factory config, the serial number will be empty again.

I hope we were able to help with your issue. If you consider it resolved please close this issue. If you have any questions left, feel free to ask!

Best regards,
Niklas Stoffers

nststx, 2023-07-19:
Closing as this issue is considered to be resolved. If the issue still persists, feel free to reopen!

closed 2023-07-19 by nststx