Sevenstax/FreeV2G #188: Charging sequence stops after first self.state loop

Vickkram, 2023-06-09:
Hello,

I have been trying to connect WB-CARRIER-BOARD-PI 1.1(EV) to a charger. The charging sequence seems to stop after the first self.state if loop. I have attached the wireshark logs below.

log (EV)
WHITE-beet-PI firmware version: V01_00_06
Set the CP mode to EV
Start the CP service
Set the CP state to State B
Set SLAC to EV mode
Start SLAC
Wait until an EVSE connects
EVSE connected
Start SLAC matching
:waiting for SLAC matched
128
SLAC matching successful
Set V2G mode to EV
Set V2G configuration
Set DC charging parameters
Start V2G
Create new charging session
ControlPilot duty cycle: 8.5
sessionStarting
sessionStarting
"Session Stopped" received
Change State to State B
EV loop finished
Goodbye!

[Failed_Session.zip](https://github.com/Sevenstax/FreeV2G/files/11709211/Failed_Session.zip)

nststx, 2023-06-12:
Hi @Vickkram,

Thanks for submitting your issue. It seems like your V2G session setup fails somewhere between the SessionSetup and PaymentServiceSelection. I've taken a look at the Wireshark log, however I'm missing the actual messages from the V2G communication. Could you please capture another Wireshark log with port mirror enabled on the board? To capture a log with port mirror please put `ev.whitebeet.networkConfigSetPortMirrorState(1)` before `ev.loop()` in the *Application.py* file.

Best regards,
Niklas Stoffers

Vickkram, 2023-06-14:
Sorry for the late reply, I was out of office for a bit. Here is the wireshark with port mirroring enabled:

[portmirroron.zip](https://github.com/Sevenstax/FreeV2G/files/11748564/portmirroron.zip)

nststx, 2023-06-15:
Hi @Vickkram,

Thanks for coming back with the logs. Unfortunately the EVSE part of the communication and packet timestamps are still missing. Do you maybe have a switch between your host and the Whitebeet? If that's the case could you please capture another log with your host being directly connected to the Whitebeet? Otherwise it's a bit difficult to tell what the issue is. 
However looking at your logs it currently seems that the communication already fails at the SECC discovery as the EV doesn't try to open a TCP connection between sending it's SECCDP request and the SessionStopped notification.

Best regards,
Niklas Stoffers

Vickkram, 2023-06-16:
Hello again,

Thank you for responding, I have attached the requested wireshark logs below.

[06-16-23.zip](https://github.com/Sevenstax/FreeV2G/files/11774231/06-16-23.zip)

lho-stx, 2023-06-23:
Hey,

we are currently working on different issues and will come back to you as soon as possible!

Thank you for your patience,
lho

nststx, 2023-06-26:
Hi @Vickkram,

I've taken a look at your logs. This issue seems to be a duplicate of #178. The session setup fails because the EVSE is responding with a response code "OK" in the SessionSetupResponse which doesn't adhere to the DIN70121 specification. However our handling of that response code is also incorrect. Instead of accepting this erronous response code we reject it and terminate the TCP connection. We will fix this in the upcoming EV firmware update.

Best regards,
Niklas Stoffers

closed 2023-06-26 by nststx