Sevenstax/FreeV2G #189: Facing Issue in downgrading WB EV firmware version

GovindMore, 2023-06-19:
**Your Setup**
Platform: EV 
Firmware Version: i.e. V01_00_06
Host Controller Interface: Ethernet 

**Describe the bug**
I had WB EV with default version V01_00_04. I successfully upgraded it to V01_00_06. But due to some reason now I want to downgrade it to V01_00_05. but I am facing issue while downgrading please give some idea how to solve this.  For upgrade and downgrade I am using latest tool "SEVENSTAX FWU Tool 2.6".  

**Logs**
StxFwUpdater_FWUtool>FWUtool.exe -f WHITEBEET_ISO15118_EV_V01_00_05.fwu -t c4:93:00:34:d0:a7 -i Ethernet
Welcome to the SEVENSTAX FWU Tool 2.6

Initiating framing interface
Starting update process
Firmware update at 0 percent; Sent: 833 Bytes
[!] Error: Device rejected update file

lho-stx, 2023-06-23:
Hey,

we are currently working on different issues and will come back to you as soon as possible!

Thank you for your patience,
lho

nststx, 2023-06-27:
Hi @GovindMore,

Thanks for submitting your issue. We are currently looking into it. I'll let you know once I got an update.

Best regards,
Niklas Stoffers

nststx, 2023-06-27:
Hi @GovindMore,

We were able to reproduce your issue and have indeed found issues with the firmware update file for EV firmware version 01_00_06. We're currently investigating potential solutions and will let you know once we got an update. In the meantime please do not use the V01_00_06 firmware update file anymore.

Best regards,
Niklas Stoffers

GovindMore, 2023-06-27:
Hi @nststx,
Thank you for the response. Please provide solution as early as possible.

Regards,
Govind MORE

nststx, 2023-07-18:
Closing, as issue should be resolved by board recall.