Sevenstax/FreeV2G #190: Question:

rwlyra, 2023-06-26:
18.12.2 Payment Selected
In Whitebeet user manual 2.16 the payment selected structure contains a note in the description field that is confusing.
"Note: The following parameters are only **present** if the
payment method contract payment was used."

This  implies it is an optional data parameter.

This goes against the normal mechanism used for 'optional' parameters since the the data type indicated and the example given show it as a non-optional parameter. So either it needs a third option in the description as 'Not applicable' (or something along those lines) or you need to redefine it to be uint8[0-1] and you need to allocate up to 2 bytes for it in the example message. (Or you need to get rid of the optional note!)

rwlyra, 2023-06-26:
So I now realise that the statement is not referring to the parameters in that description box, rather the subsequent parameters, so I don't think it's an issue, although I am still a bit confused by the syntax for parameter types. 

eg, For EMAID string[14-15] 

for previous parameters in other messages, the syntax [0 - x ] seems to indicate that the parameter is 'optional' where the first byte represents how many subsequent bytes make up that parameter. i.e. if the first byte is 0 then there won't be any subsequent bytes and the parameter can be ignored.

In this case [14-15] should surely imply it is a **non**-optional parameter either 14 or 15 bytes long... shouldn't it be  [0 - 15]
And then the first byte is always 14 or 15 (or 0)

I am confused by the syntax, and can't find a clear description of the message breakdown for these data types.

nststx, 2023-06-27:
Hi @rwlyra,

Thanks for submitting your question. Regarding the note for section 18.12.2 in the manual: As you've already guessed, that note applies to the parameters following the "Selected Payment Method" parameter. Thus you should only send them when the selected payment method is set to contract payment. 

Generally a type followed by square brackets can mean two different things. In the case of `type[0-1]` it means that the parameter is optional and can either be omitted (length 0) or be there exactly once (length 1). In the case of `type[X-Y]` (X >= 0, Y >= X, Y > 1) the parameter is either an array (for example `uint8[0-100]`) or the bounds for the length of a string. In your case `EMAID string[14-15]` just means that the `EMAID` parameter is **not** optional and has to be between 14 or 15 bytes long. Respectively `EMAID string[0-15]` would mean that the `EMAID` string has a length somewhere between 0 and 15 bytes. For optional parameters you always have one byte indicating whether the parameter is present or not. For arrays or strings where `X = Y` you **do not** have a byte for the length. The first byte will be the first value of the array/string. For arrays or strings where `X != Y` you will have either one byte (for `Y < 256`) or two bytes (for `Y >= 256`) for the length. 

This is indeed a little bit confusing and needs further clarification in the User Manual. We will update the User Manual regarding that information.

I hope I could clarify your questions. If you're still not sure about something, feel free to ask!

Best regards,
Niklas Stoffers

nststx, 2023-08-02:
Closing as this issue is fixed with the new UserManual release (v.2.17)

closed 2023-08-02 by nststx