Sevenstax/FreeV2G #191: Question: User manual 2.16  Energy Transfer Mode example is wrong.

rwlyra, 2023-06-26:
18.12.4 Energy Transfer Mode Selected (User manual 2.16)

I think the example given here is taken from the old spec before the V2G interface completely changed.
Although the example bytes given at the top of p165 use the Sub-ID 0x83 which is correct for Energy Transfer Mode, I think they have been leftover from user manual  2.11.2 before someone thought it was a good idea to reuse the same sub-ids for completely  different messages (It used to be Status message for requesting Authorization Status)

This should be corrected.

nststx, 2023-06-27:
Hi @rwlyra,

Thanks for submitting your issue. The example for the "Energy Transfer Mode Selected" message is indeed incorrect.
We will fix this with an upcoming version of the User Manual.

Best regards,
Niklas Stoffers

nststx, 2023-08-02:
Closing as this issue is fixed with the new UserManual release (v.2.17)

closed 2023-08-02 by nststx