Sevenstax/FreeV2G #194: ARRAY_SIZE_EXCEEDED and timeout before cable check request (after charge parameter discovery).

rwlyra, 2023-06-27:
**Your Setup**
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface:  SPI
Host:  Own Implementation

**Describe the bug**
I'm not 100% sure it's a bug to be honest. It might be something I'm doing wrong!
I have a setup using Whitebeet Eval board with V02_00_01 firmware acting as EVSE connected to Sevenstax EV simulator and controlled via SPI. I run into an unexpected timeout after successfully progressing through SLAC matching, 
It seems to timeout shortly after charge parameter discovery, and I'm not sure why.

**To Reproduce**
I have included a full log of SPI signals from Saleae Logic2 logic analyser. 

**Expected behavior**
I was expecting to receive  a cable check request from the Whitebeet module, but get the timeout instead. 

**Logs**

07_EV_SIM_27_06_23.txt is a log from the serial port of the EV simulator board.
07_EVSE_SIM_27_06_23.txt is a log from the serial port of the Whitebeet Eval board.
07_charging_session_stopped_27_06_23.sal (zipped for github) is a log of SPI signals from Saleae Logic2 logic analyser.
07_DecodedSPI_messages_27_06_23.docx is our decoded breakdown of the SPI messages from the Whitebeet board.

**Additional context**
I noticed in line 106 in the EVSE_SIM log from the Eval board the following message: 
**_[   65.514] [CRIT  ] V2GLIB: ARRAY_SIZE_EXCEEDED: The number of elements in an subelement array is too large_**
and shortly after that an error is sent to our host controller and the session is terminated.
[07_EV_SIM_27_06_23.txt](https://github.com/Sevenstax/FreeV2G/files/11884781/07_EV_SIM_27_06_23.txt)
[07_EVSE_SIM_27_06_23.txt](https://github.com/Sevenstax/FreeV2G/files/11884783/07_EVSE_SIM_27_06_23.txt)
[07_DecodedSPI_messages_27_06_23.docx](https://github.com/Sevenstax/FreeV2G/files/11884786/07_DecodedSPI_messages_27_06_23.docx)
[07_charging_session_stopped_27_06_23.zip](https://github.com/Sevenstax/FreeV2G/files/11884803/07_charging_session_stopped_27_06_23.zip)

nststx, 2023-06-28:
Hi @rwlyra,

Thanks for your detailed issue submission. I've taken a look at your logs and found an issue with your "Set DC charging parameters message" (frame 28). In there you set the "Peak Current Ripple" to `[0x00, 0x05, 0x81]` which corresponds to the exponential value *5x10^-127* (exponent is an sint8 type, thus twos complement of 0x81 = -127). We only allow the exponent to be in range `[-3, 3]`. I guess that you probably wanted to set it *0.05* which would be `[0x00, 0x05, 0xFE]`. The `ARRAY_SIZE_EXCEEDED` log may be a bit misleading here because that is also used when exponential types are out of bounds (this log comes from EXI serialization during the V2G session).

However there's also an issue on our side. If you take a look at the response to the "Set DC charging parameters message", the Whitebeet responds with the code "Command accepted". This is clearly a mistake as we already know at that point that the parameter is out of bounds and thus should respond with an erronous response code. We will fix this with upcoming firmware versions.

Please let me know if you're able to proceed further when setting the "Peak Current Ripple" to an exponential value inside bounds. If you have any questions left, feel free to ask!

Best regards,
Niklas Stoffers

rwlyra, 2023-06-28:
Brilliant - thanks for the rapid reply! I'll test with the new parameters, and let you know.

rwlyra, 2023-06-28:
Hi Niklas, 

That change worked, and now I can progress with the charging process - many thanks!

One thing I should mention is that the Whitebeet User Manual (v2.16)  uses those values in it's example  
**_00 05 81 (Peak Current Ripple 0.5A) p140_**, so that should probably be changed in future releases?

Also, does that mean for all other 'exponential' values that are used they are also using the same two's complement mechanism? (I think I've 'got away' with it so far, because most of the values I've used have been single figures).

Anyhow, thanks again for the support!
Rob

nststx, 2023-06-29:
Hi @rwlyra,

I'm glad to hear that the proposed solution worked for you. 

Thanks for pointing out the incorrect example. I didn't notice that myself. We definitely need to fix that as this is probably a big source of confusion. I'll add it to our internal tracker so we can fix it with an upcoming release.

Regarding your question: Yes, since the exponential type always has an exponent of type sint8 we use the two's complement representation for it. However because most of the exponential values have an exponent greater than or equal to 0 you probably didn't have any issues with it so far since the two's complement representation of 0 to 127 is basically the same as "normal" binary representation.

I'm glad that we were able to fix your issue. If you have any questions left, feel free to ask!

Best regards,
Niklas Stoffers

nststx, 2023-08-02:
Closing as this issue is fixed with the new Whitebeet User Manual (v.2.17)

closed 2023-08-02 by nststx