Sevenstax/FreeV2G #196: Socket Communication close after the payment Service selection Response

nikunjp26, 2023-07-06:
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface: Ethernet
Host: Own Implementation

We are trying to charge the Nextion XUV car with our charger but communication is always stop after the payment Service selection Response. please find the attached Wireshark logs (packet No. 945,1195,1629, etc...). we tried with single GUN and Dual GUN charger but same issue facing with this CAR.

[tata_nexon_issue.zip](https://github.com/Sevenstax/FreeV2G/files/11967245/tata_nexon_issue.zip)

Please support us to identify the issue.

Thanks & Best Regards,
Nikunj Patel

nststx, 2023-07-13:
Hi @nikunjp26,

Thanks for submitting your issue. We're currently working on different issues and will come back to you in the upcoming days.

Best regards and thanks for your patience,
Niklas Stoffers

nststx, 2023-07-18:
Hi @nikunjp26,

I've taken a look at your logs and from the EVSE side everything seems to be okay. I don't see a reason why the EV should close the TCP connection after the PaymentServiceSelectionResponse. The exchanged messages and parameters are normal and timeout requirements also met. Do you have any sort of debug information from the car? This reminds me of your issue ticket #173 just one step further. Just out of curiosity have you been able to charge a car from an **Indian** manufacturer with your charger yet? I have the feeling that there's something wrong with the way the car handles the charge service (like in #173). Both the Mahindra XUV400 and the Tata Nexon car behave very strange in that way...

Best regards,
Niklas Stoffers

nikunjp26, 2023-07-19:
Dear 

> Do you have any sort of debug information from the car?

no.

> Just out of curiosity have you been able to charge a car from an **Indian** manufacturer with your charger yet?

yes. Tata Nexon/Tigor/Tiago we already charged with single connector. 

best regards,
nikunj patel

nststx, 2023-07-20:
Hi @nikunjp26,

I've taken a look at your logs again with a colleague. We're both not sure why the car stops the communication after the PaymentServiceSelectionResponse. It's pretty strange as on communication level there's almost nothing that could go wrong at that stage. You said that you were already able to charge the Tata Nexon. Does this issue only occur with ISO and charging with DIN works fine?

Best regards,
Niklas Stoffers

nikunjp26, 2023-07-21:
Dear Niklas,

Yes DIN is working fine and facing issue with ISO only as per our trial.

Best Regards,
Nikunj Patel

nststx, 2023-07-21:
Hi @nikunjp26,

I've taken a look at your issue #170 again and in there you already had a log of a Tata Nexon which successfully charged using ISO. It seems to be a different car from the one you shared in this logs as the EVCC ID doesn't match. However was the successful charging session in #170 done with one of your chargers? I inspected both logs side by side and there's no difference on communication level until the EV suddenly decides to close the TCP connection.

Best regards,
Niklas Stoffers

nststx, 2023-08-01:
New Nextion logs by @nikunjp26:  [nextion_260723(1).zip](https://github.com/Sevenstax/FreeV2G/files/12226400/nextion_260723.1.zip)

Moved from Ticket #173 as issues got a little bit mixed up.

nststx, 2023-08-01:
Hi @nikunjp26,

I've taken a look at your logs. The car doesn't comply to the message timeouts defined in the ISO standard. It only waits about 400ms for the ChargeParameterDiscoveryResponse. However since you haven't set the schedules yet the Whitebeet will wait 1500ms to send the ChargeParameterDiscoveryResponse with Processing Status "Ongoing" to prevent unnecessary message exchanges. This timeout is compliant with the ISO standard.

You should be able to do a workaround for this issue by setting the schedules immediately after they are requested by the Whitebeet. However note that the whole process has to take less than 400ms so the car doesn't close the TCP connection.

Please let me know if the proposed solution works for you!

Best regards,
Niklas Stoffers

nikunjp26, 2023-08-01:
**Dear Niklas,**

We applied the changes suggest by you but still nextion car is not charging and it stuck at cable check request.

**Best Regards,
Nikunj Patel**

nststx, 2023-08-01:
Hi @nikunjp26,

Thanks for trying. Could you please capture another Wireshark log then? The issue should be something different now.

Best regards,
Niklas Stoffers

nikunjp26, 2023-08-07:
Dear Niklas,
Attached Wireshark log file with Tata Nextion in which schedule set response time is changed from 1sec to 100ms as per your suggestion. However, we didn't success to charge the Nextion EV.

[nextion_Schedule_time100ms.zip](https://github.com/Sevenstax/FreeV2G/files/12274890/nextion_Schedule_time100ms.zip)

Best Regards,
Nikunj Patel

nststx, 2023-08-07:
Hi @nikunjp26,

Thanks for coming back with the logs. I've had a quick look at them and the ChargeParameterDiscoveryResponse is now sent in time. However due to some reason that I'm currently not aware of the EV still decides to close the TCP connection right after that. I will take a more detailed look soon and let you know once I've got an update!

Best regards and thanks for your patience,
Niklas Stoffers

nststx, 2023-08-09:
Hi @nikunjp26,

I've taken a look at your logs again and I'm still not sure why the EV is closing the TCP connection after the ChargeParameterDiscoveryResponse. The exchanged parameters conform to the standard and timeout requirements are all met. You could try modifying the parameters slightly, for example already setting your Isolation Status to valid or setting the EVSE Status to ready before the ChargeParameterDiscovery. These modifications shouldn't be required for a successfull charging session, however I'm currently suspecting that the car doesn't conform to the ISO standard so maybe you can proceed with those modifications. The EVSE side of the communication is looking fine.

Please let me know, if you were able to proceed any further with the suggested steps. You can also check if there's an update available for the car. Maybe this has been a known issues which was fixed with newer software versions for the cars charging controller.

Best regards,
Niklas Stoffers

nikunjp26, 2023-08-25:
Dear Niklas,

We have improved the cp line wiring and now able to charge tata Nexon car.

Best regards
Nikunj Patel

nststx, 2023-08-28:
Hi @nikunjp26,

Thanks for coming back with that information. I'm glad that you're now able to perform a charging session with the car.

Best regards,
Niklas Stoffers

nststx, 2023-08-28:
Closing as this issue is resolved.

closed 2023-08-28 by nststx