Sevenstax/FreeV2G #197: Session Stopped after protocol shared

aimwho, 2023-07-10:
Setup::
Platform: EVSE
Firmware Version: i.e. V02_01_00
Host Controller Interface:  SPI
Host: FreeV2G (EV_V01_00_06) 

Hello,

I am trying to do the following:
Connect two raspberry pi's with EV and EVSE respectively.
Communicate via SPI and run the application Software.

ISSUE:
'EV'
### Welcome to Codico Whitebeet EV reference implementation
Initiating framing interface
Start SPI on bus 0 device 0 with 12000000 MHz
iface: SPI, name: spidev0.0
WHITE-beet-PI firmware version: V01_00_04
Set the CP mode to EV
Start the CP service
Set the CP state to State B
Set SLAC to EV mode
Start SLAC
Wait until an EVSE connects
EVSE connected
Start SLAC matching
SLAC matching successful
Set V2G mode to EV
Set V2G configuration
Set DC charging parameters
Set AC charging parameters
Create new charging session
ControlPilot duty cycle: 5.6
"Session started" received
        Protocol: 1
        Session ID: cc596ac7ba38599b
        EVSE ID: 44452a4132332a453435422a3738430000000000000000000000000000000000000000000000
        Payment method: 0
        Energy transfer mode: 0

"Session Stopped" received
Change State to State B
EV loop finished
Goodbye!

'EVSE'
###
Stops after SLAC matching and stays in a loop until new session start is received.

Is it necessary to make any kind of changes in the code that i am not aware of before starting the application?

Hope to get an answer from you soon.

Sushant Joshi

nststx, 2023-07-13:
Hi @aimwho,

Thanks for submitting your issue. We're currently working on different issues and will come back to you in the upcoming days.

Best regards,
Niklas Stoffers

nststx, 2023-07-18:
Hi @aimwho,

To further investigate your issue I will need the UART logs from both your EV and EVSE Whitebeet. Please capture the UART logs and upload them here. I'm also confused on the versions you're using. You have a V02_01_00 EVSE and a V01_00_04 EV Whitebeet and use it with FreeV2G tag EV_V01_00_06, right?

Best regards,
Niklas Stoffers

nststx, 2023-08-01:
Closing due to inactivity, if your issue still persists feel free to reopen!