Sevenstax/FreeV2G #199: Unexpected timeout after Start Charging Request (and subsequent unknown error code).

rwlyra, 2023-07-11:
**Your Setup**
Platform: EVSE
Firmware Version: V02_01_00
Host Controller Interface:  SPI
Host:  Own Implementation

**Describe the bug**
Whitebeet Eval board with V02_00_01 firmware acting as EVSE connected to Sevenstax EV simulator and controlled via SPI. 
I run into an unexpected timeout after successfully progressing through authorisation, cable check and precharge, and I'm not sure why.

The Whitebeet module sends us a Start Charging Request, and almost immediately after (less than 100ms) we receive a session error before we reply with the command to start charging (about 29 seconds into the charging session on the Saleae log).

Additionally, the response back from the Whitebeet to our Start Charging Command, seems to be 0x11, which is not a code included in table _10.1.5 Responses to Configuration Commands_  (User Manual 2.16)

**To Reproduce**
I have included a full log of SPI signals from Saleae Logic2 logic analyser.

**Expected behavior**
I was expecting to progress to the charge loop as per: _21.8.9 Charge Loop_

**Logs**
EV_SIM_11_07_23_.log is a log from the serial port of the EV simulator board.
EVSE_SIM_11_07_23_.log is a log from the serial port of the Whitebeet Eval board.
26_charging_session_250V_73A_stopped_11_07_23.sal (zipped for github) is a log of SPI signals from Saleae Logic2 logic analyser.
26_DecodedSPI_messages_11_07_23.docx is our decoded breakdown of the SPI messages from the Whitebeet board.

**Additional context**
Add any other context about the problem here.
[EV_SIM_11_07_23_.log](https://github.com/Sevenstax/FreeV2G/files/12016303/EV_SIM_11_07_23_.log)
[EVSE_SIM_11_07_23_.log](https://github.com/Sevenstax/FreeV2G/files/12016304/EVSE_SIM_11_07_23_.log)
[26_charging_session_250V_73A_stopped_11_07_23.zip](https://github.com/Sevenstax/FreeV2G/files/12016316/26_charging_session_250V_73A_stopped_11_07_23.zip)
[26_DecodedSPI_messages_11_07_23.docx](https://github.com/Sevenstax/FreeV2G/files/12016328/26_DecodedSPI_messages_11_07_23.docx)

nststx, 2023-07-13:
Hi @rwlyra,

Thanks for submitting your issue. We're currently working on different issues and will come back to you in the upcoming days.

Best regards and thanks for your patience,
Niklas Stoffers

nststx, 2023-07-18:
Hi @rwlyra,

Thanks again for your detailed issue description. I've taken a look at both your UART and SPI logs and I can confirm that there's a timeout happening. The Whitebeet module expects you to update the charging parameters at least every 500ms starting from your "Set Schedules" command. However between your last "Update DC charging parameters" message and the actual timeout message is a time delta of 526ms. To solve this you can either call the "Update DC charging parameters" command more often or, starting from EVSE firmware version V02_01_00 increase the timeout via the command "Set Session Parameter Timeout" (V2G, subId: 0x76). However note that the former approach is preferable because increasing the timeout is mostly used as a development feature and shouldn't be used in production.

This should also solve the 0x11 response code for the "Start charging" command. You get this response code because at that point the timeout already occured and thus the Whitebeet is in a wrong state for "Start charging".

I hope I was able to help. If you have any questions left, feel free to ask!

Best regards,
Niklas Stoffers

nststx, 2023-08-01:
Closing due to inactivity. If the proposed solution didn't fix your issue, please feel free to reopen!

closed 2023-08-01 by nststx