Sevenstax/FreeV2G #200: Documentation: Error in example for 18.10.13 Set Schedules (User Manual 2.16)

rwlyra, 2023-07-11:
There seems to be an error in the  Available power value in the second entry :
(Either that, or there's a strange inconsistency with the first tuple:

00 00 0E 10 (Duration of 1st entry: 3600s (1h))
**01 18 03** (Available power: **280kW**)
00 00 0E 10 (2nd entry starts after 3600s (1h))
00 01 43 70 (Duration of 2nd entry: 82800s (23h))
**00 05 01** (Available power: **50kW**)

nststx, 2023-07-13:
Hi @rwlyra,

Thanks for submitting your question. You're right the example given in the manual for the second schedule entry power value is incorrect. It should be `00 32 03` instead. Thanks for spotting this mistake. We will fix the example with an upcoming firmware release.

Best regards,
Niklas Stoffers

nststx, 2023-08-02:
Closing as this issue is fixed with the new User Manual release (v.2.17)

closed 2023-08-02 by nststx