Sevenstax/FreeV2G #201: Why EV is stopping after session started and EVSE is going on a time-out after start V2G

ArijitChatterjee-BlinkCharging, 2023-07-13:
EV is receiving the session started  and the protocol, session id, EVSE id, payment method and energy transfer mode is coming up and after that it is freezing. 
In case of the EVSE it is going to a time-out and the program is stopping. 
for reference please refer to uploaded documents.

EV is running on a raspberry pi 3 Model B+ and both Raspberry and Ev is connected to a router.
EVSE is running on a PC directly connected with a LAN cable.

After a good amount of time(3- 4 mins approx) EV is giving "Session stopped" received and gradually stopping the Application.py

Can you please provide a solution?

Firmware version of EV- V01_00_06
Firmware version of EVSE- V02_01_00
  
[ev_response.txt](https://github.com/Sevenstax/FreeV2G/files/12038501/ev_response.txt)
[evse_response.txt](https://github.com/Sevenstax/FreeV2G/files/12038503/evse_response.txt)

nststx, 2023-07-18:
Hi @ArijitChatterjee-BlinkCharging,

Thanks for your patience. I've taken a look at your issue and currently suspect that the FreeV2G version you're using is incompatible with your Whitebeet version (for example if I use FreeV2G EV_v1.0.6_1 with both EV and EVSE Whitebeet I get the exact same error). May I ask you which version you're using? You can also try using the version from the *master* branch as this should be compatible with both EV and EVSE version you're using.

Best regards,
Niklas Stoffers

ArijitChatterjee-BlinkCharging, 2023-07-25:
Hi,
 I am using the master branch.
The version for EVSE - v02_01_00 and EV - V01_00_06.
Can you please suggest what do next.
Best Regards
Arijit Chatterjee

nststx, 2023-07-25:
Hi @ArijitChatterjee-BlinkCharging,

In that case I will need a Wireshark log of your communication session. To capture such a log please enable the port mirror on your EVSE Whitebeet as this is directly connected to your computer. To enable port mirror please put `evse.whitebeet.networkConfigSetPortMirrorState(1)` before `evse.loop()` in the *Application.py* file.

If you need any further help in creating the log, please let me know!

Best regards,
Niklas Stoffers

closed 2023-08-31 by github-actions[bot]