Sevenstax/FreeV2G #202: Ford Mustang Mach-E charging current max 54A

Ivangel1972, 2023-07-14:
WB - that is, the WhiteBeet module has FW version 01_01_07

Hello, i have problem with Ford Mustang Mach-E
Session allways start propelly, but DC current never exceed 54A. Other cars charge with highest curren, depend on stattion / charger max possibilitties and/or car request.
Marcin

nststx, 2023-07-18:
Hi @Ivangel1972,

Thanks for submitting your issue. The DC current never exceeding 54A doesn't have to be an issue. The delivered current obviously depends on what the car requests and what the charging station is able to deliver. Maybe the car just never requests more than 54A in its CurrentDemandReq. However this is impossible to know without having access to the actual V2G communication. You're using communication via SPI, right? In that case access to the direct V2G communication is a bit more tricky, as you would've to enable the port mirror and capture the traffic on the Ethernet port of the module. However access to the Framing logs would also be sufficient. In that case I would need a capture of your SPI traffic from a Logic Analyzer. Could you create this log so I can further investigate this issue?

Best regards,
Niklas Stoffers

Ivangel1972, 2023-07-18:
Hi Niklas
We have more than 180 car charging stations, DC - to be clear- with your Whitebeet modules, in single and dual CCS versions, on which ALL other cars charge with the correct current, only one Ford machE charges with a max current of 54A. 
The same car on every other station from a different manufacturer: Delta, Efasec (!!!) or anything else can reach powers of over 120kW (even 200) - if the station allows. Our stations are a minimum of 40kW and nominally 80 - 160kW, where the maximum power is consumed by, for example, a Volvo truck driving for IKEA. 
So where do we limit the current ONLY for one car model?
No, I have no way of adding an ethernet port to the controllers in the stations, forwarding the port and doing logs.
For what it's worth, the data exchange via SPI between my CPU and a module or two has already been analysed by you twice and you have not found ANY error - so the problem is NOT with anything on our side.
Sincerely
@Ivangel1972

nststx, 2023-07-18:
Hi @Ivangel1972,

I can understand your frustration, however it's impossible to help without being able to see what is happening. The communication between car and charging station is complex and there are many things that could go wrong. I also don't know which logs that I've allegedly taken a look at you're referring to. If you're referring to #148 please note that this issue is being worked on by a colleague and not by me.

I'd really like to help you out but again my capabilities are limited if I don't have access to your logs. I do not need the whole V2G communication. As I've already said the framing messages would be sufficient. Thus could you please capture an SPI log of your charger charging with the Ford Mustang or let me know which log in #148 I should take a look at?

Best regards,
Niklas Stoffers

AlbertoGilardoni-eaton, 2023-07-18:
Hello,
We once had a similar issue with this very car and only with this one, and in our case it was the schedules power value that wasn't correctly configured, so the car was charging with a limited current. We had to set the Pmax value to the correct maximum value then the car was charging full power.
We tested with almost every EV on the market and we had this issue only with the Mach-E.
Well, I don't know if it's your case but any hints can help.
Best,
Alberto

AlbertoGilardoni-eaton, 2023-07-18:
Hi,
Ok it seems correct, so maybe it's something else.
Sorry it seemed too much the same issue I had.
I'd try to force between DIN and ISO to see if there's a difference, I know this car supports ISO because has Plug&Charge capability, but I don't know in which priority.
Regards,
Alberto

Ivangel1972, 2023-07-18:
Hi Alberto
This is the first car that this could cause a problem - so we haven't even had a chance to test it - I'll still check it carefully
Regards
@Ivangel1972

nststx, 2023-07-20:
Hi @Ivangel1972,

Your schedule settings seem to be correct, however if you want you can try and play around with them a bit and see if that helps with the issue. Please let me know what your testing results in. Otherwise I still would need the logs to help.

Best regards,
Niklas Stoffers

Ivangel1972, 2023-07-20:
Hi Niklas
We'll have Ford for testing tomorrow, I'll know the answer and let you know.
Regards
Ivangel1972

Ivangel1972, 2023-07-21:
Hi Hiklas
it was indeed a problem of the wrong power value in the schedule. Once corrected, the Ford charges with the correct current. Thank you for your help and I am closing the thread.
Regards
Ivangel1972

closed 2023-07-21 by Ivangel1972