Sevenstax/FreeV2G #203: Question:

FumiyaMiura-0626, 2023-07-25:
EV and EVSE are connected to separate Rasberry Pi 3B, router and LAN cable respectively, and EV and EVSE are connected to PE and CP.
When I ran Application.py for each of EV and EVSE, the process failed in v2gSetConfiguration on EV side.

Also, when v2gSetConfiguration was being executed on the EV side, the _sendReceiveAck was getting into the "response.payload[0] ! = 0:"

Can someone please tell me why I got that result?

EV : https://www.codico.com/en/wb-carrier-board-pi-1-1-pev-embedded-iso15118-sw-stack.
EVSE : https://www.codico.com/en/wb-carrier-board-ei-1-1-evse-embedded-iso15118-sw-stack-ev
Firmware version of EV-V01_01_07
Firmware version of EVSE- V01_00_04
Software and firmware versions are compatible.
[EVSE.log](https://github.com/Sevenstax/FreeV2G/files/12157365/EVSE.log)
[EV.log](https://github.com/Sevenstax/FreeV2G/files/12157366/EV.log)

adirschl, 2023-07-25:
Please update the boards to the latest firmware:
EVSE: V02_01_00
EV: V01_00_07
You can find the update files and tools in the Sevenstax folder on your CODICO PLC download area.

BR,
Andreas Dirschl
CODICO

FumiyaMiura-0626, 2023-07-25:
Dear Niklas 

Thanks for the answer.
I would like to confirm a few things before firmware updating.
Is the issue in the firmware update currently fixed?
https://github.com/Sevenstax/FreeV2G/discussions/193

Best Regards
FumiyaMiura

nststx, 2023-07-25:
Hi @FumiyaMiura-0626,

Thanks for submitting your issue. The note regarding the broken firmware update file is for EV firmware V01_00_06. We lately released a new firmware version for EV V01_00_07. Please use the firmware update file for EV release version V01_00_07 to update your board.

Please let me know if updating your firmware has resolved your issues!

Best regards,
Niklas Stoffers

FumiyaMiura-0626, 2023-07-26:
Dear Niklas

Thanks for the answer.

After updating to the latest firmware, I ran the Application.py in master branch ,
and I can confirm that it is working fine under my environment.
Thank you for your cooperation.

EVSE: V02_01_00
EV: V01_00_07

Best Regards
Fumiya Miura

closed 2023-07-26 by FumiyaMiura-0626