Sevenstax/FreeV2G #205: Python script: Run on Windows 10 platform

nikunjp26, 2023-07-26:
Following Error is getting when we tried to run script on Windows 10 (22H2) x64.

>python3 Application.py "Ethernet" -m c4:93:00:34:AE:5A -r EVSE

Welcome to Codico Whitebeet EVSE reference implementation
Initiating framing interface
iface: Ethernet, mac: c4:93:00:34:AE:5A
Traceback (most recent call last):
  File "C:\Users\Codico_Documents\FreeV2G-EVSE_v2.0.1_3\Whitebeet.py", line 157, in _sendReceive
    response = self.framing.receive_next_frame(filter_mod=[mod_id, 0xFF], filter_req_id=req_id, timeout=time_end - time_start)
  File "C:\Users\FreeV2G-EVSE_v2.0.1_3\FramingInterface.py", line 237, in receive_next_frame
    raise AssertionError("Frame reception timed out")
AssertionError: Frame reception timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Codico_Documents\FreeV2G-EVSE_v2.0.1_3\Whitebeet.py", line 105, in __init__
    self.slacStop()
  File "C:\Users\Codico_Documents\FreeV2G-EVSE_v2.0.1_3\Whitebeet.py", line 421, in slacStop
    response = self._sendReceive(self.slac_mod_id, self.slac_sub_stop, None)
  File "C:\Users\Codico_Documents\FreeV2G-EVSE_v2.0.1_3\Whitebeet.py", line 172, in _sendReceive
    raise ConnectionError("Problem with send/receive - Please check your connection!")
ConnectionError: Problem with send/receive - Please check your connection!

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Codico_Documents\FreeV2G-EVSE_v2.0.1_3\Application.py", line 57, in <module>
    with Evse(args.interface, args.mac) as evse:
  File "C:\Users\Codico_Documents\FreeV2G-EVSE_v2.0.1_3\Evse.py", line 8, in __init__
    self.whitebeet = Whitebeet(iface, mac)
  File "C:\Users\Codico_Documents\FreeV2G-EVSE_v2.0.1_3\Whitebeet.py", line 109, in __init__
    raise ConnectionError("Failed to initialize the framing interface on \"{}\"".format(self.framing.sut_interface))
ConnectionError: Failed to initialize the framing interface on "Ethernet"

For interface connectivity, checked on Wireshark and we got the WhiteBeat Init packet in WireShark.

Best Reagrds,
Nikunj Patel

nststx, 2023-07-27:
Hi @nikunjp26,

Thanks for submitting your issue. Are you sure that you're selecting the correct network interface? Please try to capture a Wireshark log on that interface, reset the Whitebeet and check whether you get the `Network: Interface Up` packet on that interface. Or is that what you already tried and meant with "Whitebeet Init packet"?

Furthermore do you actually see the initial `Get Firmware Version` packet that is sent by FreeV2G on the interface? If you capture another log please upload it here so i can take a look.

Best regards,
Niklas Stoffers

nikunjp26, 2023-08-01:
Dear Niklas ,
Please take a look of attached Wireshark logs.

[wb_windows_pyscript_log.zip](https://github.com/Sevenstax/FreeV2G/files/12224501/wb_windows_pyscript_log.zip)

Best Regards,
Nikunj Patel

nikunjp26, 2023-08-25:
Dear Niklas ,
We checked as per your reply and there are packets are showing when we used sniff command. However, still it is not working.
Here I attached snapshot.

[image]

nststx, 2023-08-29:
Hi @nikunjp26,

Thanks for coming back with that information. The output that you get looks fine. I currently suspect that this is some kind of permissions problem. Do you start the application as administrator? If not could you please try if that has any effect?

Best regards,
Niklas Stoffers

nikunjp26, 2023-09-01:
Dear Niklas,
I tried with Admin Mode to run application. However, it is not working.

Best Regards,
Nikunj Patel

lho-stx, 2023-09-04:
Hey @nikunjp26,

thank you for trying out! Can you tell me what the output of `ipconfig` is? I just need the interface names (you can remove MAC and IP addresses).

Best regards,
lho

nststx, 2023-09-14:
Hi @nikunjp26,

Thanks for coming back with the screenshot. The output is what's expected. I've tried reproducing your issue but have yet failed to do so. Thus it's a little hard to debug. Could you do me a favor and checkout the *packet-debugs* branch of this repository, run FreeV2G and attach the output here? I added some debug prints in that version which I hope help with finding the source of this issue.

Best regards,
Niklas Stoffers

nikunjp26, 2023-09-19:
Dear Niklas,
Here I attached snapshot of packet-debugs branch output.

[image]

Best Regards,
Nikunj Patel

lho-stx, 2023-09-29:
Hey @nikunjp26,

it seems the packets are getting lost on the way from the interface to the FreeV2G application. We can see the packet in Wireshark, so we know it needs to be there. In the branch you tested the sniffing process is started, but no packet is seen at all. When using the scapy standalone the packets are seen. Unfortunately we can not reproduce the problem on our side. 

Do you use a restricted PC, where only specific programs are allowed to record on interfaces? Maybe a installed firewall is prevent FreeV2G from operating normally. But it would be strange if Wireshark and standalone scapy is working and FreeV2G not.

Best regards,
lho

nikunjp26, 2023-10-16:
Dear lho,
We disabled the Firewall setting from our Laptop but still facing the same problem.
[image]

Best Regards,
Nikunj Patel

nikunjp26, 2023-11-09:
Dear Team,

any update on this?

best regards,
Nikunj patel

lho-stx, 2023-12-14:
Hey @nikunjp26,

I am sorry for the long delay. There are not any news and we can not reproduce your problem. There are many reasons why the setup may fail and debugging customer environments in this detail is out of scope of this issue tracker. 

I am sorry for your inconvinence,
lho

closed 2023-12-14 by lho-stx